coolsnowwolf/lede #7413: Mi R3P 无法 sysupgrade

xylz0928, 2021-07-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：小米路由3 Pro在5.4内核固件下，无法通过网页升级到最新的5.10.50版本固件
在手动pbboot全新刷入5.10.50后，成功使用5.10版本。
但在5.10.50页面进行升级，仍然提示固件不支持。
**注意：是5.10上升5.10都不支持**
不支持所上传的映像文件格式，请选择适合当前平台的通用映像文件。

### (2) 路由器型号和固件版本

A：Xiaomi MT7621 R3P， kernel 5.10.50

### (3) 详细日志

A：
Sat Jul 17 15:19:26 2021 daemon.err uhttpd[3169]: Sat Jul 17 15:19:26 CST 2021 upgrade: The device is supported, but the config is incompatible to the new image (1.0->1.1). Please upgrade without keeping config (sysupgrade -n).
Sat Jul 17 15:19:26 2021 daemon.err uhttpd[3169]: Sat Jul 17 15:19:26 CST 2021 upgrade: Config cannot be migrated from swconfig to DSA

[image]

xylz0928, 2021-07-17:
[image]

xylz0928, 2021-07-17:
已解决：清除原配置的/etc/config/下system, network, dhcp后可在线更新

closed 2021-07-17 by xylz0928

xsj520, 2021-07-21:
咋清除

wmyfelix, 2021-07-22:
> 已解决：清除原配置的/etc/config/下system, network, dhcp后可在线更新

还是不行我这边..