coolsnowwolf/lede #7449: 内核日志这个怎么解决？

qqsir-dev, 2021-07-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
[   37.114415] squashfs: Unknown parameter 'discard'
[   37.115007] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0xd44d89e4)
[   37.115184] F2FS-fs (sda128): Can't find valid F2FS filesystem in 1th superblock
[   37.115479] F2FS-fs (sda128): Magic Mismatch, valid(0xf2f52010) - read(0x80a345bd)
[   37.115653] F2FS-fs (sda128): Can't find valid F2FS filesystem in 2th superblock
[   37.116708] exFAT-fs (sda128): invalid boot record signature
[   37.116841] exFAT-fs (sda128): failed to read boot sector
[   37.116966] exFAT-fs (sda128): failed to recognize exfat type
[   37.933029] br-lan: port 1(eth0) entered disabled state
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

Boos4721, 2021-07-20:
`Can't find valid F2FS filesystem`
编译时带上F2FS支持

qqsir-dev, 2021-07-20:
> `Can't find valid F2FS filesystem`
> 编译时带上F2FS支持

请问怎么操作？谢谢

Boos4721, 2021-07-20:
> > `Can't find valid F2FS filesystem`
> > 编译时带上F2FS支持
> 
> 请问怎么操作？谢谢

`CONFIG_PACKAGE_kmod-fs-f2fs=y`

qqsir-dev, 2021-07-20:
> > > `Can't find valid F2FS filesystem`
> > > 编译时带上F2FS支持
> > 
> > 
> > 请问怎么操作？谢谢
> 
> `CONFIG_PACKAGE_kmod-fs-f2fs=y`

谢谢，不过这个我是要加到哪个文件中去？

qqsir-dev, 2021-07-20:
> > > `Can't find valid F2FS filesystem`
> > > 编译时带上F2FS支持
> > 
> > 
> > 请问怎么操作？谢谢
> 
> `CONFIG_PACKAGE_kmod-fs-f2fs=y`

我找到在那里，但是里面是有这一条的

qqsir-dev, 2021-07-22:
不勾选automount就没有这个错误了

closed 2021-07-22 by qqsir-dev