coolsnowwolf/lede #7451: 添加qemu-ga代理下pve执行“shutdown”，代替openwrt的关机指令”poweroff“。

naoki66, 2021-07-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：开启QEMU Guest Agent后OP虚拟机无法在pve的管理界面上正常关机/重启操作

### (2) 路由器型号和固件版本

A：X64

### (3) 详细日志

A：
目前在Actions下的笨办法解决方案，不知道有没有更好的办法，不会pull，见谅。
DIY加上
#添加qemu-ga代理下pve执行“shutdown”，代替openwrt的关机指令”poweroff“。
mkdir -p ~/openwrt/files/sbin
touch ~/openwrt/files/sbin/shutdown
echo -e '#!/bin/sh\nbusybox poweroff' > ~/openwrt/files/sbin/shutdown
chmod 0755 ~/openwrt/files/sbin/shutdown

hahach, 2021-07-20:
pve7.0下开了ga的，可以直接关机了，没注意到是怎么实现的。
6.x版本的时候确实不行

closed 2021-07-20 by coolsnowwolf

naoki66, 2021-07-22:
check_data_file_clashes: Package qemu-ga wants to install file ~/lede/build_dir/target-x86_64_musl/root-x86/sbin/shutdown
But that file is already provided by package * open-vm-tools
opkg_install_cmd: Cannot install package qemu-ga.