coolsnowwolf/lede #7456: 请问从5.4.x内核保留或者导入配置升上来原来那个luci web下root登录密码错误的问题解决了吗？

galaxyskyknight, 2021-07-20:
由于装了很多插件，有很多配置，如果要全新安装后再重新一个个配，那简直是要人抓狂....但原来6月份的code好像有一个bug,保留配置升级上来luci web的root密码会登录不进去，即便后台ssh改了root密码也不行，请问这个bug解决了吗，root cause是什么呢？解决了的话我想保留配置升一下新内核的code基线版本

谢谢！

acooler15, 2021-07-20:
我以前是不保留配置升级后，将之前的备份解压，删掉不需要的，用ftp覆盖路由器上的，或者删掉不要的文件后重新打包再传到路由器上恢复配置。

cncqcyc, 2021-07-21:
建议还是新内核重新配置一下，我之前将5.4的配置删掉不用的编译进5.10里，结果出现了dns解析异常，后来重新全新配置一遍正常了

galaxyskyknight, 2021-07-22:
> 
> 
> 建议还是新内核重新配置一下，我之前将5.4的配置删掉不用的编译进5.10里，结果出现了dns解析异常，后来重新全新配置一遍正常了

插件配置太繁琐了.....真的很头疼，我现在都是用ghost备份软路由硬盘后才敢升级

hahach, 2021-07-22:
我给你个我的解决方法，不排除有不明bug，建议先做好各种备份。
首先先5.4下做好备份，然后将备份文件里rpcd和uhttpd的配置文件删了，如果你配置过证书的，把证书也删了。
然后，用新固件启动，ssh把rpcd和uhttpd的配置文件考出来。然后还原设置，重启后应该就可以重新登陆和配置uhttpd了（如果不行就ssh把刚刚备份的俩配置导回去，/etc/init.d重启这俩服务，再开网页看看）

EnnawYang, 2021-07-22:
不是bug，不存在解决的说法。建议重新配置

kingzleshe, 2021-07-22:
> 我给你个我的解决方法，不排除有不明bug，建议先做好各种备份。
> 首先先5.4下做好备份，然后将备份文件里rpcd和uhttpd的配置文件删了，如果你配置过证书的，把证书也删了。
> 然后，用新固件启动，ssh把rpcd和uhttpd的配置文件考出来。然后还原设置，重启后应该就可以重新登陆和配置uhttpd了（如果不行就ssh把刚刚备份的俩配置导回去，/etc/init.d重启这俩服务，再开网页看看）

docker试了，成功升级恢复配置 666

yunxi993, 2021-07-22:
1

galaxyskyknight, 2021-07-23:
> 
> 
> 不是bug，不存在解决的说法。建议重新配置

说了跟没说一样，不解决问题。建议向楼上学习，或者参考下面
https://github.com/coolsnowwolf/lede/issues/7393#issuecomment-880524449

EnnawYang, 2021-07-23:
> > 不是bug，不存在解决的说法。建议重新配置
> 
> 说了跟没说一样，不解决问题。建议向楼上学习，或者参考下面
> [#7393 (comment)](https://github.com/coolsnowwolf/lede/issues/7393#issuecomment-880524449)

我问你搜索过历史issue没有？ 没有吧？
https://github.com/coolsnowwolf/lede/issues/7083#issuecomment-864354006

galaxyskyknight, 2021-07-23:
> 
> 
> 我给你个我的解决方法，不排除有不明bug，建议先做好各种备份。
> 首先先5.4下做好备份，然后将备份文件里rpcd和uhttpd的配置文件删了，如果你配置过证书的，把证书也删了。
> 然后，用新固件启动，ssh把rpcd和uhttpd的配置文件考出来。然后还原设置，重启后应该就可以重新登陆和配置uhttpd了（如果不行就ssh把刚刚备份的俩配置导回去，/etc/init.d重启这俩服务，再开网页看看）

回头试试你这个方子。谢谢

galaxyskyknight, 2021-07-23:
> 
> 
> > > 不是bug，不存在解决的说法。建议重新配置
> > 
> > 
> > 说了跟没说一样，不解决问题。建议向楼上学习，或者参考下面
> > [#7393 (comment)](https://github.com/coolsnowwolf/lede/issues/7393#issuecomment-880524449)
> 
> 我问你搜索过历史issue没有？ 没有吧？
> [#7083 (comment)](https://github.com/coolsnowwolf/lede/issues/7083#issuecomment-864354006)
搜索过，抱歉眼睛不好，没看到。
另外，我也没说这个是bug,而是个问题，还有，如果是#7083里的方法能解决，可以在init脚本或者导入配置的时候做个sanity check,自动更新一下路径就可以修复。
最后，我觉得开发者，可以有傲骨，但不可以有傲气。有则改之无则加勉。

hahach, 2021-07-23:
> 搜索过，抱歉眼睛不好，没看到。
> 另外，我也没说这个是bug,而是个问题，还有，如果是#7083里的方法能解决，可以在init脚本或者导入配置的时候做个sanity check,自动更新一下路径就可以修复。
> 最后，我觉得开发者，可以有傲骨，但不可以有傲气。有则改之无则加勉。

我当时的解决方法是试出来的，操作后可以登陆却发现是有uhttp的证书错误以及无法修改uhttp配置的问题，然后现在这个是解决了这个问题的方法，我自己没试出其他的bug，所以开头建议大家先做好备份。

这个法子也是苦于上一次完整配置是两三年前做的，有很多配置都不记得怎么改的了，没办法只能花时间测试的怎么样能保留配置升级，结果目前看来成功了，也暂时没发现其他问题，所以分享出来或许有其他相同烦恼的伙伴可以减少重新配置的痛苦。

galaxyskyknight, 2021-07-23:
> 我当时的解决方法是试出来的，操作后可以登陆却发现是有uhttp的证书错误以及无法修改uhttp配置的问题，然后现在这个是解决了这个问题的方法，我自己没试出其他的bug，所以开头建议大家先做好备份。 [...]

我看了一下，uhttpd新旧配置没有什么变化，应该不需要改，你说uhttpd无法操作可能是rpcd依赖的问题。rpcd新版的路径改了，看上去像是一个用本地unix domain的socket来提供系统rpc交互的配置。应该是luci uhttpd也有用。其实这个真的新版代码写在init脚本和导入配置的脚本里检查一下内核以及做些逻辑判断并用sed插入一下就能解决的问题，可能觉得我们这些伸手党们辛辛苦苦做起来的配置不值钱，保留配置升级懒得去diag各种疑难杂症，也不管用户的运维成本，干脆一刀切，你们折腾机器比折腾我去看issue来得简单吧....不懂.....

kingzleshe, 2021-07-23:
用docker吧，之前也是用起来总有小问题，把树莓派干脆刷了ubuntu.
现在我OP用的docker-mini版本，也就配个passwall和端口转发。其他都不配置。
比如别的ddns我也不配在OP上。单独配个容器用ddclient。
几乎所有应用容器化后包括OP。整体更方便。
自己能搞定的还是尽量自己搞定，虽然花了不少时间。

acooler15, 2021-07-23:
> 用docker吧，之前也是用起来总有小问题，把树莓派干脆刷了ubuntu.
> 现在我OP用的docker-mini版本，也就配个passwall和端口转发。其他都不配置。
> 比如别的ddns我也不配在OP上。单独配个容器用ddclient。
> 几乎所有应用容器化后包括OP。整体更方便。
> 自己能搞定的还是尽量自己搞定，虽然花了不少时间。

有空试试，我的树莓派装了内核5.4.136的openwrt后（U盘启动），每个一段时间就死了，发现是过一段时间就不读盘了，很奇怪，之前版本的openwrt到没发现这种问题。

hahach, 2021-07-23:
> > > > > > 不是bug，不存在解决的说法。建议重新配置
> > > > > 
> > > > > 
> > > > > 说了跟没说一样，不解决问题。建议向楼上学习，或者参考下面
> > > > > [#7393 (comment)](https://github.com/coolsnowwolf/lede/issues/7393#issuecomment-880524449)
> > > > 
> > > > 
> > > > 我问你搜索过历史issue没有？ 没有吧？
> > > > [#7083 (comment)](https://github.com/coolsnowwolf/lede/issues/7083#issuecomment-864354006)
> > > 
> > > 
> > > 搜索过，抱歉眼睛不好，没看到。
> > > 另外，我也没说这个是bug,而是个问题，还有，如果是#7083里的方法能解决，可以在init脚本或者导入配置的时候做个sanity check,自动更新一下路径就可以修复。
> > > 最后，我觉得开发者，可以有傲骨，但不可以有傲气。有则改之无则加勉。
> > 
> > 
> > 我当时的解决方法是试出来的，操作后可以登陆却发现是有uhttp的证书错误以及无法修改uhttp配置的问题，然后现在这个是解决了这个问题的方法，我自己没试出其他的bug，所以开头建议大家先做好备份。
> > 这个法子也是苦于上一次完整配置是两三年前做的，有很多配置都不记得怎么改的了，没办法只能花时间测试的怎么样能保留配置升级，结果目前看来成功了，也暂时没发现其他问题，所以分享出来或许有其他相同烦恼的伙伴可以减少重新配置的痛苦。
> 
> 我看了一下，uhttpd新旧配置没有什么变化，应该不需要改，你说uhttpd无法操作可能是rpcd依赖的问题。rpcd新版的路径改了，看上去像是一个用本地unix domain的socket来提供系统rpc交互的配置。应该是luci uhttpd也有用。其实这个真的新版代码写在init脚本和导入配置的脚本里检查一下内核以及做些逻辑判断并用sed插入一下就能解决的问题，可能觉得我们这些伸手党们辛辛苦苦做起来的配置不值钱，保留配置升级懒得去diag各种疑难杂症，也不管用户的运维成本，干脆一刀切，你们折腾机器比折腾我去看issue来得简单吧....不懂.....在我看来,不管是大厂还是开源，不管是家用还是商用，通信网络设备的software，保留配置升级是一个基本盘，即便可能因为官方源升级可能会带来插件配置兼容问题，但要zero config的upgrade可能是个灾难，至少具备这个选项，把这个选择权留给用户。不过，既然免费的，这点念想不敢痴心太多。

我遇到的uhttpd问题主要还是我配置了本地域名的证书。改完rpcd后能登陆了，但是证书报错，uhttpd其他功能都正常，就是不能生成新证书，用openssl生成的也不能用，改uhttpd配置里证书的部分也不行。也不知道什么毛病。但是还原备份的时候连uhttpd的一起删就解决了。

对于升级保留配置我的观点也是类似，除非完全不兼容，否则还是希望维护者至少能指点一下可能的解决方案，可以做“不推荐且有可能有严重bug”的免责声明嘛，我这次尝试也是源于coolsnowwolf提及是rpcd配置里的ubus路径更改了，才试出来的。否则我宁愿不升级，一直停留版本。但是这样以后有可能新功能就没法用了，很是可惜。

closed 2021-08-15 by AmadeusGhost