coolsnowwolf/lede #7459: 一段时间就无法上网只能重新麻烦大家看下啦

abclai, 2021-07-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述
以前很正常，现在上网一段时间只能重启要不上不网，网络连接什么是正常的只能重启要不就不能上网

### (1) 具体问题
以前很正常，现在上网一段时间只能重启要不上不网，网络连接什么是正常的只能重启要不就不能上网
A：

### (2) 路由器型号和固件版本
X86

OpenWrt R21.7.100
--

</body>
</html>
A：

### (3) 详细日志
Wed Jul 21 00:35:05 2021 daemon.err nlbwmon[7631]: Netlink receive failure: Out of memory
Wed Jul 21 00:35:05 2021 daemon.err nlbwmon[7631]: Unable to dump conntrack: No buffer space available
Wed Jul 21 02:00:00 2021 cron.err crond[3110]: USER root pid 16473 cmd /usr/share/shadowsocksr/ssrplusupdate.sh >/var/log/ssrplus.log
Wed Jul 21 02:18:55 2021 daemon.err nlbwmon[7631]: Netlink receive failure: Out of memory
Wed Jul 21 02:18:55 2021 daemon.err nlbwmon[7631]: Unable to dump conntrack: No buffer space available
Wed Jul 21 05:06:59 2021 daemon.err nlbwmon[7631]: Netlink receive failure: Out of memory
Wed Jul 21 05:06:59 2021 daemon.err nlbwmon[7631]: Unable to dump conntrack: No buffer space available
Wed Jul 21 07:55:02 2021 daemon.err nlbwmon[7631]: Netlink receive failure: Out of memory
Wed Jul 21 07:55:02 2021 daemon.err nlbwmon[7631]: Unable to dump conntrack: No buffer space available

A：

Hey-Leo, 2021-07-21:
Out of memory 翻译一下

VergilGao, 2021-07-22:
内存炸了，重启，开ssh上去htop或者top一下看看哪个小可爱在吃内存

qqsir-dev, 2021-07-22:
到 /etc/config/下 编辑nlbwmon 按照下面设置就行了
option netlink_buffer_size 10485760
具体什么原理根本不懂，反正这样设置就行了

closed 2024-11-19 by coolsnowwolf