coolsnowwolf/lede #7463: 开启Shortcut-FE后，向日葵连接不正常，连上就卡死界面，流量0kb/s

softtip, 2021-07-21:
开启Shortcut-FE后，在路由器后面的被控端，向日葵连接不正常，连上就卡死界面，流量0kb/s
经反复测试确实是由于Shortcut-FE造成，Shortcut-FE关闭后一切正常，这个能解决吗？
以前还遇到过监控摄像头软件也是类似情况的，也是关闭Shortcut-FE就好了

suizhenmin, 2021-07-21:
编译的时候换一个

[image]

softtip, 2021-07-21:
flow offload和Short-Fe其他方面有什么区别呢？

suizhenmin, 2021-07-21:
> […](#)
默认好像是大神魔改的，默认我用起来也有问题，也是在远程桌面的时候卡画面，不过不是向日葵。

jarod360, 2021-07-22:
测试了，向日葵可以链接上，微软的桌面也没问题。

suizhenmin, 2021-07-22:
mac 上远程桌面软件非微软的版本连接不上，只能使用非魔改的turnoacc才可以

zhouxiaolong920, 2021-08-06:
什么Shortcut-FE，什么是TouberACC.
都不要开启。。。。

以后也不要用这个东西。

XiaoliChan, 2021-08-24:
SFE actually cause this issue, when Sunlogin remote access connection establishes, it stuck on my remote screen and I can't do anything.

moonerick, 2021-09-11:
我是使用这个后随机性死机，取消编译安装后7天了，没用重启、死机问题

handbye, 2024-08-04:
> 开启Shortcut-FE后，在路由器后面的被控端，向日葵远控软件连接不正常，连上就卡死界面，流量0kb/s 经反复测试确实是由于Shortcut-FE造成，Shortcut-FE关闭后一切正常，这个能解决吗？ 以前还遇到过监控摄像头软件也是类似情况的，也是关闭Shortcut-FE就好了

一样的问题，开启后小米摄像头直接连不上，排查了半天，我还换了个摄像头，真坑啊。

closed 2024-11-19 by coolsnowwolf