coolsnowwolf/lede #7464: r3g不能刷入5.10.50的固件，求解

xsj520, 2021-07-21:
[image]
[image]

AmadeusGhost, 2021-07-21:
从 breed 里面刷

xsj520, 2021-07-21:
从breed里刷入后，系统启动不了。我又只好刷回5.10.46了。。。。。我觉得是5.10.50内核的问题

Lrvineshaw, 2021-07-21:
> 从breed里刷入后，系统启动不了。我又只好刷回5.10.46了。。。。。我觉得是5.10.50内核的问题

你可以看看前面的issues，好几个人都问了，事实确实如此

1582130940, 2021-07-21:
要不参考下这个？[https://github.com/coolsnowwolf/lede/issues/7413#issuecomment-881851618](url)

xsj520, 2021-07-21:
试了没有用。

xylz0928, 2021-07-21:
> 试了没有用。

你描述的这个现象跟我这边的不太一样。
我是全新刷入后可以启动并正常使用的，只是当时不能继承配置继续更新而已，用我那个方法才可以解决。

wuelliott, 2021-07-22:
> 试了没有用。

你可以试试我的方法：
1、在R3G的5.10.46固件系统下，把5.10.50固件用WINSCP上传到/tmp里
2、用SSH登录R3G，输入sysupgrade -F -n /tmp/固件名
我就是这样升级成功的，这个方法刷入后会自动重启，不保留配置！

shidahuilang, 2021-07-24:
一样的问题······

fztyp, 2021-08-06:
同问，breed刷入不启动，WEB控制台刷不进

Masterchiefm, 2021-08-15:
> > 试了没有用。
> 
> 你可以试试我的方法：
> 1、在R3G的5.10.46固件系统下，把5.10.50固件用WINSCP上传到/tmp里
> 2、用SSH登录R3G，输入sysupgrade -F -n /tmp/固件名
> 我就是这样升级成功的，这个方法刷入后会自动重启，不保留配置！

谢谢！ 问题解决！

fztyp, 2021-08-16:
谢谢，已成功刷入

closed 2021-09-16 by AmadeusGhost