coolsnowwolf/lede #7465: nlbwmon日志报内存不足，但是明明可用内存还有很多

db-one, 2021-07-21:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：[image]
nlbwmon日志报内存不足，但是明明可用内存还有很多
虚拟机下运行的，分了1G内存，空闲内存600-700M左右

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：
[image]

Wed Jul 21 16:54:02 2021 daemon.err nlbwmon[17653]: Netlink receive failure: Out of memory
Wed Jul 21 16:54:02 2021 daemon.err nlbwmon[17653]: Unable to dump conntrack: No buffer space available

qqsir-dev, 2021-07-22:
到 /etc/config/下 编辑nlbwmon 按照下面设置就行了
option netlink_buffer_size 10485760

db-one, 2021-07-22:
好的，谢谢，已经设置好了，再观察一下，翻译了一下大意是

	# The buffer size for receiving netlink conntrack results, in bytes.
	# If the chosen size is too small, accounting information might get
	# lost, leading to skewed traffic counting results

	# 用于接收netlink conntrack结果的缓冲区大小，字节数。
	# 如果选择的大小太小，会计信息可能会被
	# 丢失，导致流量统计结果的偏差

Neverends4, 2021-07-23:
如果你内网流量大，改了可能也溢出，只能缓解。这貌似是nlbwmon的bug，好几个人提了类似issue但是没更新。

db-one, 2021-07-23:
改了确实没报错了，希望大佬早日修复
话说溢出会有什么后果，CPU内存占用高吗？

EnnawYang, 2021-07-23:
早就修改过了 没想到不能满足你们

db-one, 2021-07-23:
> 早就修改过了 没想到不能满足你们

ᥬ😂᭄ᥬ😂᭄ᥬ😂᭄ᥬ🌚᭄话说我流量也不高啊，基本都是空载状态，也不怎么看视频 看电视，就是工作看看网页

bonnyandsky, 2021-11-02:
不单单是x86 我mt7621也有这个报错，看了下日志，大概3小时不到一点报错一次。具体看日志
[image]

closed 2024-11-19 by coolsnowwolf