coolsnowwolf/lede #7469: 太多错误。。。

308066400, 2021-07-21:
ERROR: please fix package/lean/luci-app-accesscontrol/Makefile - see logs/package/lean/luci-app-accesscontrol/dump.txt for details
ERROR: please fix package/lean/luci-app-adbyby-plus/Makefile - see logs/package/lean/luci-app-adbyby-plus/dump.txt for details
ERROR: please fix package/lean/luci-app-airplay2/Makefile - see logs/package/lean/luci-app-airplay2/dump.txt for details
ERROR: please fix package/lean/luci-app-amule/Makefile - see logs/package/lean/luci-app-amule/dump.txt for details
ERROR: please fix package/lean/luci-app-arpbind/Makefile - see logs/package/lean/luci-app-arpbind/dump.txt for details
ERROR: please fix package/lean/luci-app-autoreboot/Makefile - see logs/package/lean/luci-app-autoreboot/dump.txt for details
ERROR: please fix package/lean/luci-app-baidupcs-web/Makefile - see logs/package/lean/luci-app-baidupcs-web/dump.txt for details
ERROR: please fix package/lean/luci-app-cifs-mount/Makefile - see logs/package/lean/luci-app-cifs-mount/dump.txt for details
ERROR: please fix package/lean/luci-app-cifsd/Makefile - see logs/package/lean/luci-app-cifsd/dump.txt for details
ERROR: please fix package/lean/luci-app-cpufreq/Makefile - see logs/package/lean/luci-app-cpufreq/dump.txt for details
ERROR: please fix package/lean/luci-app-dnsforwarder/Makefile - see logs/package/lean/luci-app-dnsforwarder/dump.txt for details
ERROR: please fix package/lean/luci-app-docker/Makefile - see logs/package/lean/luci-app-docker/dump.txt for details
ERROR: please fix package/lean/luci-app-easymesh/Makefile - see logs/package/lean/luci-app-easymesh/dump.txt for details
ERROR: please fix package/lean/luci-app-familycloud/Makefile - see logs/package/lean/luci-app-familycloud/dump.txt for details
ERROR: please fix package/lean/luci-app-filetransfer/Makefile - see logs/package/lean/luci-app-filetransfer/dump.txt for details
ERROR: please fix package/lean/luci-app-frpc/Makefile - see logs/package/lean/luci-app-frpc/dump.txt for details
ERROR: please fix package/lean/luci-app-frps/Makefile - see logs/package/lean/luci-app-frps/dump.txt for details
ERROR: please fix package/lean/luci-app-guest-wifi/Makefile - see logs/package/lean/luci-app-guest-wifi/dump.txt for details
ERROR: please fix package/lean/luci-app-haproxy-tcp/Makefile - see logs/package/lean/luci-app-haproxy-tcp/dump.txt for details
ERROR: please fix package/lean/luci-app-ipsec-vpnd/Makefile - see logs/package/lean/luci-app-ipsec-vpnd/dump.txt for details
ERROR: please fix package/lean/luci-app-jd-dailybonus/Makefile - see logs/package/lean/luci-app-jd-dailybonus/dump.txt for details
ERROR: please fix package/lean/luci-app-kodexplorer/Makefile - see logs/package/lean/luci-app-kodexplorer/dump.txt for details
ERROR: please fix package/lean/luci-app-music-remote-center/Makefile - see logs/package/lean/luci-app-music-remote-center/dump.txt for details
ERROR: please fix package/lean/luci-app-mwan3helper/Makefile - see logs/package/lean/luci-app-mwan3helper/dump.txt for details
ERROR: please fix package/lean/luci-app-n2n_v2/Makefile - see logs/package/lean/luci-app-n2n_v2/dump.txt for details
ERROR: please fix package/lean/luci-app-netdata/Makefile - see logs/package/lean/luci-app-netdata/dump.txt for details
ERROR: please fix package/lean/luci-app-nfs/Makefile - see logs/package/lean/luci-app-nfs/dump.txt for details
ERROR: please fix package/lean/luci-app-nft-qos/Makefile - see logs/package/lean/luci-app-nft-qos/dump.txt for details
ERROR: please fix package/lean/luci-app-nps/Makefile - see logs/package/lean/luci-app-nps/dump.txt for details
ERROR: please fix package/lean/luci-app-openvpn-server/Makefile - see logs/package/lean/luci-app-openvpn-server/dump.txt for details
ERROR: please fix package/lean/luci-app-pppoe-relay/Makefile - see logs/package/lean/luci-app-pppoe-relay/dump.txt for details
ERROR: please fix package/lean/luci-app-pptp-server/Makefile - see logs/package/lean/luci-app-pptp-server/dump.txt for details
ERROR: please fix package/lean/luci-app-ps3netsrv/Makefile - see logs/package/lean/luci-app-ps3netsrv/dump.txt for details
ERROR: please fix package/lean/luci-app-qbittorrent/Makefile - see logs/package/lean/luci-app-qbittorrent/dump.txt for details
ERROR: please fix package/lean/luci-app-ramfree/Makefile - see logs/package/lean/luci-app-ramfree/dump.txt for details
ERROR: please fix package/lean/luci-app-rclone/Makefile - see logs/package/lean/luci-app-rclone/dump.txt for details
ERROR: please fix package/lean/luci-app-samba4/Makefile - see logs/package/lean/luci-app-samba4/dump.txt for details
ERROR: please fix package/lean/luci-app-softethervpn/Makefile - see logs/package/lean/luci-app-softethervpn/dump.txt for details
ERROR: please fix package/lean/luci-app-ssrserver-python/Makefile - see logs/package/lean/luci-app-ssrserver-python/dump.txt for details
ERROR: please fix package/lean/luci-app-syncdial/Makefile - see logs/package/lean/luci-app-syncdial/dump.txt for details
ERROR: please fix package/lean/luci-app-ttyd/Makefile - see logs/package/lean/luci-app-ttyd/dump.txt for details
ERROR: please fix package/lean/luci-app-turboacc/Makefile - see logs/package/lean/luci-app-turboacc/dump.txt for details
ERROR: please fix package/lean/luci-app-unblockmusic/Makefile - see logs/package/lean/luci-app-unblockmusic/dump.txt for details
ERROR: please fix package/lean/luci-app-usb-printer/Makefile - see logs/package/lean/luci-app-usb-printer/dump.txt for details
ERROR: please fix package/lean/luci-app-uugamebooster/Makefile - see logs/package/lean/luci-app-uugamebooster/dump.txt for details
ERROR: please fix package/lean/luci-app-v2ray-server/Makefile - see logs/package/lean/luci-app-v2ray-server/dump.txt for details
ERROR: please fix package/lean/luci-app-verysync/Makefile - see logs/package/lean/luci-app-verysync/dump.txt for details
ERROR: please fix package/lean/luci-app-vlmcsd/Makefile - see logs/package/lean/luci-app-vlmcsd/dump.txt for details
ERROR: please fix package/lean/luci-app-vsftpd/Makefile - see logs/package/lean/luci-app-vsftpd/dump.txt for details
ERROR: please fix package/lean/luci-app-webadmin/Makefile - see logs/package/lean/luci-app-webadmin/dump.txt for details
ERROR: please fix package/lean/luci-app-wrtbwmon/Makefile - see logs/package/lean/luci-app-wrtbwmon/dump.txt for details
ERROR: please fix package/lean/luci-app-zerotier/Makefile - see logs/package/lean/luci-app-zerotier/dump.txt for details
ERROR: please fix package/lean/luci-lib-docker/Makefile - see logs/package/lean/luci-lib-docker/dump.txt for details
ERROR: please fix package/lean/luci-proto-bonding/Makefile - see logs/package/lean/luci-proto-bonding/dump.txt for details
ERROR: please fix package/lean/luci-theme-argon/Makefile - see logs/package/lean/luci-theme-argon/dump.txt for details
ERROR: please fix package/lean/luci-theme-netgear/Makefile - see logs/package/lean/luci-theme-netgear/dump.txt for details
ERROR: please fix package/lean/mt/luci-app-mtwifi/Makefile - see logs/package/lean/mt/luci-app-mtwifi/dump.txt for details

按照README.md文件做的了
确定是Ubuntu 20.04 LTS x64
到./scripts/feeds install -a就报以上错误了

LittleRabbit8843, 2021-07-21:
你这个属于正常现象

EnnawYang, 2021-07-21:
> 你这个属于正常现象

不，不正常，但也不是源码问题

starxu1971, 2021-07-22:
正常的多执行价几遍 ./scripts/feeds update -a && ./scripts/feeds install -a 就行了。

closed 2021-07-22 by 308066400