coolsnowwolf/lede #7474: 编译失败

jy3344511, 2021-07-22:
make[1]: Entering directory '/home/jy3344511/lede'
unset ERROR; \
for package in ; do \
	export MAKEFLAGS= ;make V=ss -s -r -C package/$package prereq || ERROR=1; \
done; \
if [ -n "$ERROR" ]; then \
	echo "Package prerequisite check failed."; \
	false; \
fi
touch tmp/.prereq_packages
make[2]: Entering directory '/home/jy3344511/lede/tools/autoconf'
make[2]: Entering directory '/home/jy3344511/lede/tools/bc'
make[2]: Entering directory '/home/jy3344511/lede/tools/automake'
make[2]: Entering directory '/home/jy3344511/lede/tools/autoconf-archive'
make[2]: Entering directory '/home/jy3344511/lede/tools/bison'
make[2]: Entering directory '/home/jy3344511/lede/tools/cmake'
make[2]: Entering directory '/home/jy3344511/lede/tools/cpio'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/autoconf'
time: tools/autoconf/download#0.04#0.01#0.11
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/bison'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/autoconf-archive'
time: tools/bison/download#0.04#0.02#0.11
time: tools/autoconf-archive/download#0.04#0.01#0.11
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/automake'
time: tools/automake/download#0.02#0.03#0.12
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/bc'
time: tools/bc/download#0.04#0.00#0.12
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/cmake'
time: tools/cmake/download#0.04#0.01#0.12
make[2]: Entering directory '/home/jy3344511/lede/tools/dosfstools'
make[2]: Entering directory '/home/jy3344511/lede/tools/e2fsprogs'
make[2]: Entering directory '/home/jy3344511/lede/tools/fakeroot'
make[2]: Entering directory '/home/jy3344511/lede/tools/findutils'
make[2]: Entering directory '/home/jy3344511/lede/tools/firmware-utils'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/cpio'
time: tools/cpio/download#0.03#0.01#0.14
make[2]: Entering directory '/home/jy3344511/lede/tools/flex'
make[2]: Entering directory '/home/jy3344511/lede/tools/gengetopt'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/dosfstools'
time: tools/dosfstools/download#0.03#0.01#0.11
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/e2fsprogs'
time: tools/e2fsprogs/download#0.04#0.00#0.11
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/fakeroot'
time: tools/fakeroot/download#0.03#0.01#0.12
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/findutils'
time: tools/findutils/download#0.04#0.01#0.12
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/flex'
time: tools/flex/download#0.04#0.00#0.11
make[2]: Entering directory '/home/jy3344511/lede/tools/libressl'
make[2]: Nothing to be done for 'download'.
make[2]: Leaving directory '/home/jy3344511/lede/tools/firmware-utils'
time: tools/firmware-utils/download#0.04#0.02#0.13
make[2]: Entering directory '/home/jy3344511/lede/tools/lzma'
make[2]: Entering directory '/home/jy3344511/lede/tools/libtool'
make[2]: Entering directory '/home/jy3344511/lede/tools/m4'
make[2]: Entering directory '/home/jy3344511/lede/tools/missing-macros'
make[2]: Entering directory '/home/jy3344511/lede/tools/make-ext4fs'
make[2]: Nothing to be done for 'download'.
[... 1083 lines cut ...]
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -O2 -I/home/jy3344511/lede/staging_dir/host/include  -Wno-unused-parameter -include endian.h -L/home/jy3344511/lede/staging_dir/host/lib  -o /home/jy3344511/lede/build_dir/host/firmware-utils/bin/addfwhdr src/addfwhdr.c src/cyg_crc32.c -Wall
src/addfwhdr.c:66:19: error: conflicting types for 'cyg_crc32_accumulate'
   66 | extern cyg_uint32 cyg_crc32_accumulate(cyg_uint32 crc32val, unsigned char *s, int len);
      |                   ^~~~~~~~~~~~~~~~~~~~
In file included from src/addfwhdr.c:15:
src/cyg_crc.h:88:1: note: previous declaration of 'cyg_crc32_accumulate' was here
   88 | cyg_crc32_accumulate(cyg_uint32 crc, void *s, int len);
      | ^~~~~~~~~~~~~~~~~~~~
src/addfwhdr.c: In function 'main':
src/addfwhdr.c:119:6: warning: variable 'garbage' set but not used [-Wunused-but-set-variable]
  119 |  int garbage = 0;
      |      ^~~~~~~
src/addfwhdr.c:221:2: warning: ignoring return value of 'write', declared with attribute warn_unused_result [-Wunused-result]
  221 |  write(fd_w, fwhdr, sizeof(struct cbt_fw_header));
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [Makefile:112: /home/jy3344511/lede/build_dir/host/firmware-utils/.built] Error 1
make[3]: Leaving directory '/home/jy3344511/lede/tools/firmware-utils'
time: tools/firmware-utils/compile#0.19#0.06#0.38
    ERROR: tools/firmware-utils failed to build.
make[2]: *** [tools/Makefile:159: tools/firmware-utils/compile] Error 1
make[2]: Leaving directory '/home/jy3344511/lede'
make[1]: *** [tools/Makefile:155: /home/jy3344511/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/jy3344511/lede'
make: *** [/home/jy3344511/lede/include/toplevel.mk:230：world] 错误 2

jarod360, 2021-07-22:
我也是这个编译错误的失败，看样子是firmware-utils

Neverends4, 2021-07-22:
+1
昨天还好好的~

hzjnet, 2021-07-22:
> make[1]: Entering directory '/home/jy3344511/lede'
> unset ERROR;
> for package in ; do
> export MAKEFLAGS= ;make V=ss -s -r -C package/$package prereq || ERROR=1;
> done;
> if [ -n "$ERROR" ]; then
> echo [...]

回退firmware-utils试试吧

rshi0212, 2021-07-22:
[image]

hahaher123, 2021-07-22:
+1，我也是这里出错了

EnnawYang, 2021-07-22:
不要+1了，没看到刚修了吗

closed 2021-07-22 by coolsnowwolf