coolsnowwolf/lede #7476: 第一次编译，出错了，希望有好心人告诉我错哪里了，谢谢

KevinxYep, 2021-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：第一次编译，在闲置的笔记本安装上了ubuntu系统，软路由用openclash开了全局代理，最后报错了，也没有在lede文件夹下找到bin这个文件夹，也就没看到自己的固件编成功没有

### (2) 路由器型号和固件版本

A：软路由J4105，固件版本不清楚，跟着操作指引一步步做的，应该就是最新固件

### (3) 详细日志

A：make[3]: *** [Makefile:112: /home/kevin/lede/build_dir/host/firmware-utils/.built] Error 1
make[3]: Leaving directory '/home/kevin/lede/tools/firmware-utils'
time: tools/firmware-utils/compile#0.16#0.02#0.18
    ERROR: tools/firmware-utils failed to build.
make[2]: *** [tools/Makefile:159: tools/firmware-utils/compile] Error 1
make[2]: Leaving directory '/home/kevin/lede'
make[1]: *** [tools/Makefile:155: /home/kevin/lede/staging_dir/host/stamp/.tools_compile_yyynyynnyyyynyyyyyynyynnyyynyyyyyyyyyyyyyyyyynynnyyyyyyyyyy] Error 2
make[1]: Leaving directory '/home/kevin/lede'
make: *** [/home/kevin/lede/include/toplevel.mk:230：world] 错误 2

不知道是不是提供这样的日志就可以了，纯小白，希望问题能够解决。

EnnawYang, 2021-07-22:
https://github.com/coolsnowwolf/lede/issues/7474

closed 2021-07-22 by coolsnowwolf

KevinxYep, 2021-07-22:
太小白了，请教一下回退firmware-utils应该怎么操作呀，7474他遇到的问题和我是一样的