coolsnowwolf/lede #7479: 怎样更改 make menuconfig中的默认设置？

qqsir-dev, 2021-07-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
比如我要默认设置是英文？
A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

tanst, 2021-07-22:
默认设置在这里：
package/lean/default-settings/files/zzz-default-settings

qqsir-dev, 2021-07-23:
谢谢，不过没有成功，uci set luci.main.lang=zh_cn 改为 en 编译后仍然是中文默认

Neverends4, 2021-07-23:
如果你编译时用了files文件夹带入自己的上机配置，Luci有设置语言的配置文件。

qqsir-dev, 2021-07-23:
> 如果你编译时用了files文件夹带入自己的上机配置，Luci有设置语言的配置文件。

兄台您这句话我没读明白，能否讲的再口语化一些...

Neverends4, 2021-07-23:
在你的lede文件夹下新建files文件夹，可以带入目前工作的路由器配置，譬如
`/lede/files/etc/config/luci`
就是luci的配置，里面有这一段
`config internal 'languages'` 
`option zh_cn '简体中文 (Simplified Chinese)'`	
`option en 'English'`	
但是有风险，如果内核升级，配置文件格式变了，带入配置后刷机会导致起不来，重置都没用，因为配置文件固化在固件里了。
network和dhcp之类的配置容易导致这个问题。

qqsir-dev, 2021-07-23:
多谢，一会试试这个办法，我是虚拟平台跑的软路由，不怕折腾。

qqsir-dev, 2021-07-23:
我在想生成菜单make menuconfig的时候好像也是个编译过程，那里面的默认选项应该是这个编译过程带入的吧，所以我结合1楼兄弟的设置，同时又去改了同文件夹下的Makefile文件，将里面的改为：
DEPENDS:=+luci-base +luci +@LUCI_LANG_en，
po2lmo ./po/zh-cn/default.po $(1)/usr/lib/lua/luci/i18n/default.en.lmo
上述两个地方我是猜着改的，en部分原来是zh_cn和zh-cn，这样保存后，我删除原有的菜单配置文件，重新make menuconfig，进去后发现默认语言选项已经是-*-English了，而中文变成了可选项<*>，这应该就是成功了吧？
我编译了一个出来，更新后发现默认还是中文，是不是因为我保留了配置更新的固件呢？还是哪里还需要改？对于我这个外行实在是费脑子。

281677160, 2021-07-23:
> 我在想生成菜单make menuconfig的时候好像也是个编译过程，那里面的默认选项应该是这个编译过程带入的吧，所以我结合1楼兄弟的设置，同时又去改了同文件夹下的Makefile文件，将里面的改为：
> DEPENDS:=+luci-base +luci +@LUCI_LANG_en，
> po2lmo ./po/zh-cn/default.po [...]

源码编译出来就自带英文跟中文两种语言了啊，为什么还要去修改这些东西？直接openwrt后台系统-->系统-->语言和界面，把语言修改成auto的就是英文了啊

qqsir-dev, 2021-07-23:
还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面

281677160, 2021-07-23:
> 还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面

没看懂你说什么，改成auto后还是不显示英文？
还是可以显示英文，不保存配置安装后就变成中文了？如果是这样的话，编译之前
你到package/lean/default-settings/files/zzz-default-settings文件里面把默认中文改成luci.main.lang=zh_cn改成luci.main.lang=auto不就得了？

qqsir-dev, 2021-07-23:
> > 还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面
> 
> 没看懂你说什么，改成auto后还是不显示英文？
> 还是可以显示英文，不保存配置安装后就变成中文了？如果是这样的话，编译之前
> 你到package/lean/default-settings/files/zzz-default-settings文件里面把默认中文改成luci.main.lang=zh_cn改成luci.main.lang=auto不就得了？

是吗？我试试，谢谢

qqsir-dev, 2021-07-23:
> > 还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面
> 
> 没看懂你说什么，改成auto后还是不显示英文？
> 还是可以显示英文，不保存配置安装后就变成中文了？如果是这样的话，编译之前
> 你到package/lean/default-settings/files/zzz-default-settings文件里面把默认中文改成luci.main.lang=zh_cn改成luci.main.lang=auto不就得了？

顺便问一下高手，怎么更改默认IP，就是编译出来就是我想要的IP，省得改了，谢谢

281677160, 2021-07-23:
> > > 还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面
> > 
> > 
> > 没看懂你说什么，改成auto后还是不显示英文？
> > 还是可以显示英文，不保存配置安装后就变成中文了？如果是这样的话，编译之前
> > 你到package/lean/default-settings/files/zzz-default-settings文件里面把默认中文改成luci.main.lang=zh_cn改成luci.main.lang=auto不就得了？
> 
> 顺便问一下高手，怎么更改默认IP，就是编译出来就是我想要的IP，省得改了，谢谢

我就一小白好吧，你是云编译的，还是本地编译的？

EnnawYang, 2021-07-23:
> 怎么更改默认IP
> 
https://github.com/coolsnowwolf/lede/blob/master/package/base-files/files/bin/config_generate#L150

qqsir-dev, 2021-07-23:
> > > > 还是失败，只是看到默认英文，但实际上不保留配置刷机启动仍然是中文界面
> > > 
> > > 
> > > 没看懂你说什么，改成auto后还是不显示英文？
> > > 还是可以显示英文，不保存配置安装后就变成中文了？如果是这样的话，编译之前
> > > 你到package/lean/default-settings/files/zzz-default-settings文件里面把默认中文改成luci.main.lang=zh_cn改成luci.main.lang=auto不就得了？
> > 
> > 
> > 顺便问一下高手，怎么更改默认IP，就是编译出来就是我想要的IP，省得改了，谢谢
> 
> 我就一小白好吧，你是云编译的，还是本地编译的？

我更白，我本地编译

qqsir-dev, 2021-07-23:
> > 怎么更改默认IP
> 
> https://github.com/coolsnowwolf/lede/blob/master/package/base-files/files/bin/config_generate#L150

这是高手，这是高手.....多谢，我去试试

281677160, 2021-07-23:
uci set network.lan.ipaddr='192.168.2.2'                                    # IPv4 地址(openwrt后台地址)
uci set network.lan.netmask='255.255.255.0'                                 # IPv4 子网掩码
uci set network.lan.gateway='192.168.2.1'                                   # IPv4 网关
uci set network.lan.broadcast='192.168.2.255'                               # IPv4 广播
uci set network.lan.dns='114.114.114.114 223.5.5.5'                         # DNS(多个DNS要用空格分开)
uci set network.lan.delegate='0'                                            # 去掉LAN口使用内置的 IPv6 管理
uci commit network                                                          # 不要删除跟注释,除非上面全部删除或注释掉了

uci set dhcp.lan.ignore='1'                                                 # 关闭DHCP功能
uci commit dhcp                                                             # 跟‘关闭DHCP功能’联动,同时用
把这些放到zzz-default-setting文件里面就可以了，那些IP，DNS什么的，按需求改，如果有你不要就别放zzz-default-setting文件就得了，还有就是把我那些注释删除了，用这个方法改IP有好处也有坏处，坏处就是如果你安装固件后，再去修改这些数值，下次保存配置升级都会还原成你编译的设置，其实zzz-default-setting文件就是你安装好固件后执行一次

qqsir-dev, 2021-07-23:
> > > 怎么更改默认IP
> > 
> > 
> > https://github.com/coolsnowwolf/lede/blob/master/package/base-files/files/bin/config_generate#L150
> 
> 这是高手，这是高手.....多谢，我去试试

> > 怎么更改默认IP
> 
> https://github.com/coolsnowwolf/lede/blob/master/package/base-files/files/bin/config_generate#L150

成功了，多谢！！！！

qqsir-dev, 2021-07-23:
> uci set network.lan.ipaddr='192.168.2.2' # IPv4 地址(openwrt后台地址)
> uci set network.lan.netmask='255.255.255.0' # IPv4 子网掩码
> uci set network.lan.gateway='192.168.2.1' # IPv4 网关
> uci set [...]

这个好NB，收藏了！！！多谢多谢！
另外语言那个还是不行，auto默认，不保留配置刷完我就想是英文界面，可无论怎么都是中文

qqsir-dev, 2021-07-23:
感谢二位！！
我把二位的合并了一下，在本地的package/base-files/files/bin/config_generate文件中，按照281677160的建议添加了

                        netg=${gateway:-"192.168.*.*"}
                        netdns=${dns:-"114.114.114.114 119.29.29.29 223.5.5.5"}

                        uci -q batch <<-EOF
                                set network.$1.proto='static'
                                set network.$1.ipaddr='$ipad'
                                set network.$1.netmask='$netm'
                                set network.$1.gateway='$netg'
                                set network.$1.dns='$netdns'
                                set dhcp.lan.ignore='1'
                                commit dhcp
然后完美的设置了IP，网关，DNS 和关闭了DHCP 功能！！
特此感谢二位！下面就得研究怎么能默认英文显示了！

281677160, 2021-07-23:
> 这个好NB，收藏了！！！多谢多谢！
> 另外语言那个还是不行，auto默认，不保留配置刷完我就想是英文界面，可无论怎么都是中文

我刚测试了下，修改zzz-default-setting没效果，不知道什么原因，不过你可以弄一个开机启动来搞他，在package/base-files/files/etc/rc.local文件里面加一个命令就行了，加在exit 0上面 sed -i "s/'zh_cn'/'auto'/g" /etc/config/luci

qqsir-dev, 2021-07-24:
语言这个有个特点，和你昨天说的性质一样，就是每次刷机更新固件，无论保存不保存配置，都恢复成默认设置，也就是中文，这个是不是也写在了那个地方，只要去掉就应该行了，我再慢慢找找，不过先试试你这个，谢谢！

qqsir-dev, 2021-07-24:
我把package/lean/整个default-settings目录删除了，编译也没出错，但还是默认中文，看来默认设置不在这。

qqsir-dev, 2021-07-25:
> 我刚测试了下，修改zzz-default-setting没效果，不知道什么原因，不过你可以弄一个开机启动来搞他，在package/base-files/files/etc/rc.local文件里面加一个命令就行了，加在exit 0上面 sed -i "s/'zh_cn'/'auto'/g" /etc/config/luci

还是你这个最靠谱！！大赞！！多谢！

closed 2021-07-25 by qqsir-dev