coolsnowwolf/lede #7485: 请教以一个解析DDNS域名的问题

smallprogram, 2021-07-23:
我有一个动态公网IP的机器，通过某个运营商实现了DDNS解析，用域名映射到了这个动态公网IP的机器上。
例如：
例如域名 abc.com某一时刻映射为1.2.3.4，在外部网络使用openwrt路由进行上网，浏览abc.com可以正常访问这台机器。
但是当某个时刻abc.com映射变为1.2.3.5，通过各种在线网络工具、DNS在线解析工具都能看到abc.com映射关系已经变为1.2.3.5。
但是使用openwrt上网的机器访问abc.com时还是访问到1.2.3.4。必须要等上几个小时的时间才能正常通过abc.com访问到1.2.3.5。有没有什么方法让这种域名解析快速生效？

我曾尝试过重置PDNSD、重置本地DNS缓存，都无效。

Neverends4, 2021-07-23:
DNS Propagation的速度是你的DNS解析服务商决定的，服务商已经更新了就没问题了
这说明是你op指向的DNS没有更新，默认op是用的本地ISP的DNS，你可以试试改一下，譬如如果你确定阿里或者腾讯的公共DNS能在机器IP更新后快速更新IP，换op的DNS就是。

smallprogram, 2021-07-23:
> DNS Propagation的速度是你的DNS解析服务商决定的，服务商已经更新了就没问题了
> 这说明是你op指向的DNS没有更新，默认op是用的本地ISP的DNS，你可以试试改一下，譬如如果你确定阿里或者腾讯的公共DNS能在机器IP更新后快速更新IP，换op的DNS就是。

DNS解析服务商的DNS Propagation的速度是非常快的，通常在5到10分钟内基本上国内各大DNS服务器都已经更新了，这点我通过在线DNS解析工具都已经验证过。OP的DNS采用Turbo ACC中DNS加速设置的DNS服务器获取DNS解析结果并缓存。曾尝试过更换DNS服务器地址、清空DNS缓存等，均无效。

Neverends4, 2021-07-23:
dig your.domain 看看你域名的ttl是多少，例如
www.baidu.com.          47      IN      CNAME   www.a.shifen.com
说明百度的ttl是47s
最好的法子是降低ddns解析你域名的ttl，但是看你10分钟都解析成功了这点没问题
所以降低pdnsd的ttl，去搜一下pdnsd的man page，
再…………重启pdnsd应该会清除cache，不会有这个问题呀？不过我一般都用默认的dnsmasq，不清楚pdnsd的情况。

再再……我也用ddns，而且我那个解析服务商慢的离谱，最短ttl只能设成3600，怎么都要40分钟才能更新，所以我都辅助在客户端用脚本发邮件，一旦ip变了直接用ip连接………………

smallprogram, 2021-07-23:
也只能这样了。。。。感谢你的帮助与回复。谢谢~

closed 2021-07-23 by smallprogram