coolsnowwolf/lede #7486: 昨天的samba4更新后至4.14.6后编译失败，退回原4.13.9后通过了

codesnas, 2021-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：更新：https://github.com/coolsnowwolf/packages/commit/d53c2ec7906dfc43a612ea9724736076ab74a058

### (2) 路由器型号和固件版本

A：QEMU ARM Virtual Machine [ QEMU ARMv8 Virtual Machine(cortex-a53) ]

### (3) 详细日志

A：Checking for system installation of Python module markdown                        : not found 
        Unable to find Python module 'markdown'. Please install the system package: python3-markdown'.

make[3]: *** [Makefile:461: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/samba-4.14.6/.configured_1ef03153caf3dd6ed1c74f6a4c3d1c78] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/compile#25.05#13.52#32.12
    ERROR: package/feeds/packages/samba4 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/samba4/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

昨天的samba4更新后至4.14.6后编译失败，日志如上
退回原4.13.9后通过了。
请大佬确认下是不是新版有什么问题。

codesnas, 2021-07-23:
```
lede/feeds/packages/net/samba4$ cat Makefile 

                https://download.samba.org/pub/samba/stable/
PKG_HASH:=b97a773ed3b4dae6d5ebd3e09337c897ae898b65f38abad550f852b594d4e07f

PKG_MAINTAINER:=Andy Walsh <andy.walsh44+github@gmail.com>
PKG_LICENSE:=GPL-3.0-only
PKG_LICENSE_FILES:=COPYING
PKG_CPE_ID:=cpe:/a:samba:samba

# samba4=(asn1_compile,compile_et) rpcsvc-proto=(rpcgen)
HOST_BUILD_DEPENDS:=python3/host rpcsvc-proto/host perl/host perl-parse-yapp/host
PKG_BUILD_DEPENDS:=samba4/host libtasn1/host
```

是不是这个文件需要更新下？关系到python3

codesnas, 2021-07-23:
package/feeds/packages/samba4/Makefile
```
PKG_VERSION:=4.13.9
PKG_HASH:=b97a773ed3b4dae6d5ebd3e09337c897ae898b65f38abad550f852b594d4e07f
```

倒回去可以编译通过

Boos4721, 2021-07-23:
[自行查看](https://github.com/Boos4721/packages/commit/e86c2be5aabb57b4306c2ec3b38be3294d9a4532)

closed 2021-07-29 by codesnas

zhangguanzhang, 2021-08-29:
@Boos4721 回退到4.15.5还是报错

Boos4721, 2021-08-29:
> @Boos4721 回退到4.15.5还是报错

？

zhangguanzhang, 2021-08-29:
> > @Boos4721 回退到4.15.5还是报错
> 
> ？

从本条issue前到后看，samba4编译报错

Boos4721, 2021-08-29:
> > > @Boos4721 回退到4.15.5还是报错
> > 
> > 
> > ？
> 
> 从本条issue前到后看，samba4编译报错

然后呢

zhangguanzhang, 2021-08-29:
> > > > @Boos4721 回退到4.15.5还是报错
> > > 
> > > 
> > > ？
> > 
> > 
> > 从本条issue前到后看，samba4编译报错
> 
> 然后呢

我回退到你link的这个pr的4.14.5的也编译报错

Boos4721, 2021-08-29:
> > > > > @Boos4721 回退到4.15.5还是报错
> > > > 
> > > > 
> > > > ？
> > > 
> > > 
> > > 从本条issue前到后看，samba4编译报错
> > 
> > 
> > 然后呢
> 
> 我从最新源码的 `4.14.6` 回退到你 link 的这个 pr 里的的 `4.14.5` 的也编译报错

我什么时候pr了

zhangguanzhang, 2021-08-29:
> [自行查看](https://github.com/Boos4721/packages/commit/e86c2be5aabb57b4306c2ec3b38be3294d9a4532)

这个里

Boos4721, 2021-08-29:
> > [自行查看](https://github.com/Boos4721/packages/commit/e86c2be5aabb57b4306c2ec3b38be3294d9a4532)
> 
> 这个里

你真的知道编译samba4需要什么么

zhangguanzhang, 2021-08-29:
> > > [自行查看](https://github.com/Boos4721/packages/commit/e86c2be5aabb57b4306c2ec3b38be3294d9a4532)
> > 
> > 
> > 这个里
> 
> 你真的知道编译samba4需要什么么

我安装了python3-markdown，Python2的markdown也安装了，python3 -m markdown --help也有输出

zhangguanzhang, 2021-08-30:
@Boos4721 我同样的步骤回退了后可以编译了
```
if grep -Eq '^PKG_VERSION:=4.14.6' package/feeds/packages/samba4/Makefile; then
    sed -ri -e '/^PKG_VERSION:=/s#4.14.6#4.13.9#' \
        -e '/^PKG_HASH:=/s#:=.+#:=b97a773ed3b4dae6d5ebd3e09337c897ae898b65f38abad550f852b594d4e07f#' \
    package/feeds/packages/samba4/Makefile 
fi
```