coolsnowwolf/lede #7494: 请求恢复mt7621默认超频1000MHz

Luastonsu, 2021-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
请求恢复MT7621 CPU的默认超频1000MHz，breed在很早之前的版本中就取消了超频选项
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
Xiaomi Redmi AC2100
A：

### (3) 详细日志

A：

AmadeusGhost, 2021-07-23:
然后就有人启不动了…

Lrvineshaw, 2021-07-23:
夏天还是保守点，我一般天冷才超频

Luastonsu, 2021-07-23:
> 然后就有人启不动了…

那大佬能不能指导下5.4内核如何超频，现在使用4.14内核总是编译失败

Luastonsu, 2021-07-23:
> 夏天还是保守点，我一般天冷才超频

就是往冒烟去的，不怕

itfish92, 2021-07-23:
传家宝k2p的breed超频一直在啊，看了下最新的是2020年的breed

hzjixiang, 2021-07-23:
体质不同 超频很多不稳定的现象，甚至不能开机！！！如果觉得性能不够建议换ARM的路由

Luastonsu, 2021-07-24:
> 传家宝k2p的breed超频一直在啊，看了下最新的是2020年的breed

也是mt7621？可能我的breed版本问题

Luastonsu, 2021-07-24:
> 传家宝k2p的breed超频一直在啊，看了下最新的是2020年的breed

刚更新了breed，还是没有。查询k2p等型号有，小米r3g没有
现在只能从固件编译方面想办法了，毕竟默认880运行开开插件后连luci界面都刷不出来

yinzhefuyun, 2021-07-28:
联发科去breed超频

Luastonsu, 2021-08-03:
> 联发科去breed超频

部分mtk路由breed没有超频选项，比如小米r3g

LeviMarvin, 2022-03-23:
> > 联发科去breed超频
> 
> 部分mtk路由breed没有超频选项，比如小米r3g

是的，我的R3G上Breed一看也没有超频选项，请问除了编译OpenWRT外还有没有办法超频呀

lovehackintosh, 2022-07-17:
全新克隆一下源码,下载https://raw.githubusercontent.com/coolsnowwolf/lede/5be81314bb08934fbe03c2e26191b80f021b0d2b/target/linux/ramips/patches-5.4/102-mt7621-fix-cpu-clk-add-clkdev.patch这个文件替换到lede/target/linux/ramips/patches-5.4/ ，然后正常编译

closed 2022-09-06 by aiamadeus