coolsnowwolf/lede #7500: 小米AC2100 网络没有交换机

313913743, 2021-07-23:
1.小米AC2100 网络没有交换机

## 2. 详细叙述

### (1) 具体问题

A：最新版本编译，没有交换机功能，kom-macvlan是默认勾选的状态，挺需要这个功能的不知道怎么打开，找不到相关解决办法

### (2) 路由器型号和固件版本

A：小米AC2100

313913743, 2021-07-24:
lede/target/linux/ramips/mt7621/base-files/etc/board.d/02_network

找到问题出在这个文件上，文件对比旧版本有明显配置不同，但是本人小白刚刚入门，仅仅想编译一个自己能用的稳定系统，这个问题过于复杂了，我把旧版本文件中的配置直接复制到新版源码中，再编译发现确实有交换机了，但是并不能用，所有接口虽然有数据传输，但是没用，应该是内核版本有大变化的原因导致的，现在我只能找旧版本的源码库使用，希望大佬日后提供技术支持，感谢大佬的源码

Neverends4, 2021-07-24:
官方op新内核MT7621用了DSA架构，没有交换机了，然后这里的Luci还是老版本，界面没有改，所以暂时没有vlan
再话说官方op，vlan也不好用，N多人在抱怨不知道怎么整，我也没看懂……
如果你对vlan有刚性需求，建议reset到大更新前的源码编译。

313913743, 2021-07-24:
是的是的，我已经回到老版本了

dukmonki, 2021-07-26:
如何回到大更新前的版本？用什么命令？

313913743, 2021-07-26:
> 如何回到大更新前的版本？用什么命令？

我不是用命令，是找到目标文件更新之前的那个历史版本，把库打包到本地编译，云编译我试过直接改分支号不成功，不知道怎么回事，分支号搜不到那个版本，但又确实存在，可能是管理问题吧

bonnyandsky, 2021-07-27:
> 是的是的，我已经回到老版本了

请问老版本的编号多少？我也遇到同样问题

PR #7560 (2021-07-30): ramips: mt7621: add back switch support
fix: #7500 , fix: #7508 , fix: #7517 , fix: #7537
closed 2021-07-30 by coolsnowwolf