coolsnowwolf/lede #7504: 请教Lean大神如何更改固件的默认显示语言

qqsir-dev, 2021-07-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
我想刷机或更新后默认语言是英文，请问改哪里？我是本地完全按照网站指导编译。谢谢
A：

### (2) 路由器型号和固件版本
ESXi 7 软路由
A：

### (3) 详细日志

A：

yeheigcy, 2021-07-24:
https://github.com/coolsnowwolf/lede/blob/c5a034330471099cb73c5f1745b13dbd0764998e/package/lean/default-settings/files/zzz-default-settings#L3

qqsir-dev, 2021-07-24:
> https://github.com/coolsnowwolf/lede/blob/c5a034330471099cb73c5f1745b13dbd0764998e/package/lean/default-settings/files/zzz-default-settings#L3

这个改过设置了，uci set luci.main.lang=en，或者auto 都不行，改过后删除配置文件，重新make menuconfig去编译的，仍然不行，还是默认中文。

Neverends4, 2021-07-24:
所以还是我在另一个issue里说的，用files带入Luci的上机现成设置是最简单的法子，肯定不如改源码彻底，但是一样能达到。

qqsir-dev, 2021-07-24:
我刚刚重新编译没有选中文，只有英文，编译完刷好固件，登录路由器发现里面的配置默认还是中文，只不过由于没有中文包不能显示，但默认配置那里居然还是zh_cn，这是那个程序默认写进去的？？？

lphgor, 2021-07-24:
这里设置了默认值zh-cn
https://github.com/coolsnowwolf/luci/blob/aa12cbf9240365801d7f8a089e0f7284c79cef02/luci.mk#L117-L119

qqsir-dev, 2021-07-24:
我本地没这个目录啊，我找找

lphgor, 2021-07-24:
> 我本地没这个目录啊，我找找

本地执行完`./scripts/feeds update -a && ./scripts/feeds install -a`就有了，这个文件`feeds/luci/luci.mk`

qqsir-dev, 2021-07-24:
好的，我刚刚编辑了zzz-default-settings 
uci set luci.main.lang=en
而且还make clean重新编译了，刷完固件后进入路由器发现是这样的

config core 'main'
        option resourcebase '/luci-static/resources'
        option mediaurlbase '/luci-static/argon'
        option lang 'zh_cn'

config extern 'flash_keep'
        option uci '/etc/config/'
        option dropbear '/etc/dropbear/'
        option openvpn '/etc/openvpn/'
        option passwd '/etc/passwd'
        option opkg '/etc/opkg.conf'
        option firewall '/etc/firewall.user'
        option uploads '/lib/uci/upload/'

config internal 'languages'
        option zh_cn '............ (Simplified Chinese)'
        option en 'English'
        option main 'en'

zzz文件里那个是不是不应该那么写啊？好像生效的地方不对啊

qqsir-dev, 2021-07-24:
> > 我本地没这个目录啊，我找找
> 
> 本地执行完`./scripts/feeds update -a && ./scripts/feeds install -a`就有了，这个文件`feeds/luci/luci.mk`
> 
> openwrt官方版是没有第3行的
> https://github.com/openwrt/luci/blob/d354f0b107c0d54e5f9d2ec39c7f600909807e94/luci.mk#L265-L266

我看到了，不过文件里应该是按照配置文件生成的，已经是en了哦

qqsir-dev, 2021-07-24:
我总算找到了，但也不是最底层的，本地执行完./scripts/feeds update -a && ./scripts/feeds install -a就有了
feeds/luci/modules/luci-base/root/etc/uci-defaults/luci-base，怎么生成的还有待研究，编辑文件改里面的zh-cn为en，重新生成固件就是默认就是英文了。
set luci.main.lang=en
感谢所有出主意给思路的人！！

closed 2021-07-25 by qqsir-dev