coolsnowwolf/lede #7506: RT4230W 512MB刷写固件后启动失败

a920025608, 2021-07-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [√ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：关键log位置
[   46.313245] wlan1: Created a NSS virtual interface.
[   46.318698] wlan1: RX-Q and tasklet initialized.
[   46.327591] kmodloader: done loading kernel modules from /etc/modules.d/*
[   78.996154] 8<--- cut here ---
[   78.996194] Unable to handle kernel NULL pointer dereference at virtual address 00000000
猜测可能是固件默认分区布局只适合与256MB版本，但手头没有256MB版rac2v1k，无法求证

### (2) 路由器型号和固件版本

A：RT4230W-D187A Rev:9

### (3) 详细日志

A：
UBI: physical eraseblock size:   131072 bytes (128 KiB)
UBI: logical eraseblock size:    126976 bytes
UBI: smallest flash I/O unit:    2048
UBI: VID header offset:          2048 (aligned 2048)
UBI: data offset:                4096
UBI: attached mtd1 to ubi0
UBI: MTD device name:            "mtd=0"
UBI: MTD device size:            416 MiB
UBI: number of good PEBs:        3328
UBI: number of bad PEBs:         0
UBI: max. allowed volumes:       128
UBI: wear-leveling threshold:    4096
UBI: number of internal volumes: 1
UBI: number of user volumes:     3
UBI: available PEBs:             47
UBI: total number of reserved PEBs: 3281
UBI: number of PEBs reserved for bad PEB handling: 33
UBI: max/mean erase counter: 26/5
Read 7208960 bytes from volume kernel to 44000000
   Loading Kernel Image ... OK
OK

device nand0 <nand0>, # parts = 1
 #: name                size            offset          mask_flags
 0: firmware            0x1a000000      0x02400000      0

active partition: nand0,0 - (firmware) 0x1a000000 @ 0x02400000

defaults:
mtdids  : none
mtdparts: none
Setting up atags for msm partition: firmware
Using machid 0x177d from environment

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (runner@fv-az116-417) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3561-737e0fd86)) #0 SMP Fri Jul 23 13:09:30 2021
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[... 484 lines cut ...]
[   79.515304] Exception stack(0xef1f3fb0 to 0xef1f3ff8)
[   79.522352] 3fa0:                                     00000000 00000000 00000000 00000000
[   79.527498] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   79.535649] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   79.543808] Code: eb09384f e5d43da4 e3130002 1a00002a (e5da3000) 
[   79.550350] ---[ end trace d382b88acb79c820 ]---
[   79.556451] Kernel panic - not syncing: Fatal exception
[   79.561150] CPU0: stopping
[   79.566090] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G      D  C        5.4.132 #0
[   79.568860] Hardware name: Generic DT based system
[   79.576437] [<c030f954>] (unwind_backtrace) from [<c030b968>] (show_stack+0x14/0x20)
[   79.581047] [<c030b968>] (show_stack) from [<c0900b98>] (dump_stack+0x94/0xa8)
[   79.588932] [<c0900b98>] (dump_stack) from [<c030eb80>] (handle_IPI+0x184/0x1b8)
[   79.595971] [<c030eb80>] (handle_IPI) from [<c05dd588>] (gic_handle_irq+0xb4/0xb8)
[   79.603516] [<c05dd588>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[   79.610876] Exception stack(0xc0c01ee0 to 0xc0c01f28)
[   79.618466] 1ee0: 00000000 00000012 2ebcf000 ef710a80 ee8e0000 00000000 ef70fe30 00000012
[   79.623505] 1f00: 00000012 00000000 86369000 86360360 00000015 c0c01f30 c0731050 c0731054
[   79.631632] 1f20: 80000013 ffffffff
[   79.639800] [<c0301a8c>] (__irq_svc) from [<c0731054>] (cpuidle_enter_state+0x94/0x498)
[   79.643099] [<c0731054>] (cpuidle_enter_state) from [<c073149c>] (cpuidle_enter+0x30/0x4c)
[   79.651083] [<c073149c>] (cpuidle_enter) from [<c034ad6c>] (do_idle+0x1d8/0x240)
[   79.659412] [<c034ad6c>] (do_idle) from [<c034b07c>] (cpu_startup_entry+0x1c/0x20)
[   79.666964] [<c034b07c>] (cpu_startup_entry) from [<c0b00e5c>] (start_kernel+0x4dc/0x4ec)
[   80.674617] Rebooting in 3 seconds..

AmadeusGhost, 2021-07-24:
```
[ 24.314476] UBIFS error (ubi0:2 pid 151): ubifs_iget: failed to read inode 276, error -2
[ 24.314511] UBIFS error (ubi0:2 pid 151): ubifs_lookup: dead directory entry 'index.htm', error -2
[ 24.321690] UBIFS warning (ubi0:2 pid 151): ubifs_ro_mode.part.1: switched to read-only mode, error -2
[ 24.330510] CPU: 0 PID: 151 Comm: mount_root Tainted: G C 5.4.132 #0
[ 24.339676] Hardware name: Generic DT based system
[ 24.347239] [] (unwind_backtrace) from [] (show_stack+0x14/0x20)
[ 24.352104] [] (show_stack) from [] (dump_stack+0x94/0xa8)
[ 24.359997] [] (dump_stack) from [] (ubifs_lookup+0x26c/0x304)
[ 24.367029] [] (ubifs_lookup) from [] (__lookup_slow+0x88/0x14c)
[ 24.374578] [] (__lookup_slow) from [] (lookup_slow+0x3c/0x60)
[ 24.382475] [] (lookup_slow) from [] (walk_component+0x1bc/0x358)
[ 24.389855] [] (walk_component) from [] (path_lookupat+0x60/0x1e0)
[ 24.397755] [] (path_lookupat) from [] (filename_lookup.part.24+0x78/0xf0)
[ 24.405575] [] (filename_lookup.part.24) from [] (vfs_statx+0x68/0xd0)
[ 24.414163] [] (vfs_statx) from [] (__do_sys_fstatat64+0x28/0x4c)
[ 24.422409] [] (__do_sys_fstatat64) from [] (ret_fast_syscall+0x0/0x54)
[ 24.430303] Exception stack(0xedc35fa8 to 0xedc35ff0)
[ 24.438467] 5fa0: bef410d8 b6f34200 00000004 b6f342bb bef41000 00000100
[ 24.443683] 5fc0: bef410d8 b6f34200 bef41000 00000147 bef410d8 b6eb80ec b6ea75e0 00000048
[ 24.451835] 5fe0: b6f342bb bef40fe8 00000004 b6efb188
[ 24.467491] UBIFS error (ubi0:2 pid 151): ubifs_iget: failed to read inode 429, error -2
[ 24.467521] UBIFS error (ubi0:2 pid 151): ubifs_lookup: dead directory entry '15-automount', error -2
[ 24.510704] UBIFS error (ubi0:2 pid 151): ubifs_iget: failed to read inode 322, error -2
[ 24.510736] UBIFS error (ubi0:2 pid 151): ubifs_lookup: dead directory entry 'out_header_login.htm', error -2
[ 24.528844] mount_root: switching to ubifs overlay
[ 24.529743] overlayfs: upper fs is r/o, try multi-lower layers mount
[ 24.532559] mount_root: mount failed: Invalid argument, options lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work
[ 24.539114] mount_root: switching to ubifs failed - fallback to ramoverlay
```
挂载 nand 的时候 GG 了

AmadeusGhost, 2021-07-24:
试试这样改下再编译能不能解决：
```
--- a/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-rt4230w-rev6.dts
+++ b/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-rt4230w-rev6.dts
@@ -112,6 +112,9 @@
                nand-bus-width = <8>;
                nand-ecc-step-size = <512>;
 
+               nand-is-boot-medium;
+               qcom,boot_pages_size = <0x1180000>;
+
                partitions {
                        compatible = "fixed-partitions";
                        #address-cells = <1>;
```

a920025608, 2021-07-24:
> 试试这样改下再编译能不能解决：
>
> ```
> --- a/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-rt4230w-rev6.dts
> +++ b/target/linux/ipq806x/files/arch/arm/boot/dts/qcom-ipq8065-rt4230w-rev6.dts
> @@ -112,6 [...]

可能会让您有点生气，我没有尝试过打patch，请问如何打相对应的patch呢。。。

EnnawYang, 2021-07-24:
> 请问如何打相对应的patch呢。。。
> 
在lede根目录创建个 `.patch` 文件，然后把上面的内容粘贴进去  
然后 `git apply` 刚刚的patch文件，然后编译

Micah365, 2021-07-24:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [√ ] 没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> A：关键log位置
> [ 46.313245] wlan1: Created a NSS virtual interface.
> [ 46.318698] wlan1: RX-Q and tasklet initialized.
> [ 46.327591] kmodloader: done loading kernel modules from /etc/modules.d/*
> [ 78.996154] 8<--- cut here ---
> [ 78.996194] Unable to handle kernel NULL pointer dereference at virtual address 00000000
> 猜测可能是固件默认分区布局只适合与256MB版本，但手头没有256MB版rac2v1k，无法求证
> 
> ### (2) 路由器型号和固件版本
> A：RT4230W-D187A Rev:9
> 
> ### (3) 详细日志
> A：
> UBI: physical eraseblock size: 131072 bytes (128 KiB)
> UBI: logical eraseblock size: 126976 bytes
> UBI: smallest flash I/O unit: 2048
> UBI: VID header offset: 2048 (aligned 2048)
> UBI: data offset: 4096
> UBI: attached mtd1 to ubi0
> UBI: MTD device name: "mtd=0"
> UBI: MTD device size: 416 MiB
> UBI: number of good PEBs: 3328
> UBI: number of bad PEBs: 0
> UBI: max. allowed volumes: 128
> UBI: wear-leveling threshold: 4096
> UBI: number of internal volumes: 1
> UBI: number of user volumes: 3
> UBI: available PEBs: 47
> UBI: total number of reserved PEBs: 3281
> UBI: number of PEBs reserved for bad PEB handling: 33
> UBI: max/mean erase counter: 26/5
> Read 7208960 bytes from volume kernel to 44000000
> Loading Kernel Image ... OK
> OK
> 
> device nand0 , # parts = 1
> #: name size offset mask_flags
> 0: firmware 0x1a000000 0x02400000 0
> 
> active partition: nand0,0 - (firmware) 0x1a000000 @ 0x02400000
> 
> defaults:
> mtdids : none
> mtdparts: none
> Setting up atags for msm partition: firmware
> Using machid 0x177d from environment
> 
> Starting kernel ...
> 
> [ 0.000000] Booting Linux on physical CPU 0x0
> [ 0.000000] Linux version 5.4.132 (runner@fv-az116-417) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3561-737e0fd86)) #0 SMP Fri Jul 23 13:09:30 2021
> [ 0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
> [ 0.000000] CPU: div instructions available: patching division code
> [ 0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
> [ 0.000000] OF: fdt: Machine model: Askey RT4230W REV6
> [ 0.000000] Memory policy: Data cache writealloc
> [ 0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
> [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 252224
> [ 0.000000] Kernel command line: rootfstype=squashfs noinitrd
> [ 0.000000] Bootloader command line not present
> [ 0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[... 482 lines cut ...]
> [ 79.527498] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
> [ 79.535649] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
> [ 79.543808] Code: eb09384f e5d43da4 e3130002 1a00002a (e5da3000)
> [ 79.550350] ---[ end trace d382b88acb79c820 ]---
> [ 79.556451] Kernel panic - not syncing: Fatal exception
> [ 79.561150] CPU0: stopping
> [ 79.566090] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G D C 5.4.132 #0
> [ 79.568860] Hardware name: Generic DT based system
> [ 79.576437] [] (unwind_backtrace) from [] (show_stack+0x14/0x20)
> [ 79.581047] [] (show_stack) from [] (dump_stack+0x94/0xa8)
> [ 79.588932] [] (dump_stack) from [] (handle_IPI+0x184/0x1b8)
> [ 79.595971] [] (handle_IPI) from [] (gic_handle_irq+0xb4/0xb8)
> [ 79.603516] [] (gic_handle_irq) from [] (__irq_svc+0x6c/0x90)
> [ 79.610876] Exception stack(0xc0c01ee0 to 0xc0c01f28)
> [ 79.618466] 1ee0: 00000000 00000012 2ebcf000 ef710a80 ee8e0000 00000000 ef70fe30 00000012
> [ 79.623505] 1f00: 00000012 00000000 86369000 86360360 00000015 c0c01f30 c0731050 c0731054
> [ 79.631632] 1f20: 80000013 ffffffff
> [ 79.639800] [] (__irq_svc) from [] (cpuidle_enter_state+0x94/0x498)
> [ 79.643099] [] (cpuidle_enter_state) from [] (cpuidle_enter+0x30/0x4c)
> [ 79.651083] [] (cpuidle_enter) from [] (do_idle+0x1d8/0x240)
> [ 79.659412] [] (do_idle) from [] (cpu_startup_entry+0x1c/0x20)
> [ 79.666964] [] (cpu_startup_entry) from [] (start_kernel+0x4dc/0x4ec)
> [ 80.674617] Rebooting in 3 seconds..

请教一下，make menuconfig时选哪个选项，怎么加载上的ath10k_pci驱动，我编译的时候选了firmware和kernel里面的ath10，可是我的QCA9887网卡就是认不到

a920025608, 2021-07-25:
修好了，能启动

a920025608, 2021-07-25:
修复完毕，感谢暗云大佬

closed 2021-07-25 by a920025608