coolsnowwolf/lede #7512: AX3600仍然内存泄漏

cath-palug, 2021-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

cath-palug, 2021-07-25:
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.285406] clash invoked oom-killer: gfp_mask=0x100dca(GFP_HIGHUSER_MOVABLE|__GFP_ZERO), order=0, oom_score_adj=0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.285457] CPU: 1 PID: 18982 Comm: clash Not tainted 5.10.51 #0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.294645] Hardware name: Xiaomi AX3600 (DT)
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.300805] Call trace:
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.305065]  dump_backtrace+0x0/0x1b0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.307319]  show_stack+0x18/0x30
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.311137]  dump_stack+0xdc/0x11c
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.314433]  dump_header+0x44/0x194
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.317736]  oom_kill_process+0x1d0/0x1d8
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.321120]  out_of_memory+0x20c/0x320
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.325285]  __alloc_pages_slowpath.constprop.150+0x6c8/0x9b0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.328933]  __alloc_pages_nodemask+0x1f4/0x238
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.334751]  handle_mm_fault+0x4f8/0xc38
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.339086]  do_page_fault+0x1ac/0x3d8
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.343252]  do_translation_fault+0x50/0x60
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.346812]  do_mem_abort+0x40/0xa0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.350894]  el0_da+0x34/0x48
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.354366]  el0_sync_handler+0x68/0xb8
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.357489]  el0_sync+0x184/0x1c0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.361404] Mem-Info:
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624] active_anon:208 inactive_anon:35776 isolated_anon:0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624]  active_file:20 inactive_file:0 isolated_file:0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624]  unevictable:0 dirty:0 writeback:0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624]  slab_reclaimable:2287 slab_unreclaimable:9574
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624]  mapped:29 shmem:492 pagetables:280 bounce:0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.364624]  free:3977 free_pcp:216 free_cma:0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.375745] Node 0 active_anon:832kB inactive_anon:143104kB active_file:80kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:116kB dirty:0kB writeback:0kB shmem:1968kB writeback_tmp:0kB kernel_stack:2064kB all_unreclaimable? yes
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.398472] DMA free:15716kB min:16384kB low:20480kB high:24576kB reserved_highatomic:0KB active_anon:832kB inactive_anon:142192kB active_file:0kB inactive_file:96kB unevictable:0kB writepending:0kB present:524288kB managed:417704kB mlocked:0kB pagetables:1120kB bounce:0kB free_pcp:1072kB local_pcp:188kB free_cma:0kB
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.426583] lowmem_reserve[]: 0 0 0 0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.448821] DMA: 365*4kB (UME) 264*8kB (UE) 216*16kB (UME) 60*32kB (UE) 23*64kB (UE) 9*128kB (ME) 3*256kB (E) 2*512kB (ME) 0*1024kB 1*2048kB (M) 0*4096kB = 15412kB
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.452574] 536 total pagecache pages
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.466961] 0 pages in swap cache
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.470774] Swap cache stats: add 0, delete 0, find 0/0
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.474073] Free swap  = 0kB
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.479103] Total swap = 0kB
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.482230] 131072 pages RAM
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.485094] 0 pages HighMem/MovableOnly
Sun Jul 25 08:32:31 2021 kern.warn kernel: [ 4249.487954] 26646 pages reserved
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.491519] Tasks state (memory values in pages):
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.494993] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.499624] [    887]    81   887      301       28    28672        0             0 ubusd
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.508115] [    895]     0   895      205        9    24576        0             0 askfirst
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.516363] [   1934]   514  1934      296       39    32768        0             0 logd
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.524871] [   1982]     0  1982      391       32    36864        0             0 rpcd
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.532941] [   2332]     0  2332     1550      107    45056        0             0 hostapd
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.541014] [   2333]     0  2333     1522       86    40960        0             0 wpa_supplicant
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.549087] [   2389]     0  2389      470       61    32768        0             0 netifd
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.558114] [   2568]     0  2568      305        9    24576        0             0 udhcpc
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.566361] [   2701]     0  2701      360       35    28672        0             0 odhcpd
[... cut ...]
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.706727] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=clash,pid=6066,uid=65534
Sun Jul 25 08:32:31 2021 kern.err kernel: [ 4249.715305] Out of memory: Killed process 6066 (clash) total-vm:788468kB, anon-rss:133320kB, file-rss:4kB, shmem-rss:0kB, UID:65534 pgtables:376kB oom_score_adj:0
Sun Jul 25 08:32:31 2021 kern.info kernel: [ 4249.761025] oom_reaper: reaped process 6066 (clash), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

madcat0513, 2021-07-25:
借楼问下，现在ath11对ipq807x系列的无线支持如何？有点想入手

cath-palug, 2021-07-25:
> 借楼问下，现在ath11对ipq807x系列的无线支持如何？有点想入手

只要你能忍受内存泄漏就行，其他倒是没问题

miccjing, 2021-07-25:
我AX6也这样更严重了

hzjixiang, 2021-07-25:
> > 借楼问下，现在ath11对ipq807x系列的无线支持如何？有点想入手
> 
> 只要你能忍受内存泄漏就行，其他倒是没问题

每天凌晨定时重启是否可以解决？

ja5onwu, 2021-07-25:
> > > 借楼问下，现在ath11对ipq807x系列的无线支持如何？有点想入手
> > 
> > 
> > 只要你能忍受内存泄漏就行，其他倒是没问题
> 
> 每天凌晨定时重启是否可以解决？

最新编译的版本.我的AX3600甚至没把无线编译进去,开机剩下300左右.然后连接的AP用,大概设备30+.半天时间依然会OOM,感觉这个内存泄露不是无线的问题了.

madcat0513, 2021-07-26:
好吧，我再等等吧😂

coolsnowwolf, 2021-07-26:
[image]

coolsnowwolf, 2021-07-26:
[image]

My-uum, 2021-07-29:
[image]
用最新版的编译内存还是泄露

My-uum, 2021-07-29:
[image]

cath-palug, 2021-07-29:
我先跑了兄弟，QSDK顶上先

My-uum, 2021-07-29:
> 我先跑了兄弟，QSDK顶上先

关了无线，内存恢复了百分之5然后还是一点一点少，剩下20兆左右内存路由器就重启了

ja5onwu, 2021-07-29:
[image]

今天编译的,内存好像没问题了.但负载好高

My-uum, 2021-07-29:
[image]
内存溢出又重启了

LilithBristol, 2021-08-01:
8月2号 现在还漏吗？

hohhhhh, 2021-08-04:
8.1号的依旧会

ja5onwu, 2021-08-05:
测了几天,现在无线不会漏了.其实是SSRP占用内存越来越大.然后OOM.大家可以试下,内存剩下很少的时候,重新应用一次SSRP是不是释放内存出来了

closed 2023-12-22 by aiamadeus