coolsnowwolf/lede #7514: PPTP连接家里vpn后无法从ssr走流量

bonnyandsky, 2021-07-25:
1年多没折腾，最近编译了固件，然后发现pptp连接到op后可以访问家里局域网设备，可以上外网，就是不会走ssr的代理。
看了下之前的帖子，反馈的都是ipsec的问题（ipsec我手机或者win10都连不上，不知道什么情况，所以一直用pptp）

但是我1年前编译的固件，pptp连接上去后是可以通过ssr访问外网的。不知道1年多发生了什么。需要额外设置还是其他什么问题？还请各位指导

coolsnowwolf, 2021-07-26:
[image]

bonnyandsky, 2021-07-27:
按照上图设置，pptp连接后还是无法通过ssr出去，路由器重启，开关ssr服务，开关pptp服务都试过了
此openwrt是安装在群晖nas的vmm下。
版本号：OpenWrt R21.7.15 / LuCI Master (git-21.206.25982-60d43ca)
1：pptp设置如下：op本地lan地址：192.168.0.3，同时发现一个bug，见https://github.com/coolsnowwolf/lede/issues/7513
除非修改/etc/ppp/options.pptpd文件或者客户端本地指定dns。不然连上pptp后无法访问外网，只能访问本地隧道网络。
[image]
2：ssr设置如下：
[image]
3：ssr测试链接正常
[image]

bonnyandsky, 2021-07-27:
然后发现有客户端通过pptp连上来以后，会有自己生成的一个ppp0接口，断开这个接口就没有了。但是我不能新建一个自定义叫ppp0的接口，这样会冲突，web配置网页都打不开
[image]

bonnyandsky, 2021-07-28:
看了下 pptp连上去后，查ip能确认是家里的网络ip但是流量走不了v2出去。

hahaher123, 2021-07-29:
用wireguard吧

bonnyandsky, 2021-07-29:
> 用wireguard吧

在网友的帮助下暂时用ss server+sstap客户端解决了我的需求。
sstap可以可以让电脑的全局流量都走ss。然后ss server又可以在路由器端把流量通过v2转发出去。这样既可以留学，也可以访问局域网设备。关键是支持smb方式。

ipsec死活连不上我估计应该是运营商封了对应的端口导致的。
pptp这个暂时不折腾了。过一段时间不知道会不会突然变好了。这样就不要电脑上用额外软件了

wangweitung, 2021-08-06:
> > 用wireguard吧
> 
> 在网友的帮助下暂时用ss server+sstap客户端解决了我的需求。
> sstap可以可以让电脑的全局流量都走ss。然后ss server又可以在路由器端把流量通过v2转发出去。这样既可以留学，也可以访问局域网设备。关键是支持smb方式。
> 
> ipsec死活连不上我估计应该是运营商封了对应的端口导致的。
> pptp这个暂时不折腾了。过一段时间不知道会不会突然变好了。这样就不要电脑上用额外软件了

我的PPTP也连不上很久了，估计也是运营商封了端口。

jashok2, 2021-08-13:
你们为何不用openvpn?

wangweitung, 2021-08-13:
> 
> 
> 你们为何不用openvpn?

因为是受控制的电脑，不能安装其他软件，PPTP是系统自带。更方便。

bonnyandsky, 2021-08-14:
> 你们为何不用openvpn?

你这个逻辑有问题。反馈pptp的问题，然后提出的方案是使用openvpn。这不对啊，哈哈。暂时用ss解决了我的需求，但是还是希望pptp能修好。最近的版本编译后pptp都出不了外网了

closed 2024-11-19 by coolsnowwolf