coolsnowwolf/lede #7515: RT4230W 512mb micron slc nand刷入大于38MB固件后无法启动问题

a920025608, 2021-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [√ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷入大于38mb固件后会重启，提示 Internal error: Oops: 17 [#1] SMP ARM

### (2) 路由器型号和固件版本

A：RT4230W rev.9

### (3) 详细日志

A：
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (runner@fv-az181-341) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3569-1c3806e11)) #0 SMP Sun Jul 25 05:38:42 2021
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line not present
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996172K/1015808K available (6265K kernel code, 198K rwdata, 1568K rodata, 1024K init, 232K bss, 19636K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000008] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000022] Switching to timer-based delay loop, resolution 160ns
[    0.000273] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000293] pid_max: default: 32768 minimum: 301
[    0.000514] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000537] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001610] CPU: Testing write buffer coherency: ok
[    0.002751] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002928] rcu: Hierarchical SRCU implementation.
[    0.003421] smp: Bringing up secondary CPUs ...
[    0.005328] smp: Brought up 1 node, 2 CPUs
[    0.005345] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005356] CPU: All CPU(s) started in SVC mode.
[    0.016554] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016697] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016726] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016875] pinctrl core: initialized pinctrl subsystem
[    0.018024] NET: Registered protocol family 16
[    0.018346] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019674] cpuidle: using governor ladder
[    0.019743] cpuidle: using governor menu
[    0.055225] usbcore: registered new interface driver usbfs
[    0.055303] usbcore: registered new interface driver hub
[    0.055375] usbcore: registered new device driver usb
[    0.055433] pps_core: LinuxPPS API ver. 1 registered
[    0.055447] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.055478] PTP clock support registered
[... 480 lines cut ...]
[   89.402096] Exception stack(0xef127fb0 to 0xef127ff8)
[   89.409136] 7fa0:                                     00000000 00000000 00000000 00000000
[   89.414271] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   89.422427] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   89.430584] Code: eb0938a7 e5d43da4 e3130002 1a00002a (e5da3000) 
[   89.437104] ---[ end trace b5f8f6fa0d1f57d7 ]---
[   89.443309] Kernel panic - not syncing: Fatal exception
[   89.447946] CPU1: stopping
[   89.452889] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G      D  C        5.4.132 #0
[   89.455663] Hardware name: Generic DT based system
[   89.463238] [<c030f954>] (unwind_backtrace) from [<c030b968>] (show_stack+0x14/0x20)
[   89.467832] [<c030b968>] (show_stack) from [<c0900cf8>] (dump_stack+0x94/0xa8)
[   89.475731] [<c0900cf8>] (dump_stack) from [<c030eb80>] (handle_IPI+0x184/0x1b8)
[   89.482772] [<c030eb80>] (handle_IPI) from [<c05dd588>] (gic_handle_irq+0xb4/0xb8)
[   89.490318] [<c05dd588>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[   89.497679] Exception stack(0xef08ff18 to 0xef08ff60)
[   89.505238] ff00:                                                       00000000 00000014
[   89.510308] ff20: 2ebde000 ef71fa80 ee8e0400 00000000 ef71ee30 00000014 00000014 00000000
[   89.518465] ff40: d3831840 d3828600 00000015 ef08ff68 c07312a0 c07312a4 80000013 ffffffff
[   89.526612] [<c0301a8c>] (__irq_svc) from [<c07312a4>] (cpuidle_enter_state+0x94/0x498)
[   89.534763] [<c07312a4>] (cpuidle_enter_state) from [<c07316ec>] (cpuidle_enter+0x30/0x4c)
[   89.542573] [<c07316ec>] (cpuidle_enter) from [<c034ad6c>] (do_idle+0x1d8/0x240)
[   89.550902] [<c034ad6c>] (do_idle) from [<c034b07c>] (cpu_startup_entry+0x1c/0x20)
[   89.558453] [<c034b07c>] (cpu_startup_entry) from [<423024cc>] (0x423024cc)
[   90.565945] Rebooting in 3 seconds..

110010110, 2021-07-26:
冒昧问一下，这固件 “rev.9” 怎么来的呢？

a920025608, 2021-07-26:
> 冒昧问一下，这固件 “rev.9” 怎么来的呢？

拆机时主板丝印标明的，实质上是 rev6，只不过nand换成了镁光

110010110, 2021-07-26:
> > 冒昧问一下，这固件 “rev.9” 怎么来的呢？
> 
> 拆机时主板丝印标明的，实质上是 rev6，只不过nand换成了镁光

好的，了解了，谢谢。

closed 2021-07-26 by a920025608