coolsnowwolf/lede #7523: [AX6] 开启nss之后防火墙端口转发 本地环回失效 以及 随机重启

xsm1997, 2021-07-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

1. 最新版本开启nss之后，防火墙端口转发本地回环会失效，DDNS访问无法打开，且仅发生在内网里面，外网可以正常访问。一两天前的版本，可以内网访问DDNS，但会有数据截断等一系列奇葩问题，导致网页间歇性无法正常加载。一切相关问题在取消编译nss、nss-pppoe、nss-ecm后消失。

2. 从开始支持nss之后，我的这台AX6只要刷入开启了nss-ecm的固件，就会随机重启，频率很高，平均5分钟一次。在恢复出厂设置之后正常联网，也会随机重启。这个现象跟SFE无关，关闭SFE编译之后也会重启。在取消编译nss-ecm之后，可以稳定运行一天不重启。是否可能跟温度有关？但无法解释不编译nss就不重启，开启nss之后负载应该更低，温度也应该更低。或者是我这台AX6中奖了，应该换货？有多少人编译nss后随机重启呢？

### (2) 路由器型号和固件版本

Redmi AX6

### (3) 详细日志

因为是随机重启，所以不能直接抓取日志。考虑可能发生了Kernel Panic，但无crashlog文件。能否告知这种情况下如何抓取log？我愿意继续跟进。

excelwang, 2021-07-30:
遇到了同样的问题。最新版本会出现随机重启。

xsm1997, 2021-07-30:
可不可能是由于CPU的差别（比如步进等）导致的不稳定问题呢？我身边不止我一个出现了随机重启问题，而且是在nss支持之初就有问题；而另一部分人从始至终都没有问题。

ramondelee, 2021-08-03:
我也是10分钟左右重启，个人怀疑因CPU超频至1.4G（原频率1G）导致NPU也自动超了，CPU超频稳定，但NPU超频后NSS驱动网络使用NPU导致NPU负载比较高，然后死机重启。不重启都是买到了体质好的。所以建议调整CPU频率试试。

xsm1997, 2021-08-04:
@ramondelee 改dts删掉了1.4G的频率，同样重启。

hohhhhh, 2021-08-04:
AX3600刷了lede也是无限重启 大概五六分钟一次 有时候大概二十分钟

ramondelee, 2021-08-04:
昨天晚上试了试把NSS的频率改了，编译后发现NSS不起作用了，但不会重启。另外看有人提AX3600还有内存泄漏问题，AX6倒没看是不是开NSS导致内存泄漏原因引起的。

hohhhhh, 2021-08-05:
之前看了个帖子 有个人说NSS和openclash冲突 所以会不会是它的问题 试下编译的时候去掉openclash？

xsm1997, 2021-08-05:
没编译openclash。

Rhystar, 2021-09-07:
你开ipv6了吗？我的mf269，开启ipv6后会重启，关闭ipv6后能正常使用，会不会是ipq8071a的nss和ipv6冲突

closed 2022-08-16 by xsm1997