coolsnowwolf/lede #7533: luci-app-cpufreq 编译异常

bilibilistack, 2021-07-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译过程报错，使用本地和github action环境编译均失败。之前此配置可顺利编译

### (2) 路由器型号和固件版本

A：R7800

### (3) 详细日志

A：

Configuring luci-i18n-nlbwmon-zh-cn.
	But that file is already provided by package  * base-files
 * opkg_install_cmd: Cannot install package luci-app-cpufreq.
 * check_data_file_clashes: Package luci-app-cpufreq wants to install file /workdir/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/etc/init.d/cpufreq
	But that file is already provided by package  * base-files
 * opkg_install_cmd: Cannot install package luci-i18n-cpufreq-zh-cn.
make[2]: *** [package/install] Error 255
make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make: *** [world] Error 2
package/Makefile:66: recipe for target 'package/install' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:111: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:228: recipe for target 'world' failed
Error: Process completed with exit code 2.

bilibilistack, 2021-07-26:
类似问题：https://github.com/coolsnowwolf/lede/issues/7148

coolsnowwolf, 2021-07-27:
ipq806x 用的 CPU 跳频驱动是 krait v2 ，高通手机用的，不适合这个 app

bilibilistack, 2021-07-27:
明白了，谢谢

AmadeusGhost, 2021-07-27:
其实只是文件名冲突了🌚

closed 2021-09-15 by bilibilistack