coolsnowwolf/lede #7536: 总有那么几天404error

smallprogram, 2021-07-27:
总有那么几天cryptopp564.zip Download会间歇性抽风404.
什么
https://sources.cdn.openwrt.org/cryptopp564.zip
https://sources.openwrt.org/cryptopp564.zip
https://mirror2.openwrt.org/sources/cryptopp564.zip
啊，间歇性404 Not Found

Boos4721, 2021-07-27:
自己mirror下不就行了 或者换个好点的梯

smallprogram, 2021-07-27:
> 自己mirror下不就行了 或者换个好点的梯

GitHub action。编译环境本来就在国外。mirror啥啊mirror。

Chandler-Lu, 2021-08-10:
我今天 mirror2 的 libcap 也404去了。。没啥办法，土豆服务器

1-1-2, 2021-08-15:
试了好久，卡到了一台下载失败的机器SSH上去，同时安排一个可以正常访问的，分别执行

```shell
curl -fvO --connect-timeout 20 --retry 5 --location --insecure http://www.cryptopp.com/cryptopp564.zip
```

直接上对比图。
[image]

左：无法下载的机器；右：可以正常下载的机器。

综上，我觉着是微软某处机房的网络设置的问题导致的下不下来。

1-1-2, 2021-08-15:
TCPing also
[image]

closed 2021-09-16 by AmadeusGhost

smallprogram, 2021-10-04:
写了个注入，自动将cryptopp564.zip在make download的时候注入到dl文件夹，临时解决了这个问题。
https://github.com/smallprogram/OpenWrtAction/commit/6450f524a18e4555480b6c555e685c5c733fbcc7
但是不是很完美，一旦源码更新了，不用cryptopp564.zip用更高的版本了。我还得手动更新zip文件。。。