coolsnowwolf/lede #7537: 关于小娱c5没有交换机选项的问题

bonnyandsky, 2021-07-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 有类似的issue

## 2. 详细叙述
最近从r9升到R21,用以前修改network文件的方式设置vlan好像失效了。然后查了一下发现图形界面也没有交换机的设置选项。看了前几天别人的IS说小米的路由器也没有了相关选项，因为xiaoyi用的也是7621，不知道是不是同样的问题。不能设置vlan，家里的IPTV就没法用了，还请大神指导一下，在官方不修复的情况下面怎么临时性解决问题。
### (1) 具体问题

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

Neverends4, 2021-07-27:
[新DSA架构配置说明](https://forum.openwrt.org/t/mini-tutorial-for-dsa-network-config/96998)
来自官方论坛，适用的Luci这里还没有，不过有配置文件格式，你可以依样画葫芦。
我没画出来。

bonnyandsky, 2021-07-27:
> [新DSA架构配置说明](https://forum.openwrt.org/t/mini-tutorial-for-dsa-network-config/96998)
> 来自官方论坛，适用的Luci这里还没有，不过有配置文件格式，你可以依样画葫芦。
> 我没画出来。

是的我今天也看了半天英文的文档。死活没看懂

Neverends4, 2021-07-28:
架构变了，配置文件格式也变了。如果是这样估计没法整，得等作者修复
最快的方法就是用大更新前老版本的源码，其实我觉得不开HWNAT，3月份前的源码也很稳定呀。路由器固件没必要一直更新。

bonnyandsky, 2021-07-28:
> 架构变了，配置文件格式也变了。如果是这样估计没法整，得等作者修复
> 最快的方法就是用大更新前老版本的源码，其实我觉得不开HWNAT，3月份前的源码也很稳定呀。路由器固件没必要一直更新。

我上一次编译固件是1年半之前，因为那个时候的固件可能没设置好 ipv6支持有问题。所以最近才折腾固件，结果发现好多坑。现在还有2个没填。包括pptp的流量走不了v2出去

PR #7560 (2021-07-30): ramips: mt7621: add back switch support
fix: #7500 , fix: #7508 , fix: #7517 , fix: #7537
closed 2021-07-30 by coolsnowwolf