coolsnowwolf/lede #7544: samba访问速度问题

whfweb, 2021-07-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由器samba服务器只连接电脑的时候读写速度正常，都是80M/s左右，可是一但连接上机顶盒，自电脑的写入速度大幅下降至10M/s。实测石像鬼固件没有这个现象。

### (2) 路由器型号和固件版本

A：wrt1900ac

### (3) 详细日志

A：没有日志的错误反应。

Neverends4, 2021-07-28:
IPTV多播包广播风暴？抓包看看？

whfweb, 2021-07-28:
> IPTV多维多姆·德科·德尼奥特？特夫特·科瓦德？

> IPTV多播包广播风暴？抓包看看？

不是 机顶盒只要连线开机 不打开任何app就会影响到

whfweb, 2021-07-28:
> IPTV多播包广播风暴？抓包看看？

不是 机顶盒只要连线开机 不打开任何app就会影响到

coolsnowwolf, 2021-07-28:
你的机顶盒可能只支持 SMB v1 ，做了 NetBIOS 广播，导致降级为 v1 了

whfweb, 2021-07-28:
> 你的机顶盒可能只支持 SMB v1 ，做了 NetBIOS 广播，导致降级为 v1 了

一语中的了，确实是这样 机顶盒只支持SMB v1 ，应该就是smb版本问题，石像鬼很长时间没有更新了。

whfweb, 2021-07-29:
> 你的机顶盒可能只支持 SMB v1 ，做了 NetBIOS 广播，导致降级为 v1 了

老大一语中的了，机顶盒只支持SMBv1，石像鬼也是很久没有更新了。请问应该怎么解决呢？

closed 2021-08-06 by whfweb