coolsnowwolf/lede #7547: 建议：IPSec VPN服务器Luci界面提供Keepalive选项

uapolysci, 2021-07-28:
目前的IPSec VPN服务器无法设置KeepAlive，导致iOS 客户端连接后无流量时或者息屏后很快断连。希望能够增加KeepAlive选项，可以开启并且设置KeepAlive指令的间隔时间/时长。谢谢！

galaxyskyknight, 2021-07-28:
/etc/init.d/ipsec里自己添加一行

galaxyskyknight, 2021-07-28:
我以前也搞softethervpn/ipsec/openvpn,现在这些都被我统统抛弃了，论vpn好使，只有n2n，当然ios客户端没有.....openvpn可以做备用，ipsec太难用了，而且只能v1, IKEv2的安卓端原生不支持好像。

uapolysci, 2021-07-28:
> /etc/init.d/ipsec里自己添加一行

加在这里的参数在下次通过Luci修改IPSec配置后会被覆盖吗？

uapolysci, 2021-07-28:
> 我以前也搞softethervpn/ipsec/openvpn,现在这些都被我统统抛弃了，论vpn好使，只有n2n，当然ios客户端没有.....openvpn可以做备用，ipsec太难用了，而且只能v1, IKEv2的安卓端原生不支持好像。

是的。主要还是没有好用的原生客户端。IPSec iOS内置支持，而且也可以比较容易穿透公司的防火墙。

galaxyskyknight, 2021-07-29:
> > /etc/init.d/ipsec里自己添加一行
> 
> 加在这里的参数在下次通过Luci修改IPSec配置后会被覆盖吗？

不会，luci界面改的是etc/config下的配置，init脚本起来后会读/etc/config下的配置文件解析后结合Init脚本里非界面可配置的固化配置，最终拼合成完整的内存加载配置输出给应用binary加载（一般在/tmp/下或/var下）， openwrt lede里配置文件生成大多数是这个套路。

dwj0, 2021-08-05:
> 我以前也搞softethervpn/ipsec/openvpn,现在这些都被我统统抛弃了，论vpn好使，只有n2n，当然ios客户端没有.....openvpn可以做备用，ipsec太难用了，而且只能v1, IKEv2的安卓端原生不支持好像。

请问n2n有好用的win和安卓客户端吗，需要能设置自定义路由的。安卓的hin2n没这功能，win的easyn2n也没有

zhouxiaolong920, 2021-08-06:
openvpn。非常好用。
我公司2k的人。都一直用openvpn，非常稳定，速度快

dwj0, 2021-08-07:
> openvpn。非常好用。
> 我公司2k的人。都一直用openvpn，非常稳定，速度快

openvpn 我的安卓手机提示耗电异常，手机发热

liangchaofu, 2022-03-05:
> /etc/init.d/ipsec里自己添加一行
在init.d的ipsec里面添加了就连不上了，
keep_alive对吗，还是我添加的地方错了，请大佬指点，谢谢。

closed 2024-11-19 by coolsnowwolf