coolsnowwolf/lede #7556: docker编译后无法运行

haohaoget, 2021-07-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译最新版固件运行后Docker CE 未运行，解锁网易云音乐也无法运行

### (2) 路由器型号和固件版本

A：x86_64，最新版

### (3) 详细日志

A：Thu Jul 29 16:31:38 2021 daemon.err dockerd[6238]: time="2021-07-29T08:31:38.078605643Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Thu Jul 29 16:31:38 2021 daemon.err dockerd[6238]: time="2021-07-29T08:31:38.080556683Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Thu Jul 29 16:31:38 2021 daemon.err dockerd[6238]: time="2021-07-29T08:31:38.136150326Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Thu Jul 29 16:31:38 2021 daemon.err dockerd[6238]: time="2021-07-29T08:31:38.136451323Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Thu Jul 29 16:31:39 2021 daemon.err dockerd[6238]: failed to start daemon: Devices cgroup isn't mounted
Thu Jul 29 16:31:44 2021 daemon.err modprobe: failed to find a module named aufs
Thu Jul 29 16:31:44 2021 daemon.err dockerd[6399]: time="2021-07-29T08:31:44.888521310Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Thu Jul 29 16:31:44 2021 daemon.err dockerd[6399]: time="2021-07-29T08:31:44.889300216Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Thu Jul 29 16:31:44 2021 daemon.err dockerd[6399]: time="2021-07-29T08:31:44.991122509Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Thu Jul 29 16:31:44 2021 daemon.err dockerd[6399]: time="2021-07-29T08:31:44.991251023Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Thu Jul 29 16:31:45 2021 daemon.err dockerd[6399]: failed to start daemon: Devices cgroup isn't mounted

ZqinKing, 2021-07-30:
lxc中也遇到问题，编译时选了tar.gz。几个月前的编译出来放在lxc中运行没问题。现在的版本放进lxc运行后没有网卡，ifconfig手动up是可以起来。网桥也没有。tun之类的已经挂载进容器。有一点就是现在编译的版本作为容器运行后，容器内无法运行modprobe，提示/bin/ash: modprobe: not found

shanhai2333, 2021-08-18:
我也这个问题

lo-w, 2021-08-22:
> 我也这个问题

是否有使用privileged 参数, 默认的话我也是看不到网络 dropbear 提示没权限, 使用privileged 参数后一切正常了.
docker run -itd -P --privileged
BR//lo

shanhai2333, 2021-08-23:
> > 我也这个问题
> 
> 是否有使用特权参数，自然的话我也是看不到网络 dropbear 提示没有权限，使用特权参数后一切正常了。
> docker run -itd -P --privileged
> BR//lo

我的是和日志一样，docker都没启动起来，

lo-w, 2021-08-30:
> 我的是和日志一样，docker都没启动起来，
加了privileged 也没起来?
BR//lo

shanhai2333, 2021-09-03:
> > 我的是和日志一样，docker都没启动起来，
> > 加了privileged 也没起来?
> > BR//lo

root@OpenWrt-ShanHai:~# docker run -itd -P --privileged
"docker run" requires at least 1 argument.
See 'docker run --help'.

Usage:  docker run [OPTIONS] IMAGE [COMMAND] [ARG...]

Run a command in a new container
，
我百度到是docker没启动，然后输入：/etc/init.d/dockerd start
系统日志就是这样
Fri Sep  3 16:58:39 2021 daemon.err modprobe: failed to find a module named aufs
Fri Sep  3 16:58:39 2021 daemon.err dockerd[17969]: time="2021-09-03T08:58:39.637907456Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Fri Sep  3 16:58:39 2021 daemon.err dockerd[17969]: time="2021-09-03T08:58:39.639398617Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Fri Sep  3 16:58:40 2021 daemon.err dockerd[17969]: time="2021-09-03T08:58:40.465352675Z" level=warning msg="Your kernel does not support swap memory limit"
Fri Sep  3 16:58:40 2021 daemon.err dockerd[17969]: time="2021-09-03T08:58:40.465503530Z" level=warning msg="Your kernel does not support cgroup blkio weight"
Fri Sep  3 16:58:40 2021 daemon.err dockerd[17969]: time="2021-09-03T08:58:40.465547460Z" level=warning msg="Your kernel does not support cgroup blkio weight_device"
Fri Sep  3 16:58:41 2021 daemon.err dockerd[17969]: failed to start daemon: Devices cgroup isn't mounted

lo-w, 2021-09-09:
你这个看起来是想要在op 上面run docker ? 还是自己编译的OP 想要run 在docker 里?
OP run docker:
Install both docker & dockerd 
root@OpenWrt:~# opkg install dockerd --force-depends 

For GUI. 
https://github.com/lisaac/luci-lib-docker/ 
https://github.com/lisaac/luci-app-dockerman/ 
root@OpenWrt:~# ll luci-* 
-rwxr-xr-x    1 root     root         57489 Jul 17 13:33 luci-app-dockerman_v0.5.23-2_all.ipk* 
-rwxr-xr-x    1 root     root          4432 Jul 17 13:35 luci-lib-docker_v0.3.4_all.ipk* 
root@OpenWrt:~# opkg install luci-lib-docker_v0.3.4_all.ipk 
… 
Configuring luci-lib-docker. 
root@OpenWrt:~# opkg install luci-app-dockerman_v0.5.23-2_all.ipk 
Configuring luci-app-dockerman. 
BR//lo

closed 2024-11-19 by coolsnowwolf