coolsnowwolf/lede #7557: 请问怎么改4.19内核？Makefile里面只有5.4  5.10而已

yuzhe1129, 2021-07-29:
能否说下怎么改4.19的内核，我看Makefile这个文件只有5.4  5.10  我改4.19编译不成功

itfish92, 2021-07-29:
在这里   https://github.com/coolsnowwolf/openwrt
去年是4.19，今年是4.14.
感觉都挺稳

yuzhe1129, 2021-07-29:
> 在这里 https://github.com/coolsnowwolf/openwrt
> 去年是4.19，今年是4.14。
> 感觉都挺稳的

原来是op，谢谢

itfish92, 2021-07-30:
> > 在这里https://github.com/coolsnowwolf/openwrt
> > 去年是4.19，今年是4.14。
> > 感觉都挺稳的
> 
> 原来是op，谢谢

那就把这个issues关了吧

yuzhe1129, 2021-07-30:
> > > 在这里https://github.com/coolsnowwolf/openwrt
> > > 去年是4.19，今年是4.14。
> > > 感觉都挺稳的
> > 
> > 
> > 原来是op，谢谢
> 
> 那就把这个issues关了吧
atching file net/ipv4/tcp_offload.c
patching file net/ipv6/netfilter/ip6table_mangle.c
Hunk #2 FAILED at 59.
1 out of 2 hunks FAILED -- saving rejects to file net/ipv6/netfilter/ip6table_mangle.c.rej
Patch failed! Please fix /home/xxoo/openwrt/target/linux/ath79/patches-4.14/910-unaligned_access_hacks.patch!
make[4]: *** [Makefile:29: /home/xxoo/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-4.14.231/.prepared_4b1d96f28098506d2a9786e48db24a74] Error 1
make[4]: Leaving directory '/home/xxoo/openwrt/target/linux/ath79'
make[3]: *** [Makefile:13: compile] Error 2
make[3]: Leaving directory '/home/xxoo/openwrt/target/linux'
time: target/linux/compile#1.97#10.99#10.19
make[2]: *** [target/Makefile:25: target/linux/compile] Error 2
make[2]: Leaving directory '/home/xxoo/openwrt'
make[1]: *** [target/Makefile:18: /home/xxoo/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xxoo/openwrt'
make: *** [/home/xxoo/openwrt/include/toplevel.mk:227: world] Error 2
我编译了，但是弹出这个错误，这个怎么解决啊

closed 2021-08-07 by coolsnowwolf