coolsnowwolf/lede #7559: 编译三方kcptun时冲突报错

rilyuuj, 2021-07-30:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
添加luci-app-kcptun 后，编译时组件冲突，无法继续进行。

### (1) 具体问题

添加自定义应用luci-app-kcptun时，仓库地址https://github.com/kuoruan/luci-app-kcptun。这次更新sdk后，若弃用编译kcptun时候会冲突报错，原因可能是与kcptun-config使用的了相同的init.d路径，之前编译时是不会报错的。使用luci-app-kcptun.ipk安装方式也不行，必须卸载软件包kcptun-config.
与固件一同编译时，发现无法取消kcptun-config关联，不能单独kcptun-client
[image]
请问如何取消这个kcptun config script关联?这个script主要是什么作用？是否影响ssrp调用kcptun？
目前只想单独编译kcptun-client外加三方仓库的luci-app-kcptun,请问该怎么操作

### (2) 路由器型号和固件版本
sunxi-cortexa7-bpim2u

### (3) 详细日志

Configuring default-settings.
Configuring luci-i18n-base-zh-cn.
Configuring kmod-sun4i-emac.
Configuring luci-i18n-upnp-zh-cn.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Collected errors:
 * check_data_file_clashes: Package luci-app-kcptun wants to install file /home/bio/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/etc/init.d/kcptun
        But that file is already provided by package  * kcptun-config
 * opkg_install_cmd: Cannot install package luci-app-kcptun.
 * check_data_file_clashes: Package luci-app-kcptun wants to install file /home/bio/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-sunxi/etc/init.d/kcptun
        But that file is already provided by package  * kcptun-config
 * opkg_install_cmd: Cannot install package luci-i18n-kcptun-zh-cn.
package/Makefile:66: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/bio/lede'
package/Makefile:111: recipe for target '/home/bio/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install' failed
make[1]: *** [/home/bio/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/bio/lede'
/home/bio/lede/include/toplevel.mk:228: recipe for target 'world' failed
make: *** [world] Error 2

WYC-2020, 2021-08-11:
修改第三方程序的路径或者名字就行了，你认为冲突的话

closed 2021-08-23 by AmadeusGhost

rilyuuj, 2021-08-24:
> 修改第三方程序的路径或者名字就行了，你认为冲突的话

不好意思，请问是说第三方得编译路径，还是程序得存放路径？我尝试单独安装编译好得第三方ipk，也是会报同样得错误呢