coolsnowwolf/lede #7563: openwrt编译出现错误，这个怎么解决

yuzhe1129, 2021-07-30:
patching file net/ipv6/ip6_fib.c
patching file net/netfilter/nf_conntrack_proto_tcp.c
Hunk #1 succeeded at 453 (offset 6 lines).
patching file net/xfrm/xfrm_input.c
patching file net/ipv4/tcp_input.c
patching file include/uapi/linux/if_pppox.h
patching file net/ipv6/netfilter/nf_log_ipv6.c
patching file include/net/neighbour.h
patching file include/uapi/linux/netfilter_arp/arp_tables.h
patching file net/core/utils.c
patching file include/linux/etherdevice.h
patching file net/ipv4/tcp_offload.c
patching file net/ipv6/netfilter/ip6table_mangle.c
Hunk #2 FAILED at 59.
1 out of 2 hunks FAILED -- saving rejects to file net/ipv6/netfilter/ip6table_mangle.c.rej
Patch failed!  Please fix /home/xxoo/openwrt/target/linux/ath79/patches-4.14/910-unaligned_access_hacks.patch!
make[4]: *** [Makefile:29: /home/xxoo/openwrt/build_dir/target-mips_24kc_musl/linux-ath79_tiny/linux-4.14.231/.prepared_4b1d96f28098506d2a9786e48db24a74] Error 1
make[4]: Leaving directory '/home/xxoo/openwrt/target/linux/ath79'
make[3]: *** [Makefile:13: compile] Error 2
make[3]: Leaving directory '/home/xxoo/openwrt/target/linux'
time: target/linux/compile#1.97#10.99#10.19
make[2]: *** [target/Makefile:25: target/linux/compile] Error 2
make[2]: Leaving directory '/home/xxoo/openwrt'
make[1]: *** [target/Makefile:18: /home/xxoo/openwrt/staging_dir/target-mips_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xxoo/openwrt'
make: *** [/home/xxoo/openwrt/include/toplevel.mk:227: world] Error 2

ZiangGithub, 2021-07-31:
我也出现同样的问题，求助！

WYC-2020, 2021-08-10:
明显的打补丁失败，自己找下就知道

closed 2022-02-12 by Beginner-Go