coolsnowwolf/lede #7569: [MT7621] Newifi-D2 build failed

EnnawYang, 2021-07-31:
Target: MT7621 Newifi D2
```
Applying /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/804-staging-mt7621-pinctrl-use-ngpios-not-num-gpios.patch using plaintext: 
patching file drivers/staging/mt7621-pinctrl/pinctrl-rt2880.c

Applying /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/805-pinctrl-AW9523.patch using plaintext: 
patching file drivers/pinctrl/Kconfig
patching file drivers/pinctrl/Makefile

Applying /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/990-NET-no-auto-carrier-off-support.patch using plaintext: 
patching file drivers/net/phy/phy.c
patching file include/linux/phy.h

Applying /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/991-at803x.patch using plaintext: 
patching file drivers/net/phy/at803x.c
patch unexpectedly ends in middle of line
Hunk #6 succeeded at 593 with fuzz 1.

Applying /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/999-crypto-eip93-fix.patch using plaintext: 
patching file include/linux/crypto.h
Hunk #1 FAILED at 101.
1 out of 1 hunk FAILED -- saving rejects to file include/linux/crypto.h.rej
Patch failed!  Please fix /home/xw/openwrt/openwrt/target/linux/ramips/patches-5.4/999-crypto-eip93-fix.patch!
```

AmadeusGhost, 2021-07-31:
这个 patches 删掉就好，实际上是不需要的。

EnnawYang, 2021-07-31:
> 这个 patches 删掉就好，实际上是不需要的。

好吧（

closed 2021-07-31 by EnnawYang