coolsnowwolf/lede #7571: 前天的代码跑在esxi里，突然死机，dhcp挂掉，控制台进不去，不能滚动

YourWildDad, 2021-07-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：前天晚上的代码，action编译，只增加了ssrp，其余全部采用defconfig，运行中突然死机，dhcp挂掉，从esxi中登陆控制台，不能相应键盘事件，不能滚动

配置详情：

```
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_TARGET_KERNEL_PARTSIZE=160
CONFIG_TARGET_ROOTFS_PARTSIZE=2048
```

### (2) 路由器型号和固件版本

A：https://github.com/YourWildDad/lede/actions/runs/1077830092 编译链接

[image]

### (3) 详细日志

A：只能看到最终的控制台输出，无法查看日志

[image]

acooler15, 2021-08-07:
楼主找到解决办法了么？

YourWildDad, 2021-08-09:
没有，我在想办法抓日志。

YourWildDad, 2021-08-23:
保留了日志，但是看不出有什么问题，除了我的电视机在不断尝试ssh登陆我的路由器。
```
Sun Aug 15 22:00:41 2021 kern.notice kernel: [   10.424457] random: crng init done
Sun Aug 15 22:00:41 2021 daemon.notice wpa_supplicant[3016]: Successfully initialized wpa_supplicant
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: 8021ad
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: 8021q
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: macvlan
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: veth
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: bridge
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: Network device
Sun Aug 15 22:00:41 2021 user.notice : Added device handler type: tunnel
Sun Aug 15 22:00:41 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Sun Aug 15 22:00:41 2021 cron.err crond[3182]: crond (busybox 1.33.1) started, log level 8
Sun Aug 15 22:00:41 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.931632] vmxnet3 0000:13:00.0 eth0: intr type 3, mode 0, 5 vectors allocated
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.932508] vmxnet3 0000:13:00.0 eth0: NIC Link is Up 10000 Mbps
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.933117] 8021q: adding VLAN 0 to HW filter on device eth0
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.933760] br-lan: port 1(eth0) entered blocking state
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.934232] br-lan: port 1(eth0) entered disabled state
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.934782] device eth0 entered promiscuous mode
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.935550] br-lan: port 1(eth0) entered blocking state
Sun Aug 15 22:00:41 2021 kern.info kernel: [   10.935994] br-lan: port 1(eth0) entered forwarding state
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'lan' is enabled
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'lan' is setting up now
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'lan' is now up
Sun Aug 15 22:00:41 2021 daemon.notice netifd: bridge 'br-lan' link is up
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'loopback' is enabled
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'loopback' is setting up now
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'loopback' is now up
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'wan' is enabled
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'wan6' is enabled
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Network device 'eth0' link is up
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Network device 'lo' link is up
Sun Aug 15 22:00:41 2021 daemon.notice netifd: Interface 'loopback' has link connectivity
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90ipsec: Command failed: Not found
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = bbr
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90turboacc: killall: dnscache: no process killed
Sun Aug 15 22:00:42 2021 kern.alert kernel: [   11.324666] fast-classifier (PBR safe v2.1.4a): starting up
Sun Aug 15 22:00:42 2021 kern.alert kernel: [   11.325131] fast-classifier: registered
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90turboacc: sh: you need to specify whom to kill
Sun Aug 15 22:00:42 2021 daemon.notice procd: /etc/rc.d/S90turboacc: Stop DNS Caching
Sun Aug 15 22:00:42 2021 authpriv.warn dropbear[3706]: Failed listening on '22': Error listening: Address not available
Sun Aug 15 22:00:42 2021 authpriv.info dropbear[3706]: Not backgrounding
Sun Aug 15 22:00:42 2021 kern.info kernel: [   11.968455] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sun Aug 15 22:00:43 2021 daemon.notice procd: /etc/rc.d/S90turboacc: PDNSD: Start DNS Caching
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Network device 'eth1' link is up
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Interface 'wan6' has link connectivity
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Interface 'wan6' is setting up now
Sun Aug 15 22:00:44 2021 kern.info kernel: [   13.891098] e1000e: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
Sun Aug 15 22:00:44 2021 kern.info kernel: [   13.891672] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
Sun Aug 15 22:00:44 2021 daemon.notice netifd: Interface 'wan6' is now up
Sun Aug 15 22:00:44 2021 daemon.err insmod: module is already loaded - slhc
Sun Aug 15 22:00:44 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Aug 15 22:00:44 2021 daemon.err insmod: module is already loaded - pppox
Sun Aug 15 22:00:44 2021 daemon.err insmod: module is already loaded - pppoe
Sun Aug 15 22:00:44 2021 daemon.notice netifd: wan (3948): uci: Entry not found
Sun Aug 15 22:00:44 2021 daemon.notice netifd: wan (3948): sh: out of range
Sun Aug 15 22:00:44 2021 daemon.info pppd[3958]: Plugin rp-pppoe.so loaded.
Sun Aug 15 22:00:44 2021 daemon.info pppd[3958]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
[... cut ...]
Sun Aug 22 22:34:33 2021 daemon.notice procd: /etc/rc.d/S90ipsec: Command failed: Not found
Sun Aug 22 22:34:33 2021 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Sun Aug 22 22:34:33 2021 daemon.notice procd: /etc/rc.d/S90turboacc: net.ipv4.tcp_congestion_control = bbr
Sun Aug 22 22:34:33 2021 kern.alert kernel: [   11.333800] fast-classifier (PBR safe v2.1.4a): starting up
Sun Aug 22 22:34:33 2021 kern.alert kernel: [   11.334265] fast-classifier: registered
```
因为前几天不在家，所以日志保留的比较久，路由器死机是19日晚间死掉的，日志只记录到16日，22日回家后重启才有的新日志。

zhangguanzhang, 2021-09-05:
看样子kernal panic了

coolsnowwolf, 2026-09-06:
ESXi 中突然死机的报告没有串口、内核日志、宿主机信息或最小复现配置，无法判断是 x86 镜像、虚拟网卡还是宿主机问题，关闭为 not planned。

closed 2026-09-06 by coolsnowwolf