coolsnowwolf/lede #7573: Beta Test: R3P(MT7621, wifi:MT7615D) 测试5.10 内核编译mtwifi报错

xylz0928, 2021-07-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：5.10 内核下，选择mtwifi，勾选MT7615D给R3P，目前还存在问题，如下日志，编译报错。
先前还有“5.10内核下，使用mtwifi后，R3P无法更新，sysupgrade后重启，仍然回到当前固件，无法更新上去”
还有“5.10内核下，使用mtwifi，出厂设置下，无法打开无线开关”
这些功能。

### (2) 路由器型号和固件版本

A：xiaomi router-3-pro, MT7621, OP Kernel: 5.10

### (3) 详细日志

A：
make[5]: *** [/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_profile.o] Error 1
scripts/Makefile.build:280: recipe for target '/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap/../mt_wifi/os/linux/rt_profile.o' failed
make[4]: *** [/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap] Error 2
Makefile:1821: recipe for target '/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/mt_wifi_ap' failed
make[4]: Leaving directory '/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.51'
Makefile:251: recipe for target '/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/.built' failed
make[3]: *** [/workdir/***/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt7615d-5.0.4.0/.built] Error 2
make[3]: Leaving directory '/workdir/***/package/lean/mt/drivers/mt7615d'
time: package/lean/mt/drivers/mt7615d/compile#1.81#0.57#2.25
    ERROR: package/lean/mt/drivers/mt7615d failed to build.

xylz0928, 2021-07-31:
另外7月16日版本源码，5.10.50内核，闭源驱动，R3P是可用的

AmadeusGhost, 2021-07-31:
建议不用 5.10，现在除了更多的性能损失之外没看到有什么要搞 5.10 的必要。

xylz0928, 2021-07-31:
> 建议不用 5.10，现在除了更多的性能损失之外没看到有什么要搞 5.10 的必要。

嗯，我也是说了beta测试，还是希望新内核能兼容是最好吧

Neverends4, 2021-08-01:
但是我看官网说MT76的HWNAT目前不是只在5.10上初步实现了？现在5.4的HWNAT有看到CPU soft-irq占用，3月份前的源码没有这个问题，然而3月份之前flow offload有间歇性重启……

AmadeusGhost, 2021-08-01:
> 但是我看官网说MT76的HWNAT目前不是只在5.10上初步实现了？现在5.4的HWNAT有看到CPU soft-irq占用，3月份前的源码没有这个问题，然而3月份之前flow offload有间歇性重启……

有链接嘛，之前只看到 mt7622 支持了无线的 hwnat

Neverends4, 2021-08-01:
> 
> 有链接嘛，之前只看到 mt7622 支持了无线的 hwnat

[op官方坛子21.02rc2](https://forum.openwrt.org/t/openwrt-21-02-0-second-release-candidate/98026/125)
有人也提到了flowoffloading开了占用CPU，然后有人回复 Felix Fietkau提到要等5.10 内核

TXH201622, 2021-08-02:
> > 有链接嘛，之前只看到 mt7622 支持了无线的 hwnat
> 
> [op官方坛子21.02rc2](https://forum.openwrt.org/t/openwrt-21-02-0-second-release-candidate/98026/125)
> 有人也提到了flowoffloading开了占用CPU，然后有人回复 Felix Fietkau提到要等5.10 内核

OP master源码手动切到5.4内核，有线hwnat工作完美。你自己的问题。

xylz0928, 2021-08-04:
最新源码切到5.10编译mt7621又没问题了。
但是没看到有新的commit相关这个问题。
有点奇怪
不过能用了就行。
我关单了

closed 2021-08-04 by xylz0928