coolsnowwolf/lede #7575: 路由器无法使用web模式升级固件

pk49800, 2021-08-01:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用web界面升级路由器固件，上传固件后点击提交时自动退回到登录界面。

### (2) 路由器型号和固件版本

主机型号 | Netgear Nighthawk X4S R7800 (CpuMark : 13375.384732 Scores)
固件版本 | OpenWrt R21.7.1.100 / LuCI Master (git-21.188.53404-317d4ad)
内核版本 | 5.4.128

### (3) 详细日志

A：
Sun Aug  1 06:02:46 2021 daemon.notice pppd[9542]: pppd 2.4.8 started by root, uid 0
Sun Aug  1 06:02:59 2021 cron.err crond[10064]: crond (busybox 1.33.1) started, log level 8
Sun Aug  1 06:02:59 2021 cron.err crond[10289]: crond (busybox 1.33.1) started, log level 8
Sun Aug  1 06:03:01 2021 daemon.warn pppd[9542]: Timeout waiting for PADO packets
Sun Aug  1 06:03:01 2021 daemon.err pppd[9542]: Unable to complete PPPoE Discovery
Sun Aug  1 06:03:01 2021 daemon.info pppd[9542]: Exit.
Sun Aug  1 06:03:01 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Aug  1 06:03:01 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Aug  1 06:03:01 2021 daemon.err insmod: module is already loaded - slhc
Sun Aug  1 06:03:01 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Aug  1 06:03:01 2021 daemon.err insmod: module is already loaded - pppox
Sun Aug  1 06:03:01 2021 daemon.err insmod: module is already loaded - pppoe
Sun Aug  1 06:03:01 2021 daemon.notice netifd: wan (10551): uci: Entry not found
Sun Aug  1 06:03:01 2021 daemon.notice netifd: wan (10551): sh: out of range
Sun Aug  1 06:03:01 2021 daemon.info pppd[10584]: Plugin rp-pppoe.so loaded.
Sun Aug  1 06:03:01 2021 daemon.info pppd[10584]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Aug  1 06:03:01 2021 daemon.notice pppd[10584]: pppd 2.4.8 started by root, uid 0
Sun Aug  1 06:03:16 2021 daemon.warn pppd[10584]: Timeout waiting for PADO packets
Sun Aug  1 06:03:16 2021 daemon.err pppd[10584]: Unable to complete PPPoE Discovery
Sun Aug  1 06:03:16 2021 daemon.info pppd[10584]: Exit.
Sun Aug  1 06:03:16 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Aug  1 06:03:16 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Aug  1 06:03:16 2021 daemon.err insmod: module is already loaded - slhc
Sun Aug  1 06:03:16 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Aug  1 06:03:16 2021 daemon.err insmod: module is already loaded - pppox
Sun Aug  1 06:03:16 2021 daemon.err insmod: module is already loaded - pppoe
Sun Aug  1 06:03:16 2021 daemon.notice netifd: wan (10986): uci: Entry not found
Sun Aug  1 06:03:16 2021 daemon.notice netifd: wan (10986): sh: out of range
Sun Aug  1 06:03:16 2021 daemon.info pppd[11017]: Plugin rp-pppoe.so loaded.
Sun Aug  1 06:03:16 2021 daemon.info pppd[11017]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Aug  1 06:03:16 2021 daemon.notice pppd[11017]: pppd 2.4.8 started by root, uid 0
Sun Aug  1 06:03:31 2021 daemon.warn pppd[11017]: Timeout waiting for PADO packets
Sun Aug  1 06:03:31 2021 daemon.err pppd[11017]: Unable to complete PPPoE Discovery
Sun Aug  1 06:03:31 2021 daemon.info pppd[11017]: Exit.
Sun Aug  1 06:03:31 2021 daemon.notice netifd: Interface 'wan' is now down
Sun Aug  1 06:03:31 2021 daemon.notice netifd: Interface 'wan' is setting up now
Sun Aug  1 06:03:31 2021 daemon.err insmod: module is already loaded - slhc
Sun Aug  1 06:03:31 2021 daemon.err insmod: module is already loaded - ppp_generic
Sun Aug  1 06:03:31 2021 daemon.err insmod: module is already loaded - pppox
Sun Aug  1 06:03:31 2021 daemon.err insmod: module is already loaded - pppoe
Sun Aug  1 06:03:31 2021 daemon.notice netifd: wan (11416): uci: Entry not found
Sun Aug  1 06:03:31 2021 daemon.notice netifd: wan (11416): sh: out of range
Sun Aug  1 06:03:31 2021 daemon.info pppd[11448]: Plugin rp-pppoe.so loaded.
Sun Aug  1 06:03:31 2021 daemon.info pppd[11448]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Sun Aug  1 06:03:32 2021 daemon.notice pppd[11448]: pppd 2.4.8 started by root, uid 0
Sun Aug  1 06:03:35 2021 cron.err crond[11901]: crond (busybox 1.33.1) started, log level 8
Sun Aug  1 06:03:35 2021 cron.err crond[12119]: crond (busybox 1.33.1) started, log level 8
Sun Aug  1 06:03:47 2021 daemon.warn pppd[11448]: Timeout waiting for PADO packets
Sun Aug  1 06:03:47 2021 daemon.err pppd[11448]: Unable to complete PPPoE Discovery
Sun Aug  1 06:03:47 2021 daemon.info pppd[11448]: Exit.
[... 539 lines cut ...]
Sun Aug  1 11:27:19 2021 kern.err kernel: [ 6263.317338] 	00000020: c3 81 8c 88 63 8e 13 00 44 38 22 5c 00 56 00 52 00 0c 25 d4 a8 41 30 0a 66 c9 61 ad 2d 08 0a c7
Sun Aug  1 11:27:19 2021 kern.err kernel: [ 6263.328094] 	00000040: 08 27 bb 19 5b 54 9c 3b d9 32 b9 a2 91 ec d7 67 b0 46 d4 92 71 cb fa c8 12 29 94 cf 8b bd 55 d1
Sun Aug  1 11:27:19 2021 kern.err kernel: [ 6263.338781] 	00000060: 9e 49 f2 93 ac a4 1b 8a a2 83 f6 bc 2c 8b 1a 4a df c0 72 0f 00 71 38 20 02 0a 27 38 c1 0d 0c 85
Sun Aug  1 11:27:19 2021 kern.err kernel: [ 6263.349430] 	00000080: 81 a4 e7 0a a2 68 3b 1b 38 58 c8 16 72 77 8d 4a 53 d9 be 2c cb 59 2f 6a 12 97 b3 46 f8 f2 cc c8
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.360116] 	000000a0: 33 49 cf 4f 25 17 20 8e d2 82 6d f2 4d 82 c2 d5 2b 1f 0a 0a c7 93 6c 21 6f aa 4b 43 1d f2 85 4a
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.370799] 	000000c0: 25 63 7d de fd 39 94 ab 66 a8 b3 fb 76 68 05 0c 2e 88 94 b2 b8 3a 44 1d 72 46 b8 e2 61 12 ae 09
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.381485] 	000000e0: 95 88 e1 a9 2d 74 ed 71 f8 c2 76 68 93 24 1d 07 94 26 51 9a 85 ad f2 58 4c 87 ab 79 f7 70 dd d3
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.392135] 	00000100: 95 9b 13 29 ce 3a e4 4c 48 4c e9 bb f7 93 e9 54 33 92 21 8f e7 12 c9 8c e7 dc 59 e7 71 78 ee fd
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.402819] 	00000120: 7a e9 f2 37 95 c1 58 ab 73 de bd 6f b2 ce 9f ec 6b 36 88 8f dc 4f 92 b7 e7 21 2f 64 da c8 9b 0a
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.413505] 	00000140: 57 17 6a d4 21 07 15 e7 e8 d4 c2 56 d8 dc ce 79 4e 7d f6 08 a6 f8 b8 d2 ec 3b 8a 36 91 75 cb 97
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.424190] 	00000160: 8b 2f cb 92 54 31 8e 18 7c a8 ec e7 de a1 b2 91 73 c6 40 4c 20 4a a3 49 17 05 79 1a 4d 86 6e 77
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.434872] 	00000180: 2b 4c 74 36 0f 05 b9 61 03 80 89 a8 61 91 63 4a a1 1a 40 1c 04 0d 20 82 20 55 15 f1 12 e8 61 0c
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.445525] 	000001a0: 06 01 00 00 40 10 c1 03 10 41 c4 92 18 86 61 cb 38 da 0e 29 06 3d 67 17 d4 2c 3a d1 fe b3 9b 3e
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.456214] 	000001c0: 2f 80 29 76 ad fe 1d 5d 9f d6 e7 c9 b3 f7 28 10 cc 77 72 a6 80 33 40 7b 3e 65 b5 84 32 79 70 10
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.466895] 	000001e0: 90 72 75 53 23 14 cb 6c 1d 4b ef 42 69 58 72 20 aa 42 8b 17 94 67 43 3a 80 d0 03 86 fd e9 c7 92
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.477578] 	00000200: d8 8f e3 61 18 cb b8 ab 9b be b0 df 31 a9 35 3d 93 9b 72 60 93 e7 8e 84 c9 bc a6 a3 27 b5 88 82
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.488228] 	00000220: 50 51 bc 80 52 60 53 3f 30 17 ff 0a ae 25 37 4d 8c 70 be b2 d8 8c 3a 50 b6 9e 6a d9 24 a0 18 7d
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.498915] 	00000240: a0 4c 18 64 ac 2d 61 44 37 78 04 5e 56 58 0b d7 ab 8c 30 dd 6a 35 84 5b 5e 48 fa e1 0f 10 fd be
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.509601] 	00000260: 37 5c 2b 23 b3 e2 f4 f0 a5 5c f7 d6 4d 28 e8 e2 42 2b 25 05 f8 52 dc 5c 6b a3 87 d3 1c da f3 2c
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.520283] 	00000280: b5 58 29 bb 2e fa 4b b9 89 03 1b c9 ea 83 d6 3f f5 1a f6 28 0e 43 8d f3 91 46 6a 49 d0 8d f5 c1
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.530966] 	000002a0: d0 1b e5 06 e8 48 39 cb 56 69 7c 43 c4 3e c6 53 11 80 0f ca 64 66 10 c0 6b 59 e9 d6 8b 48 6e b9
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.541619] 	000002c0: 24 41 25 86 fb 93 e5 79 a6 28 bc 1e af 38 cc b0 83 dc 4f b1 a9 7a 58 c1 12 a4 60 c9 2f 3e da a2
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.552304] 	000002e0: 00 30 62 a1 62 99 71 17 3d
Sun Aug  1 11:27:20 2021 kern.err kernel: [ 6263.562999] UBIFS error (ubi0:1 pid 30478): do_readpage: cannot read page 0 of inode 202, error -22
Sun Aug  1 11:27:22 2021 authpriv.info dropbear[30484]: Not backgrounding

randyho, 2021-08-01:
配置管理方式变成DSA了，要用命令刷

hcym, 2021-08-01:
这个晕，tftp应该可以把

AmadeusGhost, 2021-08-01:
> 配置管理方式变成DSA了，要用命令刷

？这个跟 DSA 没有关系，不要一有什么问题就觉得是 DSA 的锅

pk49800, 2021-08-02:
ssh用命令刷好了

closed 2021-08-02 by pk49800