coolsnowwolf/lede #7576: 如何编译指定版本的qbittorrent

yubanmeiqin9048, 2021-08-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：需要编译qbittorrent4.1.9.1的版本，但仅修改qbittorrent的Makefile版本号为4.1.9.1报错

### (2) 路由器型号和固件版本

A：arm64

### (3) 详细日志

A：

coolsnowwolf, 2021-08-01:
qbittorrent 对特定 qt5 , libqbittorrent , boost 版本都有特定要求，请认真修改代码支持

yubanmeiqin9048, 2021-08-02:
> qbittorrent 对特定 qt5 , libqbittorrent , boost 版本都有特定要求，请认真修改代码支持

感谢大雕回复，也就是说需要修改qttools，qtbase，rlibqbittorrent , boost 中Makefile的PKG_HASH和PKG_VERSION到对应的版本才行

yubanmeiqin9048, 2021-08-02:
> qbittorrent 对特定 qt5 , libqbittorrent , boost 版本都有特定要求，请认真修改代码支持

按照您的说法，我将BOOST 的版本修改为1.71.0 将qttools，qtbase文件删掉放入qt5版本5.12.2，OpenSSL版本改为1.1.1d这些修改都参考这次更新#2982结果还是报错

closed 2021-08-02 by yubanmeiqin9048