coolsnowwolf/lede #7582: rac2v1k  256M nand  7.31号的最新源码编译启动慢

zyh6111, 2021-08-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：[    1.402976] nand: device found, Manufacturer ID: 0xc2, Chip ID: 0xaa
[    1.403340] nand: Macronix MX30UF2G18AC
[    1.410030] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    1.413716] 21 fixed-partitions partitions found on MTD device qcom_nand.0
[    1.421209] Creating 21 MTD partitions on "qcom_nand.0":
[    1.427999] 0x000000000000-0x000000040000 : "0:SBL1"
[    1.434990] 0x000000040000-0x000000180000 : "0:MIBIB"
[    1.441528] 0x000000180000-0x0000002c0000 : "0:SBL2"
[    1.446287] 0x0000002c0000-0x000000540000 : "0:SBL3"
[    1.449105] random: fast init done
[    1.457588] 0x000000540000-0x000000660000 : "0:DDRCONFIG"
[    1.460447] 0x000000660000-0x000000780000 : "0:SSD"
[    1.464750] 0x000000780000-0x000000a00000 : "0:TZ"
[    1.471709] 0x000000a00000-0x000000c80000 : "0:RPM"
[    1.476621] 0x000000c80000-0x000001180000 : "0:APPSBL"
[    1.485608] 0x000001180000-0x000001200000 : "0:APPSBLENV"
[    1.487362] 0x000001200000-0x000001340000 : "0:ART"
[    1.493003] 0x000001340000-0x0000013a0000 : "0:BOOTCONFIG"
[    1.496213] 0x0000013a0000-0x0000014e0000 : "0:SBL2_1"
[    1.503216] 0x0000014e0000-0x000001760000 : "0:SBL3_1"
[    1.510421] 0x000001760000-0x000001880000 : "0:DDRCONFIG_1"
[    1.513243] 0x000001880000-0x0000019a0000 : "0:SSD_1"
[    1.518666] 0x0000019a0000-0x000001c20000 : "0:TZ_1"
[    1.526138] 0x000001c20000-0x000001ea0000 : "0:RPM_1"
[    1.531182] 0x000001ea0000-0x000001f00000 : "0:BOOTCONFIG1"
[    1.532737] 0x000001f00000-0x000002400000 : "0:APPSBL_1"
[    1.545548] 0x000002400000-0x00001c400000 : "ubi"
[    1.545582] mtd: partition "ubi" extends beyond the end of device "qcom_nand.0" -- size truncated to 0xdc00000
[    1.787533] random: crng init done

zyh6111, 2021-08-01:
还有一个512NAND的，希望大雕能看一下。
[    1.401645] nand: device found, Manufacturer ID: 0xc2, Chip ID: 0xac
[    1.401981] nand: Macronix MX30UF4G18AB
[    1.408687] nand: 512 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    1.412364] 21 fixed-partitions partitions found on MTD device qcom_nand.0
[    1.419867] Creating 21 MTD partitions on "qcom_nand.0":

AmadeusGhost, 2021-08-01:
256m 版本板子上写的是 rev 几？

zyh6111, 2021-08-01:
[image]
rev  9.3

> 256m 版本板子上写的是rev几？

RT4230W-D187(RoHS)A Rev 9.3

nanqiaobeian, 2021-08-01:
sac2v1k启动慢，希望大雕你能帮忙解决。
[    1.402027] nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
[    1.402384] nand: Winbond W29N02GZ
[    1.409058] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB size: 64
[    1.412409] 21 fixed-partitions partitions found on MTD device qcom_nand.0
[    1.419837] Creating 21 MTD partitions on "qcom_nand.0":

AmadeusGhost, 2021-08-01:
能麻烦进 256m 机子的 uboot 输 smeminfo 把结果发出来嘛，我打算 256m 的另外处理

lancewing, 2021-08-02:
我也有256m的版本，PCB上的版本一样，芯片一样，问题也一样，希望我的情况也可以作为参考，uboot命令smeminfo返回的信息：

```
U-Boot 1.0.6 [spf.5.0.2_csu1.1] (Jan 15 2020 - 03:36:44)

smem ram ptable found: ver: 0 len: 5
DRAM:  1003 MiB
setting 0x177e as machine type from smem
NAND:  SF: Unsupported manufacturer 00
ipq_spi: SPI Flash not found (bus/cs/speed/mode) = (0/0/48000000/0)
256 MiB
MMC:   
PCI0 Link Intialized
PCI1 Link Intialized
In:    serial
Out:   serial
Err:   serial
MMC Device 0 not found
cdp: get part failed for 0:HLOS
Net:   MAC0 addr:c8:b4:22:cb:57:53
athrs17_reg_init: complete
athrs17_vlan_config ...done
S17c init  done
MAC1 addr:c8:b4:22:cb:57:54
eth0, eth1
Hit space key to stop autoboot:  0 
(IPQ) # smeminfo
flash_type:             0x2
flash_index:            0x0
flash_chip_select:      0x0
flash_block_size:       0x20000
partition table offset 0x0
No.: Name             Attributes            Start             Size
  0: 0:SBL1           0x0000ffff              0x0          0x40000
  1: 0:MIBIB          0x0000ffff          0x40000         0x140000
  2: 0:SBL2           0x0000ffff         0x180000         0x140000
  3: 0:SBL3           0x0000ffff         0x2c0000         0x280000
  4: 0:DDRCONFIG      0x0000ffff         0x540000         0x120000
  5: 0:SSD            0x0000ffff         0x660000         0x120000
  6: 0:TZ             0x0000ffff         0x780000         0x280000
  7: 0:RPM            0x0000ffff         0xa00000         0x280000
  8: 0:APPSBL         0x0000ffff        0x1f00000         0x500000
  9: 0:APPSBLENV      0x0000ffff        0x1180000          0x80000
 10: 0:ART            0x0000ffff        0x1200000         0x140000
 11: 0:BOOTCONFIG     0x0000ffff        0x1340000          0x60000
 12: 0:SBL2_1         0x0000ffff        0x13a0000         0x140000
 13: 0:SBL3_1         0x0000ffff        0x14e0000         0x280000
 14: 0:DDRCONFIG_1    0x0000ffff        0x1760000         0x120000
 15: 0:SSD_1          0x0000ffff        0x1880000         0x120000
 16: 0:TZ_1           0x0000ffff        0x19a0000         0x280000
 17: 0:RPM_1          0x0000ffff        0x1c20000         0x280000
 18: 0:BOOTCONFIG1    0x0000ffff        0x1ea0000          0x60000
 19: 0:APPSBL_1       0x0000ffff         0xc80000         0x500000
 20: rootfs           0x0000ffff        0x8700000        0x6300000
 21: rootfs_1         0x0000ffff        0x2400000        0x6300000
 22: ubifs            0x0000ffff        0xea00000        0x1600000
(IPQ) #
```

nanqiaobeian, 2021-08-02:
> 
> 
> 能麻烦进 256m 机子的 uboot 输 smeminfo 把结果发出来嘛，我打算 256m 的另外处理

> 
> 
> 能麻烦进 256m 机子的 uboot 输 smeminfo 把结果发出来嘛，我打算 256m 的另外处理
nand型号:
[    1.402133] nand: device found, Manufacturer ID: 0xef, Chip ID: 0xaa
[    1.402480] nand: Winbond W29N02GZ
[    1.409186] nand: 256 MiB, SLC, erase size: 128 KiB, page size: 2048, OOB siz                                                                                                                                                             e: 64
[    1.412505] 21 fixed-partitions partitions found on MTD device qcom_nand.0

输入smeminfo 结果：
(IPQ) # smeminfo
flash_type:             0x2
flash_index:            0x0
flash_chip_select:      0x0
flash_block_size:       0x20000
partition table offset 0x0
No.: Name             Attributes            Start             Size
  0: 0:SBL1           0x0000ffff              0x0          0x40000
  1: 0:MIBIB          0x0000ffff          0x40000         0x140000
  2: 0:SBL2           0x0000ffff         0x180000         0x140000
  3: 0:SBL3           0x0000ffff         0x2c0000         0x280000
  4: 0:DDRCONFIG      0x0000ffff         0x540000         0x120000
  5: 0:SSD            0x0000ffff         0x660000         0x120000
  6: 0▒TZ             0x0000ffff         0x780000         0x280000
  7: 0:RPM            0x0000ffff         0xa00000         0x280000
  8: 0:APPSBL         0x0000ffff         0xc80000         0x500000
  9: 0:APPSBLENV      0x0000ffff        0x1180000          0x80000
 10: 0:ART      "     0x0000ffff        0x1200000         0x140000
 11: 0:BOOTCONFIG     0x0000ffff        0x1340000          0x60000
 12: 0:SBL2_1         0x0000ffff        0x13a0000         0x140000
 13: 0:SBL3_1         0x0000ffff        0x14e0000         0x280000
 14: 0:DDRCONFIG_1    0x0000ffff        0x1760000         0x120000
 15: 0:SSD_1          0x0000ffff        0x1880000         0x120000
 16: 0:TZ_1           0x0000ffff        0x19a0000         0x280000
 17: 0:RPM_1          0x0000ffff        0x1c20000         0x280000
 18: 0:BOOTCONFIG1    0x0000ffff        0x1ea0000          0x60000
 19: 0:APPSBL_1       0x▒000ffff        0x1f00000         0x500000
 20: rootfs           0x0000ffff        0x8700000        0x6300000
 21: rootfs_1`        0x0000ffff        0x2400000        0x6300000
 22: ubifs            0x0000ffff        0xea00000        0x1600000
(IPQ) #

> 
> 
> 能麻烦进 256m 机子的 uboot 输 smeminfo 把结果发出来嘛，我打算 256m 的另外处理

(IPQ) # smeminfo
flash_type:             0x2
flash_index:            0x0
flash_chip_select:      0x0
flash_block_size:       0x20000
partition table offset 0x0
No.: Name             Attributes            Start             Size
  0: 0:SBL1           0x0000ffff              0x0          0x40000
  1: 0:MIBIB          0x0000ffff          0x40000         0x140000
  2: 0:SBL2           0x0000ffff         0x180000         0x140000
  3: 0:SBL3           0x0000ffff         0x2c0000         0x280000
  4: 0:DDRCONFIG      0x0000ffff         0x540000         0x120000
  5: 0:SSD            0x0000ffff         0x660000         0x120000
  6: 0▒TZ             0x0000ffff         0x780000         0x280000
  7: 0:RPM            0x0000ffff         0xa00000         0x280000
  8: 0:APPSBL         0x0000ffff         0xc80000         0x500000
  9: 0:APPSBLENV      0x0000ffff        0x1180000          0x80000
 10: 0:ART      "     0x0000ffff        0x1200000         0x140000
 11: 0:BOOTCONFIG     0x0000ffff        0x1340000          0x60000
 12: 0:SBL2_1         0x0000ffff        0x13a0000         0x140000
 13: 0:SBL3_1         0x0000ffff        0x14e0000         0x280000
 14: 0:DDRCONFIG_1    0x0000ffff        0x1760000         0x120000
 15: 0:SSD_1          0x0000ffff        0x1880000         0x120000
 16: 0:TZ_1           0x0000ffff        0x19a0000         0x280000
 17: 0:RPM_1          0x0000ffff        0x1c20000         0x280000
 18: 0:BOOTCONFIG1    0x0000ffff        0x1ea0000          0x60000
 19: 0:APPSBL_1       0x▒000ffff        0x1f00000         0x500000
 20: rootfs           0x0000ffff        0x8700000        0x6300000
 21: rootfs_1`        0x0000ffff        0x2400000        0x6300000
 22: ubifs            0x0000ffff        0xea00000        0x1600000
(IPQ) #

AmadeusGhost, 2021-08-03:
试试这样：https://github.com/AmadeusGhost/lede/commit/1845163a052efac124f00656eb72f38947630a42
记得先进 uboot 改下 env:
```
setenv bootcmd "setenv mtdids nand0=nand0 && set mtdparts mtdparts=nand0:0xDC00000@0x2400000(firmware) && ubi part firmware && ubi read 0x44000000 kernel 0x6e0000 && bootm"
saveenv
```
然后刷上编译出来的 Rev 9.3 initramfs-uImage 固件再 sysupgrade 看看

Trbe, 2021-08-04:
> 试试这样：[AmadeusGhost@1845163](https://github.com/AmadeusGhost/lede/commit/1845163a052efac124f00656eb72f38947630a42)
> 记得先进 uboot 改下 env:
> 
> ```
> setenv bootcmd "setenv mtdids nand0=nand0 && set mtdparts mtdparts=nand0:0xDC00000@0x2400000(firmware) && ubi part firmware && ubi read 0x44000000 kernel 0x6e0000 && bootm"
> saveenv
> ```
> 
> 然后刷上编译出来的 Rev 9.3 initramfs-uImage 固件再 sysupgrade 看看

刷完断电后启动还是很长时间,而且有个问题,在ssh不能使用fw_printenv
```
root@OpenWrt:~# fw_printenv
Cannot parse config file '/etc/fw_env.config': No such file or directory
```

内核日志
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (mori@ubuntu) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3606-1845163a0)) #0 SMP Tue Aug 3 03:53:51 2021
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV9.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line not present
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996168K/1015808K available (6215K kernel code, 198K rwdata, 1556K rodata, 1024K init, 232K bss, 19640K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000006] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000275] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000297] pid_max: default: 32768 minimum: 301
[    0.000511] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000534] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001602] CPU: Testing write buffer coherency: ok
[    0.002769] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002948] rcu: Hierarchical SRCU implementation.
[    0.003448] smp: Bringing up secondary CPUs ...
[    0.005370] smp: Brought up 1 node, 2 CPUs
[    0.005387] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005397] CPU: All CPU(s) started in SVC mode.
[    0.016648] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.016794] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016822] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.016966] pinctrl core: initialized pinctrl subsystem
[    0.018110] NET: Registered protocol family 16
[    0.018432] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.019791] cpuidle: using governor ladder
[    0.019859] cpuidle: using governor menu
[    0.055541] usbcore: registered new interface driver usbfs
[    0.055625] usbcore: registered new interface driver hub
[    0.055697] usbcore: registered new device driver usb
[... 1260 lines cut ...]
[  317.245171] ath10k_pci 0001:01:00.0: rts threshold -1
[  317.245703] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  317.250853] debugfs: File 'virt_if' in directory 'stats' already present!
[  317.261003] wlan0: Created a NSS virtual interface.
[  317.267500] wlan0: RX-Q and tasklet initialized.
[  317.279708] br-lan: port 2(wlan1) entered blocking state
[  317.279757] br-lan: port 2(wlan1) entered disabled state
[  317.284628] device wlan1 entered promiscuous mode
[  317.301576] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[  317.591466] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  317.591910] br-lan: port 2(wlan1) entered blocking state
[  317.596919] br-lan: port 2(wlan1) entered forwarding state
[  324.333969] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  324.334001] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[  324.415954] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  324.416827] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13156 sram: 11224'
[... cut ...]  -1
[  324.798413] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  324.811292] br-lan: port 3(wlan0) entered blocking state
[  324.812708] br-lan: port 3(wlan0) entered disabled state
[  324.818672] device wlan0 entered promiscuous mode
[  325.444991] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  325.445425] br-lan: port 3(wlan0) entered blocking state
[  325.450575] br-lan: port 3(wlan0) entered forwarding state
```

asushugo, 2021-08-05:
> ssh不能使用fw_printenv

这个问题可能恢复出厂设置就可以用了，试试看

lancewing, 2021-08-05:
大雕在群里说了是block-mount的兼容性问题，不编译或者删除即可。编译时不可取消选中，所以直接移除block-mount，移除后，启动慢的问题解决了，附上系统日志
<details>	
<summary>系统日志</summary>	
<pre><code class="language-shell">
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Tue Aug  3 18:45:31 2021 kern.notice kernel: [    0.000000] Linux version 5.4.132 (lance@deepin) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3606-f5c0cee30)) #0 SMP Tue Aug 3 10:30:35 2021
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] CPU: div instructions available: patching division code
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Memory policy: Data cache writealloc
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 253952
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 1728 pages used for memmap
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 196608 pages, LIFO batch:63
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
Tue Aug  3 18:45:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
Tue Aug  3 18:45:31 2021 kern.notice kernel: [    0.000000] Kernel command line: rootfstype=squashfs noinitrd
Tue Aug  3 18:45:31 2021 kern.notice kernel: [    0.000000] Bootloader command line not present
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] Memory: 996172K/1015808K available (6214K kernel code, 198K rwdata, 1556K rodata, 1024K init, 232K bss, 19636K reserved, 0K cma-reserved, 229376K highmem)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Tue Aug  3 18:45:31 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000020] Switching to timer-based delay loop, resolution 160ns
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000276] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000296] pid_max: default: 32768 minimum: 301
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000509] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.000530] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.001611] CPU: Testing write buffer coherency: ok
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.002766] Setting up static identity map for 0x42300000 - 0x42300060
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.002947] rcu: Hierarchical SRCU implementation.
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.003441] smp: Bringing up secondary CPUs ...
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.005367] smp: Brought up 1 node, 2 CPUs
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.005385] SMP: Total of 2 processors activated (25.00 BogoMIPS).
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.005396] CPU: All CPU(s) started in SVC mode.
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.017492] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.017644] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.017673] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.017821] pinctrl core: initialized pinctrl subsystem
Tue Aug  3 18:45:31 2021 kern.info kernel: [    0.018989] NET: Registered protocol family 16
[... cut ...]
Tue Aug  3 18:45:59 2021 daemon.info procd: - init complete -
Tue Aug  3 18:45:59 2021 user.info wsdd2[3696]: starting.
Tue Aug  3 18:45:59 2021 user.err wsdd2[3696]: error: wsdd-mcast-v4: wsd_send_soap_msg: send: Network unreachable
Tue Aug  3 18:46:14 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Tue Aug  3 18:46:29 2021 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0.2)
Tue Aug  3 18:46:30 2021 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan6 (eth0.2)
</code></pre>	
</details>

Trbe, 2021-08-05:
> > ssh不能使用fw_printenv
> 
> 这个问题可能恢复出厂设置就可以用了，试试看

试了,还是不能用

zyh6111, 2021-08-06:
最新状况：因本人没有ttl未使用AmadeusGhost大佬的方法测试，7月6号下午大雕更新block-mount后编译出来的block-mount版本为2021-08-05-d4f01298-22，刷机后第一次不保留配置启动很快，配置完重启后依然很慢，去web页面软件包里卸载block-mount，在安装软件包里的2021-08-05-d4f01298-1版本，就没有启动慢的问题了。

rshi0212, 2022-02-20:
为啥启动一次要那么久？

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.175 (rshi0212@GPDWIN3-RSHI) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r4208-a45b0fa8a)) #0 SMP Sun Feb 20 04:00:11 2022
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 253952
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 57344 pages, LIFO batch:15
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] pcpu-alloc: s30220 r8192 d23028 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line (ignored): 
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996168K/1015808K available (6265K kernel code, 198K rwdata, 1568K rodata, 1024K init, 232K bss, 19640K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000007] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000021] Switching to timer-based delay loop, resolution 160ns
[    0.000263] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000283] pid_max: default: 32768 minimum: 301
[    0.000490] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000512] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001586] CPU: Testing write buffer coherency: ok
[    0.002730] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002912] rcu: Hierarchical SRCU implementation.
[    0.003411] smp: Bringing up secondary CPUs ...
[    0.005338] smp: Brought up 1 node, 2 CPUs
[    0.005355] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005366] CPU: All CPU(s) started in SVC mode.
[    0.017360] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.017516] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017545] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.017689] pinctrl core: initialized pinctrl subsystem
[    0.018851] NET: Registered protocol family 16
[    0.019177] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.020560] cpuidle: using governor ladder
[    0.020740] cpuidle: using governor menu
[    0.056700] usbcore: registered new interface driver usbfs
[    0.056784] usbcore: registered new interface driver hub
[    0.056857] usbcore: registered new device driver usb
[    0.056921] pps_core: LinuxPPS API ver. 1 registered
[    0.056935] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[    0.056970] PTP clock support registered
[    0.057811] workqueue: max_active 576 requested for napi_workq is out of range, clamping between 1 and 512
[    0.058979] clocksource: Switched to clocksource dg_timer
[    0.059868] thermal_sys: Registered thermal governor 'step_wise'
[    0.062538] NET: Registered protocol family 2
[    0.062724] IP idents hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.063862] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.063905] TCP established hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.063964] TCP bind hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.064058] TCP: Hash tables configured (established 8192 bind 8192)
[    0.064175] UDP hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064213] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.064450] NET: Registered protocol family 1
[    0.064502] PCI: CLS 0 bytes, default 64
[... 522 lines cut ...]
[  408.697696] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  408.698026] br-lan: port 2(wlan0) entered blocking state
[  408.703223] br-lan: port 2(wlan0) entered forwarding state
[  438.402177] device wlan1 left promiscuous mode
[  438.402410] br-lan: port 3(wlan1) entered disabled state
[  438.591162] ath10k_pci 0001:01:00.0: could not get mac80211 beacon, vdev_id: 0 addr: c8:b4:22:8e:b2:cb
[  438.594446] ath10k_pci 0001:01:00.0: mac flush null vif, drop 0 queues 0xffff
[  438.600316] ath10k_pci 0001:01:00.0: peer-unmap-event: unknown peer id 0
[... cut ...]
[  446.086165] ath10k_pci 0001:01:00.0: rts threshold -1
[  446.086518] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[  446.097886] br-lan: port 3(wlan1) entered blocking state
[  446.101254] br-lan: port 3(wlan1) entered disabled state
[  446.107013] device wlan1 entered promiscuous mode
[  448.259540] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[  448.259888] br-lan: port 3(wlan1) entered blocking state
[  448.264983] br-lan: port 3(wlan1) entered forwarding state

coolsnowwolf, 2026-09-06:
RAC2V1K 启动慢日志没有设备型号、完整启动时间线或当前版本，无法判断 NAND/驱动问题，关闭为“不计划处理”。

closed 2026-09-06 by coolsnowwolf