coolsnowwolf/lede #7584: 关于5.10内核使用X550T2万兆网卡的问题

sniper3209, 2021-08-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：自更新5.10内核后，intel X550T2万兆网卡无法正常握手2.5G速率，始终保持1G状态
B：期间更换基于大神源码的不同固件版本，情况相同

### (2) 路由器型号和固件版本

A：路由器型号：M720Q 
      固件版本：OpenWrt Ipv6-Mini-S2021.08.01 by Sirpdboy / LuCI Master (git-21.211.30217-03784c3)
      内核版本5.10.51

### (3) 详细日志

A：Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.264677] hid-generic 0003:046D:C534.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:00:14.0-7/input0
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.268088] input: Logitech USB Receiver Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:046D:C534.0002/input/input5
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.269522] input: Logitech USB Receiver Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:046D:C534.0002/input/input6
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.334683] input: Logitech USB Receiver System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:046D:C534.0002/input/input7
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.336191] hid-generic 0003:046D:C534.0002: input,hiddev96,hidraw1: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:00:14.0-7/input1
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.344765] usb 2-4: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
Sun Aug  1 21:45:48 2021 user.info kernel: [    2.800660] kmodloader: loading kernel modules from /etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    2.843962] Key type encrypted registered
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.846525] pps_core: LinuxPPS API ver. 1 registered
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.847572] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.850439] PTP clock support registered
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.852305] Button Hotplug driver version 0.4.1
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.853821] ANTFS Module: Version 07.19-b41ba529
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.855224] raid6: skip pq benchmark and using algorithm avx2x4
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.856347] raid6: using avx2x2 recovery algorithm
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.857701] xor: automatically using best checksumming function   avx
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.861963] Btrfs loaded, crc32c=crc32c-generic
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.863858] ehci-fsl: Freescale EHCI Host controller driver
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.865424] ehci-platform: EHCI generic platform driver
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.869481] sdhci-pltfm: SDHCI platform and OF driver helper
Sun Aug  1 21:45:48 2021 kern.info kernel: [    2.870767] usbcore: registered new interface driver uas
Sun Aug  1 21:45:48 2021 user.info kernel: [    2.871897] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 user.info kernel: [    2.874068] init: - preinit -
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    3.121935] random: jshn: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    3.151564] random: jshn: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    3.153818] random: jshn: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 kern.warn kernel: [    7.183433] urandom_read: 4 callbacks suppressed
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    7.183436] random: jshn: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.261944] mount_root: loading kmods from internal overlay
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.268810] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.270703] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    7.377396] random: procd: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    7.414852] F2FS-fs (loop0): Mounted with checkpoint version = 73eb8edc
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.416051] block: attempting to load /tmp/f2fs_cfg/upper/etc/config/fstab
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.417599] block: extroot: not configured
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    7.548514] F2FS-fs (loop0): Mounted with checkpoint version = 73eb8ede
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.552406] mount_root: loading kmods from internal overlay
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.580272] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.584139] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.596616] block: attempting to load /tmp/f2fs_cfg/upper/etc/config/fstab
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.598038] block: extroot: not configured
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.599285] mount_root: switching to f2fs overlay
Sun Aug  1 21:45:48 2021 kern.info kernel: [    7.601205] overlayfs: "xino" feature enabled using 32 upper inode bits.
Sun Aug  1 21:45:48 2021 kern.info kernel: [    7.631486] EXT4-fs (nvme0n1p1): mounted filesystem without journal. Opts: (null)
Sun Aug  1 21:45:48 2021 user.warn kernel: [    7.633834] urandom-seed: Seeding with /etc/urandom.seed
Sun Aug  1 21:45:48 2021 user.info kernel: [    7.771656] procd: - early -
Sun Aug  1 21:45:48 2021 kern.notice kernel: [    7.915486] random: jshn: uninitialized urandom read (4 bytes read)
Sun Aug  1 21:45:48 2021 user.info kernel: [    8.329306] procd: - ubus -
[... 583 lines cut ...]
Sun Aug  1 21:47:10 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Aug  1 21:47:13 2021 daemon.notice procd: /etc/rc.d/S99openclash: OpenClash Already Start
Sun Aug  1 21:47:13 2021 daemon.notice procd: /etc/rc.d/S99serverchan: serverchan is starting now ...
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]: ttyd 1.6.3 (libwebsockets 4.1.6-ff04283)
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]: tty configuration:
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:   start command: /bin/login
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:   close signal: SIGHUP (1)
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:   terminal type: xterm-256color
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]: LWS: 4.1.6-ff04283, loglevel 7
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]: NET CLI SRV H1 H2 WS IPV6-off
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:    /usr/lib/libwebsockets-evlib_uv.so
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:  Using foreign event loop...
Sun Aug  1 21:47:13 2021 daemon.notice ttyd[17870]:  Listening on port: 7681
Sun Aug  1 21:47:13 2021 user.info wsdd2[18116]: starting.
Sun Aug  1 21:47:13 2021 daemon.info procd: - init complete -
Sun Aug  1 21:47:13 2021 cron.err crond[18242]: crond (busybox 1.33.1) started, log level 8
Sun Aug  1 21:47:18 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Aug  1 21:47:18 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Aug  1 21:47:18 2021 daemon.info dnsmasq[18485]: Connected to system UBus
Sun Aug  1 21:47:27 2021 daemon.info procd: Instance haproxy::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
Sun Aug  1 21:48:27 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Aug  1 21:50:25 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Aug  1 21:52:28 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Aug  1 21:54:26 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)
Sun Aug  1 21:56:34 2021 daemon.err odhcp6c[10896]: Failed to send SOLICIT message to ff02::1:2 (Permission denied)

coolsnowwolf, 2021-08-01:
你可以切换到 5.4 内核测试嘛

BigDog66, 2021-08-03:
> 你切换到 5.4 可以运行测试

大佬 我也一样啊 5.4内核正常的

BigDog66, 2021-08-03:
和楼主一样的问题  https://github.com/coolsnowwolf/lede/issues/7401  这个是我发的

closed 2021-08-07 by coolsnowwolf

sniper3209, 2021-08-08:
> 你可以切换到 5.4 内核测试嘛
谢谢大神回复，我切回到5.4正常了

sniper3209, 2021-08-08:
> 你可以切换到 5.4 内核测试嘛
感谢大神，8.8日源代码更新后，X550T2正常识别！非常感谢