coolsnowwolf/lede #7588: 编译报错，请达人指点下

starambler, 2021-08-02:
编译环境github action  ubuntu20.02

报错如下：
Configuring kmod-fs-squashfs.
Configuring kmod-i2c-algo-bit.
Configuring kmod-igb.
Configuring kmod-ipt-offload.
Configuring kmod-bnx2.
Configuring vsftpd.
Configuring urngd.
Configuring libnet-1.2.x.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:109: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
make[1]: Leaving directory '/workdir/openwrt'
Error: Process completed with exit code 2.

EnnawYang, 2021-08-02:
请提供详细日志

starambler, 2021-08-02:
Installing vim-fuller (8.2-3) to root...
Installing vsftpd (3.0.3-3) to root...
Package wget-ssl (1.21.1-1) installed in root is up to date.
Package xray-core (1.4.2-1) installed in root is up to date.
Package xray-geodata (1.4.2-1) installed in root is up to date.
Package libyaml (0.2.5-1) installed in root is up to date.
Package zerotier (1.6.5-1) installed in root is up to date.
Configuring libgcc1.
Configuring libc.
Configuring terminfo.
Configuring kernel.
Configuring kmod-crypto-hash.
Configuring kmod-crypto-lib-blake2s.
Configuring kmod-crypto-lib-chacha20.
Configuring kmod-crypto-lib-poly1305.
Configuring kmod-crypto-lib-chacha20poly1305.
Configuring kmod-crypto-kpp.
Configuring kmod-crypto-lib-curve25519.
Configuring kmod-udptunnel4.
Configuring kmod-udptunnel6.
Configuring kmod-wireguard.
Configuring wireguard-tools.
Configuring luci-proto-wireguard.
Configuring libubox20210516.
Configuring libuci20130104.
Configuring liblua5.1.5.
Configuring libuci-lua.
Configuring luci-theme-material.
Configuring libopenssl1.1.
Configuring zlib.
Configuring openssh-sftp-server.
Configuring liblucihttp0.
Configuring libubus20210630.
Configuring cgi-io.
Configuring lua.
Configuring luci-lib-nixio.
Configuring libnl-tiny1.
Configuring luci-lib-ip.
Configuring libjson-c5.
Configuring luci-lib-jsonc.
Configuring liblucihttp-lua.
Configuring luci-lib-base.
Configuring libpthread.
Configuring librt.
Configuring libuuid1.
Configuring libuv1.
Configuring libmnl0.
Configuring netdata.
Configuring libf2fs6.
Configuring mkf2fs.
Configuring libuclient20201210.
Configuring uclient-fetch.
Configuring opkg.
Configuring libcap.
Configuring libwebsockets-full.
Configuring kmod-nf-reject.
Configuring kmod-nf-ipt.
Configuring kmod-ipt-core.
Configuring libxtables12.
Configuring libip4tc2.
Configuring libip6tc2.
Configuring iptables.
Configuring kmod-ipt-extra.
Configuring iptables-mod-extra.
Configuring kmod-nls-base.
Configuring kmod-usb-core.
Configuring luci-app-opkg.
Configuring libblobmsg-json20210516.
Configuring ubusd.
Configuring ubus.
[... 247 lines cut ...]
Configuring kmod-crypto-rng.
Configuring uhttpd-mod-ubus.
Configuring rpcd-mod-rrdns.
Configuring luci.
Configuring luci-i18n-opkg-zh-cn.
Configuring kmod-ipt-nat6.
Configuring kmod-e1000e.
Configuring libevent2-7.
Configuring kmod-button-hotplug.
Configuring kmod-fs-squashfs.
Configuring kmod-i2c-algo-bit.
Configuring kmod-igb.
Configuring kmod-ipt-offload.
Configuring kmod-bnx2.
Configuring vsftpd.
Configuring urngd.
Configuring libnet-1.2.x.
Configuring luci-i18n-firewall-zh-cn.
Configuring ppp-mod-pppoe.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [package/Makefile:69: package/install] Error 255
make[1]: *** [package/Makefile:109: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.

EnnawYang, 2021-08-02:
libustream-wolfssl与libustream-openssl冲突

starambler, 2021-08-02:
请问选哪一个？我在试试

EnnawYang, 2021-08-02:
libustream-openssl是ssrp的依赖，另一个我不知道是哪个的依赖

closed 2021-08-03 by starambler