coolsnowwolf/lede #7589: 求增加新机型-瑞斯康达msg1500，K2P同配置很便宜。

ldsj007, 2021-08-02:
优点就是路由很便宜
7621+7615+nand闪存+256M
有大神愿意搞我可以提供一台二手机器。

whfweb, 2021-08-02:
可以直接用K2P的固件吗

Lrvineshaw, 2021-08-02:
确实便宜，配置甚至比k2p还好一点

ldsj007, 2021-08-02:
> 可以直接用K2P的固件吗

ldsj007, 2021-08-03:
有大神愿意指导的话，我来做苦力。
网上搜到的是16年的教程，代码变化有点大，学不来。

coolsnowwolf, 2021-08-04:
便宜不如 100 元的 CR6608 了吧？ AX + 128 NAND + 256MB ，还是 新的/只拆封的

hzjnet, 2021-08-04:
Cr6608已经涨价了，而且是mtk芯片，驱动一直跟不上趟啊，主要还是大雕神没认真搞hwnat,效果就是带宽跑不满！

ldsj007, 2021-08-05:
> 便宜不如100元的CR6608吧？AX+128NAND+256MB，还是新的/只拆封的

大佬回复很荣幸。
CR6608淘宝130多元,比msg1500贵一些，但貌似CR6608也没有支持呢!

您有空适配吗？我寄过去两台机器，或者您没空的话介绍个别人对接这个事，我是分拣站，想推广下msg1500这个型号。

coolsnowwolf, 2021-08-05:
CR6606/6608/6609 统称 CR660x ，全支持了

Her0R0cket, 2021-08-06:
msg1500硬件规格上比k2p好,但是无线信号不行,我买了两台当主路由和交换机.本身是可以用hiwifi-hc5962和小米r3g的固件的

hzjnet, 2021-08-06:
> msg1500硬件规格上比k2p好,但是无线信号不行,我买了两台当主路由和交换机.本身是可以用hiwifi-hc5962的固件的,后来用的是padavan,这系统无线信号和稳定性比op好点,但我是一点都不习惯这个系统.

MTK的用padavan确实好些

coolbele, 2021-08-10:
dw33d,性价比不错.可以看这个.

hzjnet, 2021-08-11:
翻了翻 发现这个型号不同配置太多了 有7621+7615、7612、7603，甚至还有螃蟹芯片

Her0R0cket, 2021-08-13:
> 
> 
> 翻了翻 发现这个型号不同配置太多了 有7621+7615、7612、7603，甚至还有螃蟹芯片

对的,买的时候要看好版本,X.00 是7615

ldsj007, 2021-08-13:
> dw33d，不错，可以看这个。

dw33d这款用的是高通9558cpu，性能只有7621A的一半。

> 翻了翻发现这个型号不同配置了7621+7615、7612、7603，甚至还有饼干芯片
> 对的,买的时候看好版本,X.00 是7615

对的确要看好版本。

目前我还是没有发现别的带7615的其他便宜机型。

Hey-Leo, 2021-08-14:
> 便宜不如 100 元的 CR6608 了吧？ AX + 128 NAND + 256MB ，还是 新的/只拆封的

现在淘宝上CR6608已经160了...

coolbele, 2021-08-15:
> > dw33d，不错，可以看这个。
> 
> dw33d这款用的是高通9558cpu，性能只有7621A的一半。
> 
> > 翻了翻发现这个型号不同配置了7621+7615、7612、7603，甚至还有饼干芯片
> > 对的,买的时候看好版本,X.00 是7615
> 
> 对的确要看好版本。
> 
> 目前我还是没有发现别的带7615的其他便宜机型。
> 希望早日有大神适配。

高通信号不错了.带20~30台机器没问题.很稳定. mesh 无缝组网很舒服.

lunatickochiya, 2021-08-23:
这种nand的，不带spi，先破解了免拆再看吧

ldsj007, 2021-08-29:
> 这种nand的，不带spi，先破解了免拆再看吧

有免拆刷机教程
https://post.smzdm.com/p/aoozem39/

tlngln, 2021-09-02:
我可以做，联系61368554

sundog315, 2021-09-03:
同求

tlngln, 2021-09-03:
> 
> 
> > 便宜不如100元的CR6608吧？AX+128NAND+256MB，还是新的/只拆封的
> 
> 大佬回复很荣幸。
> CR6608淘宝130多元,比msg1500贵一些，但貌似CR6608也没有支持呢!
> 
> 您有空适配吗？我寄过去两台机器，或者您没空的话介绍个别人对接这个事，我是分拣站，想推广下msg1500这个型号。

我可以做

tlngln, 2021-09-03:
> 
> 
> > 这种nand的，不带spi，先破解了免拆再看吧
> 
> 有免拆刷机教程
> https://post.smzdm.com/p/aoozem39/

q61368554

ldsj007, 2021-09-03:
> q61368554
加你微了

sundog315, 2021-09-08:
跟新源码以后，好像已经添加了msg1500.x.00

closed 2021-09-16 by AmadeusGhost