coolsnowwolf/lede #7590: 请问...R4S 1G的版本 在target Profile里面怎么选?

showme2005, 2021-08-02:
请问...R4S 1G的版本 在target Profile里面怎么选?

liuyanantj, 2021-08-02:
同问啊  貌似新的里面只有4g版本的

minlang112, 2021-08-02:
本来R4S就是RMB收割机了，还买1G版本的……

liuyanantj, 2021-08-03:
> 本来R4S就是RMB收割机了，还买1G版本的……

很早买的 当时4G版本的普遍没货  而且做个软路由 1G妥妥的了 还指望着这个做all in one么 希望代码还是能支持下吧

liuyanantj, 2021-08-06:
还是希望大雕能增加下r4s  1g的支持  毕竟还有一帮子用户呢 万分感谢

liuyanantj, 2021-08-09:
大雕在热心群友的建议下  已经更新代码支持r4s 1g版本了  可以无脑刷了 貌似还调节了超频的吧 跑分从3.8w到5w了  耍起来呀

Jason6111, 2021-08-10:
> 请问...R4S 1G的版本 在target Profile里面怎么选?

选4g版应该就行

DanielChen008, 2021-08-13:
处理器一样，内存不一样，之前显示 XXX 4G，现在 L大已经把4G去掉了，只留了名字。
所以之前其实直接编译4G版本的就可以给 1G 的用，跟 X86 一样呀，那些工控机、网友自己用的旧平台啥的，内存容量五花八门，还不是只管编译 X86 即可。

coolsnowwolf, 2021-08-13:
楼上的，R4S 1G 和 4G 之前因为补丁问题，是不通刷的 现在更新支持了

showme2005, 2021-08-15:
> 楼上的，R4S 1G 和 4G 之前因为补丁问题，是不通刷的 现在更新支持了

 感谢大神。。。

DanielChen008, 2021-08-16:
> 楼上的，R4S 1G 和 4G 之前因为补丁问题，是不通刷的 现在更新支持了

L 大亲自纠错，学习了

closed 2021-08-23 by AmadeusGhost

adamgonglj, 2021-11-08:
> 楼上的，R4S 1G 和 4G 之前因为补丁问题，是不通刷的 现在更新支持了

感谢L大