coolsnowwolf/lede #7593: k2p使用最新代码编译后,使用ssrplus会出现重启等问题,可能是内存溢出?

wsspsb, 2021-08-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：打开ssrplus后出现oom(用的是xray1.4.2), 导致uhttpd长时间无法访问, 过一段时间会出现路由器重启,是内存溢出问题还是我内存不够?
因为我在2021年5月25日编译的没有这样的问题,勾选的同样的插件,xray在这期间也一直是1.4.2版本, 我固件在那时候开启ssrplus内存占用也不高,这是我在路由器web概览->内存页面看到的:
    可用数 | 36 MB / 119 MB (30%)
    -- | --
    已缓存 | 3 MB / 119 MB (2%)

### (2) 路由器型号和固件版本

A：斐讯k2p, 版本根据最新代码编译,比如May 31, 2021

### (3) 详细日志

A：
[    0.000000] Linux version 5.4.132 (spsb@spsb) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3418-4a06bf69e)) #0 SMP Sat Jul 31 09:09:57 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Phicomm K2P
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] percpu: Embedded 14 pages/cpu s26992 r8192 d22160 u57344
[    0.000000] pcpu-alloc: s26992 r8192 d22160 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00042a20
[    0.000000] Readback ErrCtl register=00042a20
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 120328K/131072K available (6124K kernel code, 257K rwdata, 1288K rodata, 1276K init, 238K bss, 10744K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x36c/0x584 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[    0.015497] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.087828] pid_max: default: 32768 minimum: 301
[    0.097164] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.111571] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.129826] rcu: Hierarchical SRCU implementation.
[... 1080 lines cut ...]
[  288.666694] [   6348]     0  6348      314       10    16384        0             0 ntpd
[  288.683695] [   6524]     0  6524      329       35    16384        0             0 pppd
[  288.700118] [   7290]     0  7290      325       22    16384        0             0 sh
[  288.716069] [   7819]     0  7819      341       39    20480        0             0 sh
[  288.731875] [   8031]     0  8031      325       21    16384        0             0 sh
[  288.747879] [   8081]     0  8081      600       95    20480        0             0 fw3
[  288.763885] [   8087]     0  8087      313        9    12288        0             0 sh
[  288.779857] [   8194]     0  8194   170913     9978    57344        0             0 v2ray
[  288.796400] [   8205]     0  8205      331       37    20480        0             0 pppd
[  288.812559] [   8516]     0  8516      268        8    16384        0             0 odhcp6c
[  288.829340] [   8900] 65534  8900      656       18    16384        0             0 pdnsd
[  288.845665] [   8933]     0  8933      355       47    20480        0             0 sh
[  288.861581] [   8982]     0  8982      313        7    12288        0             0 sleep
[  288.878014] [   9205]   453  9205     1079      751    24576        0             0 dnsmasq
[  288.894850] [   9263]     0  9263      340       36    20480        0             0 sh
[  288.910656] [   9377]     0  9377      313        9    16384        0             0 lock
[  288.926924] [   9472]     0  9472      381       46    20480        0             0 luci
[  288.943079] [   9497]     0  9497      526       12    16384        0             0 ip
[  288.958993] [   9523]     0  9523      313        7    16384        0             0 sleep
[  288.975318] [   9524]     0  9524      289        8    16384        0             0 luci
[  288.991627] [   9532]     0  9532      314        9    12288        0             0 sleep
[  289.008149] [   9533]     0  9533      313        7    16384        0             0 sleep
[  289.024571] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=v2ray,pid=8194,uid=0
[  289.049342] Out of memory: Killed process 8194 (v2ray) total-vm:683652kB, anon-rss:39912kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:56kB oom_score_adj:0
[  289.103441] oom_reaper: reaped process 8194 (v2ray), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB

AmadeusGhost, 2021-08-02:
放过 k2p 吧，它只是一个只有 128m 内存的 mips 路由…

minlang112, 2021-08-02:
Out of memory 你说的很多，放过K2P吧

Neverends4, 2021-08-02:
同样128内存的红米2100表示，如果不用ssr，只使用xray基本程序包和基本代理，可以跑起来(⊙ᗜ⊙)

wsspsb, 2021-08-02:
> 
> 
> 放过 k2p 吧，它只是一个只有 128m 内存的 mips 路由…

我也知道k2p性能挺拉的, 不过我想的是能将就用就将就用, 我主要用的也就是偶尔开下代理,平常用端口转发和ddns就行了,这个简单的功能k2p够了,屋里也就100m宽带.
主要是之前也不至于内存不够,你看我提到的之前版本,内存只占用了36M,120M的内存还剩那么多,会不会是哪里出问题了才导致新版内存不够用

lphgor, 2021-08-03:
我的k2p不知道为什么，切换到DSA架构以后的固件都无法启动，下载openwrt官方snapshot的固件也不行。
看到你们都能正常使用，难道是路由器体质问题？等改天用ttl看看日志。

NueXini, 2021-08-05:
spi-max-frequency = <80000000>;
改为
spi-max-frequency = <10000000>;

应该能解决问题

wsspsb, 2021-08-06:
看到内核版本降了,重新编译刷了一下,没有这个问题了

wsspsb, 2021-08-06:
还是大意了,再观察了一震后还是一样的oom

coolsnowwolf, 2021-08-06:
SPI 都死了，还不换等啥时候呢

closed 2021-08-06 by coolsnowwolf