coolsnowwolf/lede #7597: 系统日志里uhttpd相关信息刷屏，编译时未勾选uhttpd。

minfengyk, 2021-08-03:
## 1.关于你要提交的问题
系统日志里uhttpd相关信息刷屏，编译时未勾选uhttpd。

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

系统日志里以下信息刷屏，编译时未勾选uhttpd。openwrt能使用，暂未发现其它问题，请教大佬有什么影响吗？

### (2) 路由器型号和固件版本

主机型号 | Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz : 1 Core 2 Thread
CPU 信息 | 2400.000 MHz
固件版本 | OpenWrt R21.7.26 / LuCI Master (git-21.211.30217-03784c3)
内核版本 | 5.10.51

### (3) 详细日志

Tue Aug  3 11:01:29 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:29 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:29 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:34 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:39 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:44 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:49 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:54 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:01:59 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:02:04 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute call dispatcher target for entry '/admin/services/openclash/state'.
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: The called action terminated with an exception:
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: /usr/lib/lua/luci/controller/openclash.lua:260: bad argument #2 to 'format' (string expected, got nil)
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: stack traceback:
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: 	[C]: in function 'assert'
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
Tue Aug  3 11:02:09 2021 daemon.err uhttpd[3459]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

EnnawYang, 2021-08-03:
好家伙，openclash来这里问

minfengyk, 2021-08-03:
> 好家伙，openclash来这里问

小白，不懂代码，所以不知道什么原因引起的。确实装了openclash，但是并没有运行啊！

closed 2021-08-05 by coolsnowwolf