coolsnowwolf/lede #7598: ea4500编译失败

rshi0212, 2021-08-03:
../../deps/v8/src/objects/api-callbacks.tq ../../deps/v8/src/objects/arguments.tq ../../deps/v8/src/objects/cell.tq ../../deps/v8/src/objects/code.tq ../../deps/v8/src/objects/contexts.tq ../../deps/v8/src/objects/data-handler.tq ../../deps/v8/src/objects/debug-objects.tq ../../deps/v8/src/objects/descriptor-array.tq ../../deps/v8/src/objects/embedder-data-array.tq ../../deps/v8/src/objects/feedback-cell.tq ../../deps/v8/src/objects/feedback-vector.tq ../../deps/v8/src/objects/fixed-array.tq ../../deps/v8/src/objects/foreign.tq ../../deps/v8/src/objects/free-space.tq ../../deps/v8/src/objects/heap-number.tq ../../deps/v8/src/objects/heap-object.tq ../../deps/v8/src/objects/js-aggregate-error.tq ../../deps/v8/src/objects/js-array-buffer.tq ../../deps/v8/src/objects/js-array.tq ../../deps/v8/src/objects/js-collection-iterator.tq ../../deps/v8/src/objects/js-collection.tq ../../deps/v8/src/objects/js-generator.tq ../../deps/v8/src/objects/js-objects.tq ../../deps/v8/src/objects/js-promise.tq ../../deps/v8/src/objects/js-proxy.tq ../../deps/v8/src/objects/js-regexp-string-iterator.tq ../../deps/v8/src/objects/js-regexp.tq ../../deps/v8/src/objects/js-weak-refs.tq ../../deps/v8/src/objects/literal-objects.tq ../../deps/v8/src/objects/map.tq ../../deps/v8/src/objects/microtask.tq ../../deps/v8/src/objects/module.tq ../../deps/v8/src/objects/name.tq ../../deps/v8/src/objects/oddball.tq ../../deps/v8/src/objects/ordered-hash-table.tq ../../deps/v8/src/objects/primitive-heap-object.tq ../../deps/v8/src/objects/promise.tq ../../deps/v8/src/objects/property-array.tq ../../deps/v8/src/objects/property-cell.tq ../../deps/v8/src/objects/property-descriptor-object.tq ../../deps/v8/src/objects/prototype-info.tq ../../deps/v8/src/objects/regexp-match-info.tq ../../deps/v8/src/objects/scope-info.tq ../../deps/v8/src/objects/script.tq ../../deps/v8/src/objects/shared-function-info.tq ../../deps/v8/src/objects/source-text-module.tq ../../deps/v8/src/objects/stack-frame-info.tq ../../deps/v8/src/objects/string.tq ../../deps/v8/src/objects/struct.tq ../../deps/v8/src/objects/synthetic-module.tq ../../deps/v8/src/objects/template-objects.tq ../../deps/v8/src/objects/template.tq ../../deps/v8/src/wasm/wasm-objects.tq ../../deps/v8/test/torque/test-torque.tq ../../deps/v8/third_party/v8/builtins/array-sort.tq -o "/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/torque-output-root/torque-generated" -v8-root ../../deps/v8
  touch c071ca2dc147e78c802d124ca799566020c95edb.intermediate
  LD_LIBRARY_PATH=/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/lib.host:/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/src/inspector/protocol /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/include/inspector; python ../../deps/v8/third_party/inspector_protocol/code_generator.py --jinja_dir ../../deps/v8/third_party --output_base "/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/src/inspector" --config ../../deps/v8/src/inspector/inspector_protocol_config.json --inspector_protocol_dir ../../deps/v8/third_party/inspector_protocol
  touch 2779fdd8178f14f16658de638cda988982eede15.intermediate
  LD_LIBRARY_PATH=/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/lib.host:/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/lib.target:$LD_LIBRARY_PATH; export LD_LIBRARY_PATH; cd ../tools/v8_gypfiles; mkdir -p /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/src/inspector/protocol /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/include/inspector; python ../../deps/v8/third_party/inspector_protocol/code_generator.py --jinja_dir ../../deps/v8/third_party --output_base "/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root/src/inspector" --config ../../deps/v8/src/inspector/inspector_protocol_config.json --inspector_protocol_dir ../../deps/v8/third_party/inspector_protocol
  g++ -m32 -o /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/codegen/arm/assembler-arm.o ../deps/v8/src/codegen/arm/assembler-arm.cc '-DV8_GYP_BUILD' '-DV8_TYPED_ARRAY_MAX_SIZE_IN_HEAP=64' '-D__STDC_FORMAT_MACROS' '-DV8_TARGET_ARCH_ARM' '-DCAN_USE_ARMV7_INSTRUCTIONS' '-DV8_EMBEDDER_STRING="-node.67"' '-DENABLE_DISASSEMBLER' '-DV8_PROMISE_INTERNAL_FIELD_COUNT=1' '-DENABLE_MINOR_MC' '-DOBJECT_PRINT' '-DV8_CONCURRENT_MARKING' '-DV8_ARRAY_BUFFER_EXTENSION' '-DV8_ENABLE_LAZY_SOURCE_POSITIONS' '-DV8_USE_SIPHASH' '-DDISABLE_UNTRUSTED_CODE_MITIGATIONS' '-DV8_WIN64_UNWINDING_INFO' '-DV8_ENABLE_REGEXP_INTERPRETER_THREADED_DISPATCH' '-DV8_SNAPSHOT_COMPRESSION' '-DUSE_EABI_HARDFLOAT=0' -I/home/rshi0212/ea4500-lede/staging_dir/target-arm_xscale_musl_eabi/usr/include -I../deps/v8 -I../deps/v8/include -I/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/inspector-generated-output-root -I../deps/v8/third_party/inspector_protocol -I/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/torque-output-root -I/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj/gen/generate-bytecode-output-root -I../deps/v8/third_party/zlib -I../deps/v8/third_party/zlib/google  -pthread -Wno-unused-parameter -Wno-return-type -fno-strict-aliasing -m32 -O3 -fno-omit-frame-pointer -fdata-sections -ffunction-sections -O3 -fno-rtti -fno-exceptions -std=gnu++1y -MMD -MF /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/.deps//home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/codegen/arm/assembler-arm.o.d.raw   -c
../deps/v8/src/codegen/arm/assembler-arm.cc:178:2: error: #error "CAN_USE_ARMV7_INSTRUCTIONS should match CAN_USE_VFP3_INSTRUCTIONS"
  178 | #error "CAN_USE_ARMV7_INSTRUCTIONS should match CAN_USE_VFP3_INSTRUCTIONS"
      |  ^~~~~
make[5]: *** [tools/v8_gypfiles/v8_base_without_compiler.host.mk:637: /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/out/Release/obj.host/v8_base_without_compiler/deps/v8/src/codegen/arm/assembler-arm.o] Error 1
rm c071ca2dc147e78c802d124ca799566020c95edb.intermediate c8ff977637accc1103a2efbe1a9beefcc1e14dca.intermediate 9cfc94d7fa4766e803d7a4be01aefe91bd0d4611.intermediate 2779fdd8178f14f16658de638cda988982eede15.intermediate
make[4]: *** [Makefile:104: node] Error 2
make[4]: Leaving directory '/home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1'
make[3]: *** [Makefile:204: /home/rshi0212/ea4500-lede/build_dir/target-arm_xscale_musl_eabi/node-v14.17.1/.built] Error 2
make[3]: Leaving directory '/home/rshi0212/ea4500-lede/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#19.82#0.81#20.39
    ERROR: package/feeds/packages/node failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/node/compile] Error 1
make[2]: Leaving directory '/home/rshi0212/ea4500-lede'
make[1]: *** [package/Makefile:110: /home/rshi0212/ea4500-lede/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rshi0212/ea4500-lede'
make: *** [/home/rshi0212/ea4500-lede/include/toplevel.mk:230: world] Error 2

[image]

coolsnowwolf, 2021-08-04:
EA4500 CPU 不具备 FPU 硬件浮点，不支持 Nodejs

rshi0212, 2021-08-04:
哪个包用到了Nodejs？

发自我的iPhone

coolsnowwolf, 2021-08-04:
网易云音乐和jd签到

closed 2021-08-05 by coolsnowwolf