coolsnowwolf/lede #7604: ERROR: package/libs/pcre failed to build.

weigeloveu, 2021-08-04:
## 1.关于你要提交的问题
 ERROR: package/libs/pcre failed to build.
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
make -j8 download V=s之后会出现这个错误

### (2) 路由器型号和固件版本

A：HC5962

### (3) 详细日志

A： 
HEAD is now at d63ff86 Finish 2.5.3 ... fix some small build issue in last version
Packing checkout...
100   466  100   466    0     0    131      0  0:00:03  0:00:03 --:--:--   288
100   333  100   333    0     0     61      0  0:00:05  0:00:05 --:--:--   369
  0     0    0     0    0     0      0      0 --:--:--  0:00:06 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   360  100   360    0     0    192      0  0:00:01  0:00:01 --:--:--   192
make[2]: Leaving directory '/home/v/lede/feeds/helloworld/shadowsocksr-libev'
time: package/feeds/helloworld/shadowsocksr-libev/download#9.17#1.49#21.77
100   333  100   333    0     0     88      0  0:00:03  0:00:03 --:--:--   370
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://netix.dl.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   291    0     0    0     0      0      0 --:--:--  0:00:02 --:--:--     0
100   464  100   464    0     0    105      0  0:00:04  0:00:04 --:--:--   261
100   333  100   333    0     0     53      0  0:00:06  0:00:06 --:--:--   405
  0     0    0     0    0     0      0      0 --:--:--  0:00:06 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://freefr.dl.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   292    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100   466  100   466    0     0    124      0  0:00:03  0:00:03 --:--:--   262
100   333  100   333    0     0     59      0  0:00:05  0:00:05 --:--:--   381
  0     0    0     0    0     0      0      0 --:--:--  0:00:06 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://downloads.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   360  100   360    0     0    179      0  0:00:02  0:00:02 --:--:--   179
100   333  100   333    0     0     84      0  0:00:03  0:00:03 --:--:--   361
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://netix.dl.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0   291    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100   464  100   464    0     0    166      0  0:00:02  0:00:02 --:--:--  1237
100   333  100   333    0     0     69      0  0:00:04  0:00:04 --:--:--   333
  0     0    0     0    0     0      0      0 --:--:--  0:00:04 --:--:--     0
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to udomain.dl.sourceforge.net:443
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://freefr.dl.sourceforge.net/pcre/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
[... 74 lines cut ...]
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/pcre-8.45.tar.bz2
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
No more mirrors to try - giving up.
make[2]: *** [Makefile:113: /home/v/lede/dl/pcre-8.45.tar.bz2] Error 2
make[2]: Leaving directory '/home/v/lede/package/libs/pcre'
time: package/libs/pcre/download#1.11#0.25#77.61
    ERROR: package/libs/pcre failed to build.
make[1]: *** [package/Makefile:116: package/libs/pcre/download] Error 1
make[1]: Leaving directory '/home/v/lede'
make[1]: Entering directory '/home/v/lede'
make[2]: Entering directory '/home/v/lede/target/linux'
make[3]: Entering directory '/home/v/lede/target/linux/ramips'
make[3]: Nothing to be done for 'download'.
make[3]: Leaving directory '/home/v/lede/target/linux/ramips'
make[2]: Leaving directory '/home/v/lede/target/linux'
time: target/linux/download#0.07#0.01#0.07
make[1]: Leaving directory '/home/v/lede'

geniusyjiang, 2021-08-04:
遇到同样的问题，打开这些源查看了，里面都还没有8.45，只有8.44

Boos4721, 2021-08-04:
> 遇到同样的问题，打开这些源查看了，里面都还没有8.45，只有8.44

`https://sourceforge.net/projects/pcre/files/pcre/8.45`
?

ja5onwu, 2021-08-04:
https://ftp.pcre.org/pub/pcre/pcre-8.45.tar.bz2
手动下载,然后拷贝进去lede/dl/下

geniusyjiang, 2021-08-04:
> https://ftp.pcre.org/pub/pcre/pcre-8.45.tar.bz2
> 手动下载,然后拷贝进去lede/dl/下

最后只有这样才解决了，谢谢

closed 2021-08-05 by coolsnowwolf