coolsnowwolf/lede #7606: 编译成功的Newifi D1固件不能使用

sclsyin, 2021-08-05:
编译好的固件，通用网页更新时报不能识别的固件。通过Bread刷入后不能启动。

huming2207, 2021-08-05:
[image]

sclsyin, 2021-08-05:
问题是取不到日志啊。不能开机了怎么取。

yaott2020-666, 2021-08-05:
文件名，错误截图都可以，不然怎么帮你解决问题

sclsyin, 2021-08-05:
openwrt-ramips-mt7621-lenovo_newifi-d1-squashfs-sysupgrade.bin 网页更新时报不能识别的固件。
openwrt-ramips-mt7621-lenovo_newifi-d1-initramfs-kernel.bin 使用Bread刷入成功，但不能启动。

EnnawYang, 2021-08-05:
d1需要刷initramfs ?

yaott2020-666, 2021-08-05:
openwrt-ramips-mt7621-lenovo_newifi-d1-squashfs-sysupgrade.bin 这个用breed刷入石试试

sclsyin, 2021-08-05:
OK.晚点试一下!

sclsyin, 2021-08-05:
[image]

[image]

EnnawYang, 2021-08-05:
> 
> 这个是网页更新时错误的截图和系统日志。
> 
前几天MT7621有重要变更，请不要从网页直接更新

sclsyin, 2021-08-05:
那我从Bread更新，是更新squashfs 还是initramfs ？？？？

EnnawYang, 2021-08-05:
> squashfs

squashfs

sclsyin, 2021-08-05:
谢谢，晚上我试一下。

closed 2021-08-06 by sclsyin