coolsnowwolf/lede #7608: bug

lexin8, 2021-08-05:
config uhttpd 'main'
        list listen_http '0.0.0.0:80'
        list listen_http '[::]:80'
        list listen_https '0.0.0.0:443'
        list listen_https '[::]:443'
        option redirect_https '0'
        option home '/www'
        option max_requests '3'
        option max_connections '100'
        option cert '/etc/uhttpd.crt'
        option key '/etc/uhttpd.key'
        option cgi_prefix '/cgi-bin'
        list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
        option network_timeout '30'
        option http_keepalive '20'
        option tcp_keepalive '1'
        option ubus_prefix '/ubus'
        option script_timeout '3600'
        list index_page 'cgi-bin/luci'
        option rfc1918_filter '0'

config cert 'defaults'
        option days '730'
        option key_type 'ec'
        option bits '2048'
        option ec_curve 'P-256'
        option country 'ZZ'
        option state 'Somewhere'
        option location 'Unknown'
        option commonname 'OpenWrt'
[image]

9l, 2021-08-06:
自签证书这个提示没毛病,没有有效证书的,不要开https呀

yaott2020-666, 2021-08-06:
这个不是bug，自签证书都会这样，要么别开https，要么套上有效ssl证书

closed 2021-08-06 by coolsnowwolf

lexin8, 2021-08-07:
> 自签证书这个提示没毛病,没有有效证书的,不要开https呀

全新刷机默认状态，也没有设置https，七月份的源码没有这个情况！