coolsnowwolf/lede #7609: 版本问题 为什么从5.10又降回了5.4

xiangfeigao, 2021-08-05:
只是想问问大雕版本控制为什么又降下来了？？
路由器:hc5962

AmadeusGhost, 2021-08-05:
因为 5.10 性能更差，就这样

dwj0, 2021-08-05:
我自己编译了5.10版本，写进路由器后直接不能启动了，不知道啥原因

zhouxiaolong920, 2021-08-06:
没感觉啊。。难道是我的x86硬件性能好，无感觉？

zhouxiaolong920, 2021-08-06:
5.10，已经更新了大半年了吧。。这时候回滚。。。

galaxyskyknight, 2021-08-06:
> 5.10，已经更新了大半年了吧。。这时候回滚。。。

6月份才更新吧？哪有大半年？不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？

EnnawYang, 2021-08-06:
> 
> 不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？

要稳定可以去用隔壁 `coolsnowwolf/openwrt` 仓库，4.x内核。主要是很多人要求上5.10内核，开发者就满足他们的要求，当然出了什么问题开发者也可以不管（当然实际也出现了一些暂时无法解决问题

zhouxiaolong920, 2021-08-06:
@EnnawYang 
1. 别这么大火气，我们都是支持大神的lede
2. bug存在没啥问题，问题是大版本回退不应该

EnnawYang, 2021-08-06:
> 
> 1. 别这么大火气，我们都是支持大神的lede
> 2. bug存在没啥问题，问题是大版本回退不应该

我只是实事求是，但这些也只针对那些过分追新的人

eightsheep67, 2021-08-06:
我X86的5.10也正常，感觉非常稳定。还好前天编译了一个回滚前的最新版本……

eightsheep67, 2021-08-06:
有人解释一下R21.8.6这个版本号对应的是什么，相当于是系统版本号吗？

AmadeusGhost, 2021-08-06:
要追新的是你们，说新的有很多 bug 的也是你们，所以我们开发者就只能爬了是吧

1715173329, 2021-08-06:
我建议自行开发，或者打钱定制。

Beginner-Go, 2021-08-06:
要新上5.14啊，5.10算什么

QiuSimons, 2021-08-06:
> > 5.10，已经更新了大半年了吧。。这时候回滚。。。
> 
> 6月份才更新吧？哪有大半年？不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？

整挺好，是不是还要搞个road map，然后制定一个计划书，然后对各个主要组件做版本锁定？

eightsheep67, 2021-08-06:
我觉得就是抱着学习的态度，开发者有自己规划，如果有意见或者有疑问都可以沟通，但是不要出言不逊说什么开发者没有规划。
相信大部分人也只是想知道大版本回滚的原因，可能有什么潜在问题大家目前没有遇到，没有任何指责开发者的意思，我们都是站在他们奠定好的基石上进行学习和使用的。
如果挑三拣四的，自己开发去吧

gurongwei, 2021-08-06:
> > > 5.10，已经更新了大半年了吧。。这时候回滚。。。
> > 
> > 
> > 6月份才更新吧？哪有大半年？不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？
> 
> 整挺好，是不是还要搞个road map，然后制定一个计划书，然后对各个主要组件做版本锁定？

那不得得整个PR委员会~

zhouxiaolong920, 2021-08-06:
@AmadeusGhost 
     对于追新，意味着，快和稳定，好多人是这样认为，我每天都会看一下commit的啥，大部份都是插件升级.
-----
1. 我觉得好多人提bug，并不一定是真的bug，你们可以选择跟进和忽略。
2. 按你们自己的步调来呗？除了openwrt，难道选择的软路由系统还有很多？
3. 我是用x86的，用的插件就几个而已，openwrt，还是非常稳定的。

---

我们没啥抱怨的，更希望的是，openwrt能更好，快，稳定。

msylgj, 2021-08-06:
> ## @AmadeusGhost
> 对于追新，意味着，快和稳定，好多人是这样认为，我每天都会看一下commit的啥，大部份都是插件升级.
> 1. 我觉得好多人提bug，并不一定是真的bug，你们可以选择跟进和忽略。
> 2. 按你们自己的步调来呗？除了openwrt，难道选择的软路由系统还有很多？
> 3. 我是用x86的，用的插件就几个而已，openwrt，还是非常稳定的。
> 
> 我们没啥抱怨的，更希望的是，openwrt能更好，快，稳定。

快来看看5.14吧 都已经rc4了 再不抓紧该出5.15了…

AmadeusGhost, 2021-08-06:
我只能说显然你们没有去修过或者看过这些 issues，很多人可不会这么客气说这个 bug 能不能修下。我们不管是不修（不理）或者修不了直接回退甚至修了你们都会觉得是开发者的问题，这里不就是明显的例子吗？
另外说一句，openwrt 官方的 snapshot 分支是开发分支，真要追求稳定应该选择 release 分支，想必你们没有人会做此选择。

OleStore, 2021-08-06:
其实大家觉得很突然，5.10降级5.4，issues估计大伙都有看，这那问题都有

galaxyskyknight, 2021-08-06:
不尊重用户（特别是初级用户），表现为傲慢（爱理不理），用户无权提需求和了解why change，都是我说了算的潜在思维模式是这里的一贯风格。我可能说的比较直白，得罪人，没所谓了，你们也可以喷我。

EnnawYang, 2021-08-06:
> 不尊重用户（特别是初级用户），表现为傲慢（爱理不理），用户无权提需求和了解why change，都是我说了算的潜在思维模式是这里的一贯风格。我可能说的比较直白，得罪人，没所谓了，你们也可以喷我。

[image]
本项目不是商业项目，为什么要惯着你？

coolsnowwolf, 2021-08-06:
首先你还是要搞清楚，

1.这个基本上是我一个人在维护的项目  
2. 为何要满足所有人的需求？
3. 我觉得我比你们大多数人专业，所以我做决策

有问题？

nicholas-opensource, 2021-08-06:
本项目没有多余人手去做Buildtest，也没有多余人手去写Changelog，欢迎您可以帮忙来写。或者如果您有任何不满意的地方，show me the code，PR is welcome。另外欢迎donate or help，我们或许可以拥有固定的人手来处理这些问题。

coolsnowwolf, 2021-08-06:
Talk is cheap. Show me the code

hcym, 2021-08-06:
ok啦，大雕定就完事了么，好用最好，不好用还有其他选择，新内核最多对arm有点意思，其他基本负面影响

lonee6, 2021-08-06:
> 有人解释一下R21.8.6这个版本号对应的是什么，相当于是系统版本号吗？

21年8月6号版本，很难理解吗？

Ailick, 2021-08-06:
> 不尊重用户（特别是初级用户），表现为傲慢（爱理不理），用户无权提需求和了解why change，都是我说了算的潜在思维模式是这里的一贯风格。我可能说的比较直白，得罪人，没所谓了，你们也可以喷我。

首先要理解到，这不是给普通用户用的，而是给具备动手能力的人。
其次这个不是产品，也不盈利，只是作者和社区成员处于爱好而维护的作品，简称为爱发电。
最后你可以提需求，但是做不做得看有没有人愿意免费加班。
忠告：为爱发电的项目，普通伸手党一般付不起维护费用，且用且珍惜。

eightsheep67, 2021-08-06:
> > 有人解释一下R21.8.6这个版本号对应的是什么，相当于是系统版本号吗？
> 
> 21年8月6号版本，很难理解吗？

我的意思是，比如5.10、5.4这些是内核版本号，R21.8.6是什么组件的版本号，是Luci界面还是什么的版本号。
WEB管理界面就写个固件版本号，但有时候内核版本更新的时候这个固件版本号也不一定更新，虽然这个问题很弱智，但是我确实找不到答案，只好问一问大佬

liuyanantj, 2021-08-06:
作者挺辛苦的去维护 作为使用者 免费使用功能 所以得抱着敬畏之心吧 不能花钱去支持 最起码要有语言上的尊重吧 没事哔哔 开发者得解决bug 还得听这些话 换作自己你能坚持么

EnnawYang, 2021-08-06:
> 
> R21.8.6是什么组件的版本号

项目拥有者自定的版本号，和其他组件没有联系

coolsnowwolf, 2021-08-06:
R 多少版本是属于阶段性标记版本，觉得本月代码到这个版本比较稳定了就做个象征性标记

eightsheep67, 2021-08-06:
> > R21.8.6是什么组件的版本号
> 
> 项目拥有者自定的版本号，和其他组件没有联系

明白了，谢谢
因为我经常会通过各个版本号能去搜索到一些相关的类似changelog，release notes这样的东西，想知道有什么新变化。

eightsheep67, 2021-08-06:
> R 多少版本是属于阶段性标记版本，觉得本月代码到这个版本比较稳定了就做个象征性标记

谢谢大佬的回复！也再次感谢大佬对项目的维护。请忽略那些没有感恩之心的人的废话。我们还是拥护所有辛苦维护本项目的大佬们的规划方向。

ricksuzade-maker, 2021-08-06:
这回滚了内核，是不是最好不保留设置刷呢？然后全部重头配置

baodp, 2021-08-06:
> 这回滚了内核，是不是最好不保留设置刷呢？然后全部重头配置

好像是要不保存设置刷固件

coolsnowwolf, 2021-08-06:
可以保留配置，无影响

281677160, 2021-08-06:
呵呵，感谢大雕付出，我是无所谓什么内核版本的，有用就好了，总比用官方的要自己折腾的来的舒服，大雕的，天灵的，还有Lienol大的，我都换着玩的，感谢各位大佬无私奉献

qqsir-dev, 2021-08-06:
这么多大神出现了，感觉真好！顺便问一句，新内核对软路由有影响吗？

reaiyaogun, 2021-08-06:
保留5.10内核的方法：
make menuconfig --> 进入"Global build settings"菜单 --> 勾选"Use the testing kernel version"选项 --> 保存退出，完毕。

minlang112, 2021-08-07:
凡是认真阅读一下pull requests和commits，应该也会知道开发者遇到了什么问题和反馈……

还有扯什么尊重，觉得别人该伺候你，我是建议掏钱，越多越尊重

galaxyskyknight, 2021-08-07:
> 凡是认真阅读一下pull requests和commits，应该也会知道开发者遇到了什么问题和反馈……
> 
> 还有扯什么尊重，觉得别人该伺候你，我是建议掏钱，越多越尊重

什么叫伺候？不管是商业还是开源软件，与用户良性互动尊重用户，了解用户需求，难道不是一个开发者基本的素质？如果不需要用户输入那就自己乐呗，把issue板块关了就好了。（我指的良性，是指沟通问题时，不因彼此知识掌握差异的不对等而居高临下，在交流时保持彼此语气，口吻上的谦逊，平等，而不是一副颐指气使，给人我掌握比你，我付出我维护多，我对你就可以蔑视无视的态度，自己看看这里的有些人回复用户提出问题的某些语气.....）至于钱不钱的，对不起，我还真掏过，虽然不多，一杯咖啡钱而已,只是象征性的，但性质上我相信比绝大多数没掏过的有资格说钱的问题。所以别公开说什么掏钱越多越尊重这种拉低品味的话，只会显得自己很Low。

galaxyskyknight, 2021-08-07:
> > 凡是认真阅读一下pull requests和commits，应该也会知道开发者遇到了什么问题和反馈……
> > 还有扯什么尊重，觉得别人该伺候你，我是建议掏钱，越多越尊重
> 
> 什么叫伺候？不管是商业还是开源软件，与用户良性互动尊重用户，了解用户需求，难道不是一个开发者基本的素质？如果不需要用户输入那就自己乐呗，把issue板块关了就好了。（我指的良性，是指沟通问题时，不因彼此知识掌握差异的不对等而居高临下，在交流时保持彼此语气，口吻上的谦逊，平等，而不是你用户就只能对我唱赞歌和舔，不能有任何的不同意见和不同想法提出来，否则就是挑战大佬，变成一副颐指气使，给人我掌握比你，我付出我维护多，你没我懂，我对你就可以蔑视无视的态度，自己看看这里的有些人回复用户提出问题的某些语气.....）至于钱不钱的，对不起，我还真掏过，虽然不多，一杯咖啡钱而已,只是象征性的，但性质上我相信比绝大多数没掏过的有资格说钱的问题。所以别公开说什么掏钱越多越尊重这种拉低品味的话，只会显得自己很Low。

这是我与n2n的开发者交流一个feature test和issue的互动，人家是一个外国人，看得懂英文的人可以看看，人家老外回答问题字里行间都充满着谦逊和patient,从来不是我们这里这个样子的。
https://github.com/ntop/n2n/pull/741#issue-696404187

guochan2019, 2021-08-07:
退肯定有退的原因，支持

ricksuzade-maker, 2021-08-07:
说实话，5.10命运很坎坷，正式版没几天就爆大bug，连核心开发者都不想做长期维护，后来才勉强救回来。大雕既然决定回退就退吧，等年底的新的lts出来再说。话说，有空可否升一下5.4.138? ：)

Lrvineshaw, 2021-08-07:
@galaxyskyknight 5.10bug太多，op官方都没法彻底解决，你要想用可以自己在makefile里面修改版本号

stephenchou888, 2021-08-07:
@galaxyskyknight L大开源这项目本身就有很大压力
您一上来就"6月份才更新吧？哪有大半年？不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？" 
悄无声息加有点规划，这语句用词给谁看就不爽换位思考吧。

we2oxy, 2021-08-07:
@galaxyskyknight 你自己整一个不也行吗，别人做的东西用的着和你商量？

galaxyskyknight, 2021-08-07:
> @galaxyskyknight 你自己整一个不也行吗，别人做的东西用的着和你商量？

所以根本不想和你这种no sense的人沟通交流

AmadeusGhost, 2021-08-07:
> 不尊重用户（特别是初级用户），表现为傲慢（爱理不理），用户无权提需求和了解why change，都是我说了算的潜在思维模式是这里的一贯风格。

[image]
[image]
你给我翻译翻译什么叫“用户无权提需求”？还是说只有您的需求才算数呢😅
至于尊不尊重初级用户这个，我的态度就是伸手党爬，百度都懒得用那自然没什么好说的。
了解 why change 这个... 简单回复一下你们觉得敷衍，不回复就是没人管，我们不可能闲到每个 commit 都详细的说明为什么。
您要真想这里变得更好，那就去整理并回复 issues，像新的内核有问题相关的 issues 简单翻翻都有。
> 人家老外回答问题字里行间都充满着谦逊和patient,从来不是我们这里这个样子的。

请您先翻一下这些 issues 再来说这些 ... 但凡有 watch 或者每个都看下甚至去管几个都不会这么说。
我就这么说了吧，在这里，一个稍微明显一点的更改不管做与不做都会被喷。(e.g.) 甚至我之前动了下 mt7621 开源驱动的版本都被闭源驱动用户和 x86 用户狂喷，尽管这与他们毫不相干。我们不是受虐狂，不要指望每个 issue 我们都能好好回复 ...
说白了，你的态度是别人应该怎样而不是我自己要怎样才能让这里变好。这里不欢迎只会嘴上指点江山的人，有种就拿出实际行动去做，我相信不管每个人水平如何只要想帮忙有搜索引擎的帮助简单的做点贡献（回复 issues 也算）并非什么难事。
> 我可能说的比较直白，得罪人，没所谓了，你们也可以喷我。

同样你要喷我就喷吧，强烈建议指名道姓地喷，我只能表示没人喷我都不习惯了😅

Boos4721, 2021-08-07:
> 什么叫伺候？不管是商业还是开源软件，与用户良性互动尊重用户，了解用户需求，难道不是一个开发者基本的素质？如果不需要用户输入那就自己乐呗，把issue板块关了就好了。（我指的良性，是指沟通问题时，不因彼此知识掌握差异的不对等而居高临下，在交流时保持彼此语气，口吻上的谦逊，平等，而不是你用户就只能对我唱赞歌和舔，不能有任何的不同意见和不同想法提出来，否则就是挑战大佬，变成一副颐指气使，给人我掌握比你， [...]

😅那您自己fork下revert不就行了 去看了看您的repo commits您貌似也不会git 😅

Beginner-Go, 2021-08-07:
人家是伸手党，那会git这种操作

1715173329, 2021-08-07:
先看看下什么叫做提问的智慧吧，没必要拿什么国外的来比。提问题可以，但是提了傻逼问题还一副趾高气扬的样子，你在哪儿都只有滚蛋。
我求你解决问题啊？

别说捐了多少，可以联系我们定制固件，两万起步。

1715173329, 2021-08-07:
对了，这位 @galaxyskyknight 也是著名的友善用户了，就他那个 r4s fork 即可看出智商情况。

galaxyskyknight, 2021-08-07:
> 先看看下什么叫做提问的智慧吧，没必要拿什么国外的来比。提问题可以，但是提了傻逼问题还一副趾高气扬的样子，你在哪儿都只有滚蛋。
> 我求你解决问题啊？
> 
> 别说捐了多少，可以联系我们定制固件，两万起步。

你先滚蛋，对不起，被逼自卫骂人，只是把这两个字还给你，因为你的傻逼智慧只会张口说这两个字。

galaxyskyknight, 2021-08-07:
> 😅那您自己fork下revert不就行了 去看了看您的repo commits您貌似也不会git 😅

笑话，只不过fork一个后来废弃的仓库，跟着放那废弃了没动，就因此发现了一个攻击别人的由头，自诩高人一等讽刺别人不会revert, 想不到码农里也有孔乙己。

Boos4721, 2021-08-07:
> 笑话，只不过fork一个后来废弃的仓库，跟着放那废弃了没动，就因此发现了一个攻击别人的由头，自诩高人一等讽刺别人不会revert, 想不到码农里也有孔乙己。

😅 您就是笑话本人吧

galaxyskyknight, 2021-08-07:
> 😅 您就是笑话本人吧

无所谓了，党同伐异，同行相轻，中国人的劣根性就是如此。你们爱玩自己玩吧，恕不奉陪，辩论你们爱怎么辩随意那是自由，保持不主动骂人讲脏话这是起码的底线，但是骂人的话，你就当骂你自己就行了。我不会再回复。无聊。

nicholas-opensource, 2021-08-07:
please DO NOT use any line of code written by Chinese. Including [openwrt/openwrt](https://github.com/openwrt/openwrt)

z826540272, 2021-08-07:
> > 不过当初火急火燎上马5.10，现在又悄无声息的回滚，开发者是否能有点规划呢？
> 
> 要稳定可以去用隔壁 `coolsnowwolf/openwrt` 仓库，4.x内核。主要是很多人要求上5.10内核，开发者就满足他们的要求，当然出了什么问题开发者也可以不管（当然实际也出现了一些暂时无法解决问题

有很多吗,我怎么没看到

[51 more replies]

closed 2021-08-12 by coolsnowwolf