coolsnowwolf/lede #7611: Turbo ACC 必现BUG

PyXiaoxin, 2021-08-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：开启了截图中未运行的三个功能，路由器运行1-2天必然会死机。

### (2) 路由器型号和固件版本

A：P&W R619AC 128M OpenWrt R21.7.26 / LuCI Master (git-21.211.30217-03784c3) 
      内核版本：5.4.132

### (3) 详细日志

A：以前也遇到过，我经过几轮排查终于定位到是这个功能引发的，但是没法采集到日志，因为出问题的时候SSH WEB都无法登录，甚至IP也无法ping通，只能通过bread重刷固件恢复，然后日志就没了。
[image]

xcf13363175, 2021-08-06:
装了什么插件了吗

PyXiaoxin, 2021-08-06:
这是编译之前的config文件
CONFIG_TARGET_ipq40xx=y
CONFIG_TARGET_ipq40xx_DEVICE_p2w_r619ac-128m=y
CONFIG_TARGET_BOARD="ipq40xx"
CONFIG_NODEJS_14=y
CONFIG_NODEJS_ICU_NONE=y
# CONFIG_PACKAGE_UnblockNeteaseMusic-Go is not set
CONFIG_PACKAGE_appfilter=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_boost=y
CONFIG_PACKAGE_boost-date_time=y
CONFIG_PACKAGE_boost-program_options=y
CONFIG_PACKAGE_boost-system=y
# CONFIG_PACKAGE_dns2socks is not set
CONFIG_PACKAGE_frpc=y
CONFIG_PACKAGE_frps=y
CONFIG_PACKAGE_ipt2socks=y
CONFIG_PACKAGE_iputils-arping=y
CONFIG_PACKAGE_kcptun-client=y
CONFIG_PACKAGE_kcptun-config=y
CONFIG_PACKAGE_kmod-ifb=y
CONFIG_PACKAGE_kmod-oaf=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libcap=y
CONFIG_PACKAGE_libmaxminddb=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_libuv=y
CONFIG_PACKAGE_libwebsockets-full=y
CONFIG_PACKAGE_lua-maxminddb=y
CONFIG_PACKAGE_luasocket=y
CONFIG_PACKAGE_luci-app-eqos=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-frps=y
CONFIG_PACKAGE_luci-app-jd-dailybonus=y
CONFIG_PACKAGE_luci-app-netdata=y
CONFIG_PACKAGE_luci-app-oaf=y
CONFIG_PACKAGE_luci-app-serverchan=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-vssr=y
CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-webadmin=y
CONFIG_PACKAGE_luci-app-wrtbwmon=y
CONFIG_PACKAGE_luci-i18n-eqos-zh-cn=y
CONFIG_PACKAGE_luci-i18n-frpc-zh-cn=y
CONFIG_PACKAGE_luci-i18n-frps-zh-cn=y
CONFIG_PACKAGE_luci-i18n-netdata-zh-cn=y
CONFIG_PACKAGE_luci-i18n-oaf-zh-cn=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-webadmin-zh-cn=y
CONFIG_PACKAGE_luci-i18n-wrtbwmon-zh-cn=y
CONFIG_PACKAGE_luci-theme-ifit=y
CONFIG_PACKAGE_netdata=y
CONFIG_PACKAGE_node=y
CONFIG_PACKAGE_tc-mod-iptables=y
CONFIG_PACKAGE_tc-tiny=y
CONFIG_PACKAGE_trojan=y
CONFIG_PACKAGE_ttyd=y
CONFIG_PACKAGE_xray-plugin=y
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y

cncqcyc, 2021-08-06:
我的是x86，我刚刚出现了路由器CPU 100%占用的情况，应该是路由器死机了，因为是远程访问，现在已经访问不到了，在离线之前，日志里有一个uhttpd：sed:write error的错误日志，我没有装tuoboacc

zhouxiaolong920, 2021-08-06:
我之前，openwrt不稳定，偶尔，几天有重启。tuoboacc,之类的，
后来，我把这个取消了。什么shortfe之类的。加速。都不需要。

openwrt稳定运行2个月。

closed 2021-08-06 by coolsnowwolf

Cathgao, 2021-08-07:
```
型号 | P&W R619AC 128M (CpuMark : 10514.210613 Scores)
架构 | ARMv7 Processor rev 5 (v7l) x 4 (896MHz, 63.0°C 70.0°C )
固件版本 | OpenWrt R21.7.26 / LuCI Master (git-21.211.30217-03784c3)
内核版本 | 5.4.132
```
跟你一样路由器，除了DNS缓存全开，没任何问题
[image]

freeoa, 2021-08-29:
我之前也遇到这个情况只要一打开bbr就会死机,还有就算不重启了,如果adsl重新拔号了,但是ssr+一定手动重置一下才能正常科学上网