coolsnowwolf/lede #7616: rac2v1k 512M重新启动失败

kisaragychihaya, 2021-08-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：重新启动有概率无法进入系统

### (2) 路由器型号和固件版本

A：rac2v1k 512m

### (3) 详细日志

A：TTL日志
```
Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.132 (kiana@debian) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3616-6f2494ee5)) #0 SMP Fri Aug 6 07:24:28 2021
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Askey RT4230W REV6
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] percpu: Embedded 15 pages/cpu s30220 r8192 d23028 u61440
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 252224
[    0.000000] Kernel command line: rootfstype=squashfs noinitrd
[    0.000000] Bootloader command line (ignored): console=ttyMSM0,115200n8
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 996168K/1015808K available (6256K kernel code, 198K rwdata, 1568K rodata, 1024K init, 232K bss, 19640K reserved, 0K cma-reserved, 229376K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x4ec with crng_init=0
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000008] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000022] Switching to timer-based delay loop, resolution 160ns
[    0.000267] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000289] pid_max: default: 32768 minimum: 301
[    0.000516] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000542] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001626] CPU: Testing write buffer coherency: ok
[    0.002783] Setting up static identity map for 0x42300000 - 0x42300060
[    0.002965] rcu: Hierarchical SRCU implementation.
[    0.003467] smp: Bringing up secondary CPUs ...
[    0.005405] smp: Brought up 1 node, 2 CPUs
[    0.005422] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.005433] CPU: All CPU(s) started in SVC mode.
[    0.017730] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.017885] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017916] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.018069] pinctrl core: initialized pinctrl subsystem
[    0.019236] NET: Registered protocol family 16
[    0.019559] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.021076] cpuidle: using governor ladder
[    0.021144] cpuidle: using governor menu
[    0.056909] usbcore: registered new interface driver usbfs
[    0.056992] usbcore: registered new interface driver hub
[    0.057062] usbcore: registered new device driver usb
[    0.057120] pps_core: LinuxPPS API ver. 1 registered
[    0.057133] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
[... 381 lines cut ...]
[   44.674509] Buffer I/O error on dev mtdblock17, logical block 0, async page read
[   44.691816] Buffer I/O error on dev mtdblock17, logical block 0, async page read
[   49.320061] print_req_error: 31 callbacks suppressed
[   49.320067] blk_update_request: I/O error, dev mtdblock12, sector 0 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
[   49.330476] blk_update_request: I/O error, dev mtdblock12, sector 8 op 0x0:(READ) flags 0x80700 phys_seg 3 prio class 0
[   49.340911] blk_update_request: I/O error, dev mtdblock12, sector 16 op 0x0:(READ) flags 0x80700 phys_seg 2 prio class 0
[   49.350381] blk_update_request: I/O error, dev mtdblock12, sector 24 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[   49.370594] blk_update_request: I/O error, dev mtdblock12, sector 0 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[   49.370620] buffer_io_error: 2 callbacks suppressed
[   49.370624] Buffer I/O error on dev mtdblock12, logical block 0, async page read
[   49.395390] blk_update_request: I/O error, dev mtdblock12, sector 32 op 0x0:(READ) flags 0x80700 phys_seg 7 prio class 0
[   49.419703] blk_update_request: I/O error, dev mtdblock12, sector 40 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[   49.449855] blk_update_request: I/O error, dev mtdblock12, sector 48 op 0x0:(READ) flags 0x80700 phys_seg 6 prio class 0
[   49.505656] blk_update_request: I/O error, dev mtdblock12, sector 56 op 0x0:(READ) flags 0x80700 phys_seg 5 prio class 0
[   49.534952] blk_update_request: I/O error, dev mtdblock12, sector 64 op 0x0:(READ) flags 0x80700 phys_seg 4 prio class 0
[   49.669841] Buffer I/O error on dev mtdblock12, logical block 1, async page read
[   49.693536] Buffer I/O error on dev mtdblock13, logical block 0, async page read
[   49.705320] Buffer I/O error on dev mtdblock13, logical block 0, async page read
[   49.820522] Buffer I/O error on dev mtdblock14, logical block 0, async page read
[   49.848891] Buffer I/O error on dev mtdblock14, logical block 0, async page read
[   50.135789] Buffer I/O error on dev mtdblock16, logical block 0, async page read
[   50.139108] Buffer I/O error on dev mtdblock16, logical block 0, async page read
[   50.283106] Buffer I/O error on dev mtdblock17, logical block 0, async page read
[   50.329556] Buffer I/O error on dev mtdblock17, logical block 0, async page read
```

closed 2021-08-06 by coolsnowwolf

AmadeusGhost, 2021-08-07:
这个应该只是启动时间太长了，把 block-mount 卸了可以解决。

kisaragychihaya, 2021-08-07:
是的，后来多等一会就发现确实是这样，谢谢解答